zigpy/zha-device-handlers #352: [BUG] Ikea blind not working with google home

blair287, 2020-05-13:
Not sure when this issue starts but this is whats happening.

1) if i control the ikea blind via HA UI up or down it works fine.

2) if i ask google to open or cloase (linked through nabu casa) it will start moving then stop after about 10cm this happens every time.

3) if i unlink the blind from google and relink it works for first go then back to the 10cm and stop.

this comes up sometimes with a different number everytime i ask google to open or close the blind.

Log Details (WARNING)
Logger: zigpy_deconz.zigbee.application
Source: components/zha/core/helpers.py:32
First occurred: 8:53:38 PM (2 occurrences)
Last logged: 8:53:48 PM

Error while sending 250 req id frame: 0xd0
Error while sending 18 req id frame: 0xd0

Adminiuga, 2020-05-13:
https://www.home-assistant.io/integrations/zha/#reporting-issues

However, if controlling from UI works fine, I fail to see how this would be ZHA's fault? 🤷‍♂️  Google integration should issue same  commands as UI You may want to  enable google integration (nabu casasa?) debug logging too

blair287, 2020-05-14:
> https://www.home-assistant.io/integrations/zha/#reporting-issues
> 
> However, if controlling from UI works fine, I fail to see how this would be ZHA's fault?  Google integration should issue same commands as UI You may want to enable google integration (nabu casasa?) debug logging too

I have arrived at this conclusion too powering off the nuc running HA and turning it on again solved the issue for few hours.

How do I turn on debugging for both ZHA and Nabu Casa. There must be another command like stop or maybe it's sending open or close twice causing the blind to pause.

Adminiuga, 2020-05-14:
https://www.home-assistant.io/integrations/zha/#reporting-issues how to turn on debug logging. Add `homeassistant.components.cloud` and I have no ideas which one is for google integration.

blair287, 2020-05-17:
Probably not zigbees fault.

closed 2020-05-17 by blair287