zigpy/zha-device-handlers #353: [BUG] Ikea Blind super slow to react to command

blair287, 2020-05-16:
If the blind has not been operated for roughly 5 minutes the next command is really slow to execute if its told to go up or down or any other command from UI it can take upwards of 10 seconds before it starts moving.

Once it moves the UI is super responsive till its left for 5 minutes again then the cycle repeats.

This didn't used to happen the blind was very responsive.

Adding the blind to smartthings its quick to react less than a seconds before it moves.

Blind it connected through the included repeater which is less than 1m from the blind which is then about 3m from conbee2 stick. has 255 LQI and -43 for signal. No other zigbee devices are slow to operate all are instant.

Adminiuga, 2020-05-19:
enable debug logging as per https://www.home-assistant.io/integrations/zha/#debug-logging
once debug logging is enabled:
1. go to config panel -> ZHA -> pick the blind from the device drop down
2. Select WindowCovering cluster from the device cluster drop down
3. get value of `config_status (0x0007)` attribute, post value here
4. get value of `window_covering_mode (0x0017)` attribute, post value here
5. Get log
6. wait 6 min
7. go to entity and send it open and then close commands
8. get another log

post logs and values here

blair287, 2020-05-23:
> enable debug logging as per https://www.home-assistant.io/integrations/zha/#debug-logging
> once debug logging is enabled:
>
> 1. go to config panel -> ZHA -> pick the blind from the device drop down
> 2. [...]

I haven't seen your post sorry I will do this in the morning.

One thing I have noticed though is remove the blind from HA then re add it it's fast to respond fine even after several hours.

But set the max length of the blind and suddenly it's laggy as hell since the 110 update its now a 5 second plus delay on every command not just the first one after a while.

Makes controlling it to a specific length and stopping it through HA impossible.

Adminiuga, 2020-05-24:
> But set the max length of the blind

What do you mean? and how do you do that?

blair287, 2020-05-24:
> > But set the max length of the blind
> 
> What do you mean? and how do you do that?

manually lower the blind via the buttons on the unit then press a button again to stop it when its at the length you need, then you double press the bottom button and it will limit that to the max length then it should report the 0 to 100% to HA fine.

this is the info you requested:

3) bitmap8.2|1

4) 0

5)

`2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04ab0007000000
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04ab0013000c00226b029dbd0b010000000000
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 107: 00
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x6b 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xbd9d endpoint=11>, status: 0x00
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0eac000700aa00
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x17ac000800010001
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17ac0026001f002202000001029dbd0b040106000800186a01000000100000afffbf244b00cb
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xbd9d>, 11, 260, 6, b'\x18j\x01\x00\x00\x00\x10\x00', 0, 175, 255, 191, 36, 75, 0, -53]
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy.zcl] [0xbd9d:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=106 command_id=Command.Read_Attributes_rsp>
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xbd9d>, ep: 11, profile: 0x0104, cluster_id: 0x0006, data: b'186a010000001000'
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 108 under 109 request id, data: b'006c000000'
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 109, 0, <DeconzAddressEndpoint address_mode=2 address=0xbd9d endpoint=11>, 260, 8, 1, b'\x00l\x00\x00\x00', 2, 0)
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x12ad001b0014006d00029dbd0b04010800010500006c0000000200
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x12ad0009000200226d
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 109]
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0eae000700a600
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04ae0007000000
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04ae0013000c00226d029dbd0b010000000000
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 109: 00
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x6d 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xbd9d endpoint=11>, status: 0x00
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0eaf000700aa00
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x17af000800010001
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17af0026001f002202000001029dbd0b040108000800186c0100000020fe00afffbf244b00c9
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xbd9d>, 11, 260, 8, b'\x18l\x01\x00\x00\x00 \xfe', 0, 175, 255, 191, 36, 75, 0, -55]
2020-05-24 11:36:23 DEBUG (MainThread) [zigpy.zcl] [0xbd9d:11:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=108 command_id=Command.Read_Attributes_rsp>
[... cut ...] 24T11:44:36.355829+01:00>>
2020-05-24 11:44:36 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.prusa_mk3s_actual_tool0_temp, old_state=<state sensor.prusa_mk3s_actual_tool0_temp=214.9; unit_of_measurement=°C, friendly_name=Prusa MK3s actual tool0 temp @ 2020-05-24T11:44:05.425182+01:00>, new_state=<state sensor.prusa_mk3s_actual_tool0_temp=215.1; unit_of_measurement=°C, friendly_name=Prusa MK3s actual tool0 temp @ 2020-05-24T11:44:36.366196+01:00>>`

blair287, 2020-05-24:
This is a log after i re added the blind and set the length however its not reporting its position in HA at the moment the slider never moved but the blind is being super responsive currently.

I wonder if when it starts to send position data it will slow down.

`2020-05-24 11:54:55 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 169: 00
2020-05-24 11:54:55 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xa9 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x2072 endpoint=1>, status: 0x00
2020-05-24 11:54:56 DEBUG (MainThread) [homeassistant.components.zha.entity] light.kitchen_light_2: polling current state
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 170 under 171 request id, data: b'00aa000000'
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 171, 0, <DeconzAddressEndpoint address_mode=2 address=0x1771 endpoint=11>, 260, 6, 1, b'\x00\xaa\x00\x00\x00', 2, 0)
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x128e001b001400ab000271170b0401060001050000aa0000000200
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x128e000900020022ab
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 171]
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e8f000700a600
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x048f0007000000
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x048f0013000c0022ab0271170b010000000000
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 171: 00
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xab 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x1771 endpoint=11>, status: 0x00
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e90000700aa00
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1790000800010001
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17900026001f0022020000010271170b04010600080018aa01000000100000afff5d294b00cc
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x1771>, 11, 260, 6, b'\x18\xaa\x01\x00\x00\x00\x10\x00', 0, 175, 255, 93, 41, 75, 0, -52]
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy.zcl] [0x1771:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=170 command_id=Command.Read_Attributes_rsp>
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x1771>, ep: 11, profile: 0x0104, cluster_id: 0x0006, data: b'18aa010000001000'
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 172 under 173 request id, data: b'00ac000000'
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 173, 0, <DeconzAddressEndpoint address_mode=2 address=0x1771 endpoint=11>, 260, 8, 1, b'\x00\xac\x00\x00\x00', 2, 0)
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1291001b001400ad000271170b0401080001050000ac0000000200
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1291000900020022ad
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 173]
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e92000700a600
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04920007000000
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04920013000c0022ad0271170b010000000000
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 173: 00
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xad 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x1771 endpoint=11>, status: 0x00
2020-05-24 11:54:56 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e93000700aa00
[... cut ...]
2020-05-24 11:55:04 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04990013000c0022af02ba1800000000000000
2020-05-24 11:55:04 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 175: 00
2020-05-24 11:55:04 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xaf 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x18ba endpoint=0>, status: 0x00
2020-05-24 11:55:05 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1c9a000c00050002f317ffc5`

blair287, 2020-05-24:
If i press reconfigure device it start reporting position data but then everything goes super laggy or just plain up ignores commands.

blair287, 2020-05-24:
Im done was attempting to reset and add everything again added the repeater ok added the remote ok then added the blind then the remote and repeater disappear from ZHA, HA restart NG try again NG try again NG i give up with is pre alpha software if i ever saw it.

Adminiuga, 2020-05-24:
Repeater disappearing was a bug unrelated to zha, fixed in 0.110.2

blair287, 2020-05-25:
> Repeater disappearing was a bug unrelated to zha, fixed in 0.110.2

Ah ok I was pulling my hair out over this. This blind will be the death of me I got 48 ZigBee devices and only this blind had issues.

blair287, 2020-05-25:
> Repeater disappearing was a bug unrelated to zha, fixed in 0.110.2

Find anything in the info i posted need anymore info?

blair287, 2020-05-29:
Its got even worse after the first command it hangs for about 6-8 seconds then starts to move but now every command after that whilst it moving is ignored pressing stop or anything does nothing. Is this issue ever going to be solved at all?

blair287, 2020-05-29:
> Repeater disappearing was a bug unrelated to zha, fixed in 0.110.2

Its got even worse after the first command it hangs for about 6-8 seconds then starts to move but now every command after that whilst it moving is ignored pressing stop or anything does nothing. Is this issue ever going to be solved at all?

blair287, 2020-05-30:
asked for logs then silence......

blair287, 2020-06-27:
Starting to give up that this will ever be resolved.

blair287, 2020-07-09:
> Repeater disappearing was a bug unrelated to zha, fixed in 0.110.2

Anything wrong in logs this issue still going on gets worse and worse till I need to unpaid and repair blind again.

Adminiuga, 2020-07-09:
In the logs I did not see any requests to control the blinds. Just to requests refreshing the state of a 0x1171 light.
And please post the logs between triple back-ticks, so it is easier to read the logs on mobile.

closed 2020-09-04 by Adminiuga