zigpy/zha-device-handlers #355: [BUG] Ikea E1524, 5-button, Remote dont work, only battery status.

jampem, 2020-05-18:
And no zha events. SMaBiT (Bitron Video) ZigBee USB Wireless Stick, Raspberry 3B+

[image]

Adding the remote log:

[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0xf54a (00:0b:57:ff:fe:38:30:99) left the network
[0xf54a](TRADFRI remote control): started initialization
[0xf54a:ZDO](TRADFRI remote control): 'async_initialize' stage succeeded
[0xf54a] Extending timeout for 0x9d request
[0xf54a] Extending timeout for 0x9f request
[0xf54a:1:0x1000]: initializing channel: from_cache: False
[0xf54a:1:0x0008]: initializing channel: from_cache: False
[0xf54a:1:0x0006]: initializing channel: from_cache: False
[0xf54a:1:0x0019]: initializing channel: from_cache: False
[0xf54a:1:0x0005]: initializing channel: from_cache: False
Device 0xf54a (00:0b:57:ff:fe:38:30:99) joined the network
Skip initialization for existing device 00:0b:57:ff:fe:38:30:99
[0xf54a:zdo] ZDO request ZDOCmd.Device_annce: [0xf54a, 00:0b:57:ff:fe:38:30:99, 128]
[0xf54a:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=157 command_id=Command.Read_Attributes_rsp>
[0xf54a:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=159 command_id=Command.Read_Attributes_rsp>
[0xf54a:1:0x0000]: initializing channel: from_cache: False
[0xf54a:1:0x0001]: initializing channel: from_cache: False
[0xf54a] Extending timeout for 0xa1 request
[0xf54a:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=161 command_id=Command.Read_Attributes_rsp>
[0xf54a:1:0x0000]: 'async_initialize' stage succeeded
[0xf54a:1:0x0001]: 'async_initialize' stage succeeded
[0xf54a:1:0x1000]: 'async_initialize' stage succeeded
[0xf54a:1:0x0008]: 'async_initialize' stage succeeded
[0xf54a:1:0x0006]: 'async_initialize' stage succeeded
[0xf54a:1:0x0019]: 'async_initialize' stage succeeded
[0xf54a:1:0x0005]: 'async_initialize' stage succeeded
[0xf54a](TRADFRI remote control): power source: Battery or Unknown
[0xf54a](TRADFRI remote control): completed initialization

Adminiuga, 2020-05-19:
What version of core?

jampem, 2020-05-19:
System Health
arch	armv7l
dev	false
docker	true
hassio	true
os_name	Linux
os_version	4.19.114-v7
python_version	3.7.7
timezone	Europe/Helsinki
version	0.109.6
virtualenv	false
Lovelace

dashboards	1
mode	storage
resources	1
views	1

Adminiuga, 2020-05-19:
1. enable debug logging for zigpy/quirks/radio as per https://www.home-assistant.io/integrations/zha/#debug-logging

2. remove device 1st so you don't get `Skip initialization for existing device 00:0b:57:ff:fe:38:30:99` and wait at least one minute

3. Re-pair the device, wait till it finishes pairing, keep device awake while pairing. 
4. press some buttons on the device
5. post full debug logs

jampem, 2020-05-19:
Removed only some personal info:

[home-assistant_modified.log](https://github.com/zigpy/zha-device-handlers/files/4652236/home-assistant_modified.log)

Adminiuga, 2020-05-19:
```
Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4476 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
Discovered endpoints: [1]
Discovered endpoint information: <SimpleDescriptor endpoint=1 profile=260 device_type=2064 device_version=2 input_clusters=[0, 1, 3, 9, 2821, 4096] output_clusters=[3, 4, 5, 6, 8, 25, 4096]>
Manufacturer: IKEA of Sweden
Model: TRADFRI remote control
```

Doesn't seem to match quirks
```
...
2020-05-19 19:59:30 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
2020-05-19 19:59:30 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
2020-05-19 19:59:30 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event device_registry_updated[L]: action=create, device_id=61b9344f8ee44dba9ac5b96577d10475>
```

@jampem in Configuration panel -> Devices, what "Firmware" does it report in the device info?

Adminiuga, 2020-05-19:
do you have dev access to hassio console and able to run "docker" command?

Adminiuga, 2020-05-19:
if you have dev ssh access to hassio, run the following
```
docker exec -it homeassistant pip3 uninstall zha-quirks
docker exec -it homeassistant pip3 install git+https://github.com/Adminiuga/zha-device-handlers.git@ac/ikea-5btn-remote-2
```

and restart hass

jampem, 2020-05-19:
Did those docker exec things.
Now missing Configuration:Zigbee Integration so I can't add the remote, should I try to reinstall Zigbee Integration? Here is the log

[home-assistant_mod2.log](https://github.com/zigpy/zha-device-handlers/files/4653006/home-assistant_mod2.log)

Adminiuga, 2020-05-19:
ouch. Wanna upgrade to beta? There were some incompatible changes since last release.

Adminiuga, 2020-05-19:
otherwise rebuild the container with `ha core rebuild` in hassio CLI

jampem, 2020-05-19:
I can upgrade to beta (how I do that?) because I only started 3-4 weeks ago with Home Assistant and my setup is little bit experimental, planning to move away from Zipabox and my other site where I'am using Tellstick Net. So if the beta works with my another old Tellstick Classic USB stick, which I'm already using, that should be ok

Adminiuga, 2020-05-19:
Beta supports all same sticks. 
In HA go to super visor panel and join the beta channel. It then should allow you update to 0.110.b4 version

jampem, 2020-05-20:
Same same, only battery and no zha events, should I upgrade also the system to HassOS Release-4 build 6 (RC)? The log:

[home-assistant_mod3.log](https://github.com/zigpy/zha-device-handlers/files/4654944/home-assistant_mod3.log)

Adminiuga, 2020-05-20:
After you updated to beta/release, did you run those docker commands?

jampem, 2020-05-20:
Noup, but doing it now.

Adminiuga, 2020-05-20:
If successful, after restart you should see quirk info applied to the remote in zha panel. After, wake up remote, press reconfigure in zha device panel and keep remote awake. Post debug logs

jampem, 2020-05-20:
Ookkay, we see some light, I think all the buttons worked first but after I tried few more times left and right did not work anymore (zha event)  but atleast all buttons are visible on automations. I will boot few more times and see whats coming. 

[home-assistant_mod4.log](https://github.com/zigpy/zha-device-handlers/files/4657030/home-assistant_mod4.log)

Adminiuga, 2020-05-20:
This seems to be a slightly different remote and I don't know if its the model or firmware update. But new remote reports itself as a scenes controller, i wonder if it sends different commands for left and right buttons. 

Can you annotate the debug log for each button press. for example for button X - this is in the logs
```
2020-05-20 15:52:20 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=8 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=193>, 255, -62, 0xbba7, 255, 255, b'\x01\x13\x02\x01+\x05\x00']
2020-05-20 15:52:20 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=19 command_id=2>
2020-05-20 15:52:20 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] ZCL request 0x0002: [1, 43, 5]
2020-05-20 15:52:20 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] No handler for cluster command 2
2020-05-20 15:52:20 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=00:0b:57:ff:fe:38:30:99, unique_id=00:0b:57:ff:fe:38:30:99:1:0x0008, endpoint_id=1, cluster_id=8, command=step, args=[1, 43, 5]>
```

Not  all may generate an event, but at least the raw and decoded commands

jampem, 2020-05-20:
Ok I'll try with this does this sound logical: 1. series up-left-right-down. 2. series up-left-right-down-middle (up=big sun down=little sun) and the log:

2020-05-20 21:42:50 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401080001ff0001c09e61efbfa7bbffff07011606002b0500'
2020-05-20 21:42:50 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=8 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=97>, 239, -65, 0xbba7, 255, 255, b'\x01\x16\x06\x00+\x05\x00']
2020-05-20 21:42:50 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=22 command_id=6>
2020-05-20 21:42:50 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] ZCL request 0x0006: [0, 43, 5]
2020-05-20 21:42:50 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] No handler for cluster command 6
2020-05-20 21:42:50 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=00:0b:57:ff:fe:38:30:99, unique_id=00:0b:57:ff:fe:38:30:99:1:0x0008, endpoint_id=1, cluster_id=8, command=step_with_on_off, args=[0, 43, 5]>
2020-05-20 21:42:52 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401050001ff0001c09e62ffbfa7bbffff09057c11170701010d00'
2020-05-20 21:42:52 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=5 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=98>, 255, -65, 0xbba7, 255, 255, b'\x05|\x11\x17\x07\x01\x01\r\x00']
2020-05-20 21:42:52 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=23 command_id=7>
2020-05-20 21:42:52 WARNING (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] Unknown cluster-specific command 7
2020-05-20 21:42:52 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL request 0x0007: b'\x01\x01\r\x00'
2020-05-20 21:42:52 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] No handler for cluster command 7
2020-05-20 21:42:53 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'00040102040101400000008dffb7e1bcffff08085c0a0000299808'
2020-05-20 21:42:53 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=1026 sourceEndpoint=1 destinationEndpoint=1 options=64 groupId=0 sequence=141>, 255, -73, 0xbce1, 255, 255, b'\x08\\\n\x00\x00)\x98\x08']
2020-05-20 21:42:53 DEBUG (MainThread) [zigpy.zcl] [0xbce1:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=92 command_id=Command.Report_Attributes>
2020-05-20 21:42:53 DEBUG (MainThread) [zigpy.zcl] [0xbce1:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=2200>>]]
2020-05-20 21:42:53 DEBUG (MainThread) [zigpy.zcl] [0xbce1:1:0x0402] Attribute report received: measured_value=2200
2020-05-20 21:42:53 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xbce1, <EmberApsFrame profileId=260 clusterId=1026 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=92>, 79, b'\x18\\\x0b\n\x00')
2020-05-20 21:42:53 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'0065'
2020-05-20 21:42:53 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401050001ff0001c09e63ffbda7bbffff09057c11180700010d00'
2020-05-20 21:42:53 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=5 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=99>, 255, -67, 0xbba7, 255, 255, b'\x05|\x11\x18\x07\x00\x01\r\x00']
2020-05-20 21:42:53 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=24 command_id=7>
2020-05-20 21:42:53 WARNING (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] Unknown cluster-specific command 7
2020-05-20 21:42:53 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL request 0x0007: b'\x00\x01\r\x00'
2020-05-20 21:42:53 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] No handler for cluster command 7
2020-05-20 21:42:53 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'00e1bc04010204010140010000654f0000'
2020-05-20 21:42:53 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 48353, <EmberApsFrame profileId=260 clusterId=1026 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=101>, 79, <EmberStatus.SUCCESS: 0>, b'']
[... cut ...]
2020-05-20 21:42:56 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=00:0b:57:ff:fe:38:30:99, unique_id=00:0b:57:ff:fe:38:30:99:1:0x0006, endpoint_id=1, cluster_id=6, command=toggle, args=[]>
2020-05-20 21:42:59 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=system_log, service=write, service_data=logger=frontend.js.latest.202005190, message=:0:0 Script error.>
2020-05-20 21:42:59 ERROR (MainThread) [frontend.js.latest.202005190] :0:0 Script error.

jampem, 2020-05-20:
Next one: short up - long up - short left - long left - short right - long right - short down - -long down - short middle - long middle:

2020-05-20 22:32:54 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2020-05-20 22:32:54 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2020-05-20 22:32:57 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401080001ff0001c09e66ffbea7bbffff07011b06002b0500'
2020-05-20 22:32:57 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=8 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=102>, 255, -66, 0xbba7, 255, 255, b'\x01\x1b\x06\x00+\x05\x00']
2020-05-20 22:32:57 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=27 command_id=6>
2020-05-20 22:32:57 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] ZCL request 0x0006: [0, 43, 5]
2020-05-20 22:32:57 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] No handler for cluster command 6
2020-05-20 22:32:57 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=00:0b:57:ff:fe:38:30:99, unique_id=00:0b:57:ff:fe:38:30:99:1:0x0008, endpoint_id=1, cluster_id=8, command=step_with_on_off, args=[0, 43, 5]>
2020-05-20 22:33:00 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401080001ff0001c09e67ffbba7bbffff05011c050053'
2020-05-20 22:33:00 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=8 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=103>, 255, -69, 0xbba7, 255, 255, b'\x01\x1c\x05\x00S']
2020-05-20 22:33:00 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=28 command_id=5>
2020-05-20 22:33:00 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] ZCL request 0x0005: [0, 83]
2020-05-20 22:33:00 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] No handler for cluster command 5
2020-05-20 22:33:00 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=00:0b:57:ff:fe:38:30:99, unique_id=00:0b:57:ff:fe:38:30:99:1:0x0008, endpoint_id=1, cluster_id=8, command=move_with_on_off, args=[0, 83]>
2020-05-20 22:33:02 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401080001ff0001c09e68ffbaa7bbffff03011d07'
2020-05-20 22:33:02 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=8 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=104>, 255, -70, 0xbba7, 255, 255, b'\x01\x1d\x07']
2020-05-20 22:33:02 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=29 command_id=7>
2020-05-20 22:33:02 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] ZCL request 0x0007: []
2020-05-20 22:33:02 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0008] No handler for cluster command 7
2020-05-20 22:33:02 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event zha_event[L]: device_ieee=00:0b:57:ff:fe:38:30:99, unique_id=00:0b:57:ff:fe:38:30:99:1:0x0008, endpoint_id=1, cluster_id=8, command=stop, args=[]>
2020-05-20 22:33:04 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2020-05-20 22:33:04 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2020-05-20 22:33:14 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2020-05-20 22:33:14 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2020-05-20 22:33:24 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2020-05-20 22:33:26 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event timer_out_of_sync[L]: seconds=1.1266970239998955>
2020-05-20 22:33:26 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2020-05-20 22:33:28 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401050001ff0001c09e69ffbfa7bbffff09057c111e0701010d00'
2020-05-20 22:33:28 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=5 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=105>, 255, -65, 0xbba7, 255, 255, b'\x05|\x11\x1e\x07\x01\x01\r\x00']
2020-05-20 22:33:28 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=30 command_id=7>
2020-05-20 22:33:28 WARNING (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] Unknown cluster-specific command 7
[... cut ...]
2020-05-20 22:34:09 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.olohuone_temperature, old_state=<state sensor.olohuone_temperature=23.2; device_file=/sys/bus/w1/devices/28-0516803837ff/w1_slave, raw_value=23187.0, unit_of_measurement=°C, friendly_name=Olohuone Temperature @ 2020-05-20T22:01:33.889693+03:00>, new_state=<state sensor.olohuone_temperature=23.2; device_file=/sys/bus/w1/devices/28-0516803837ff/w1_slave, raw_value=23250.0, unit_of_measurement=°C, friendly_name=Olohuone Temperature @ 2020-05-20T22:01:33.889693+03:00>>

jampem, 2020-05-20:
and from events:

Event 14 fired 10:33 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0008",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move_to_level_with_on_off",
        "args": [
            254,
            0
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-20T19:33:53.966933+00:00",
    "context": {
        "id": "7377e62dd31f4f4e875a246440063e8d",
        "parent_id": null,
        "user_id": null
    }
}
Event 13 fired 10:33 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0006",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-20T19:33:50.458525+00:00",
    "context": {
        "id": "30937e2a47424e3387f64b87527e74c7",
        "parent_id": null,
        "user_id": null
    }
}
Event 12 fired 10:33 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0006",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "toggle",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-20T19:33:48.589253+00:00",
    "context": {
        "id": "1a56c8bfdc964c948ab6202cece0ed71",
        "parent_id": null,
        "user_id": null
    }
}
Event 11 fired 10:33 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0008",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "stop",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-20T19:33:46.951046+00:00",
    "context": {
        "id": "98f4063807744444abf1b85066c95a66",
        "parent_id": null,
        "user_id": null
    }
}
Event 10 fired 10:33 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0008",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move",
        "args": [
            1,
            83
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-20T19:33:44.330385+00:00",
    "context": {
        "id": "c8cb81f803ca4b3c9470cc6e53235447",
        "parent_id": null,
        "user_id": null
    }
}
Event 9 fired 10:33 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0008",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            1,
            43,
            5
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-20T19:33:42.728247+00:00",
    "context": {
        "id": "24272f6c349b4d818ff5f6a3f8576e6b",
        "parent_id": null,
        "user_id": null
    }
}
Event 8 fired 10:33 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0008",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "stop",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-20T19:33:02.698182+00:00",
    "context": {
        "id": "3f50841d2c6240a39a4b18c7bde9c3d7",
        "parent_id": null,
        "user_id": null
    }
}
Event 7 fired 10:33 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0008",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move_with_on_off",
        "args": [
            0,
            83
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-20T19:33:00.151926+00:00",
    "context": {
        "id": "5d5f588d4e5f4848973382518e89d868",
        "parent_id": null,
        "user_id": null
    }
}
Event 6 fired 10:32 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0008",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step_with_on_off",
        "args": [
            0,
            43,
            5
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-20T19:32:57.577598+00:00",
    "context": {
        "id": "487e5b1ba39d48e08f5d3025f306e644",
        "parent_id": null,
        "user_id": null
    }
}

Adminiuga, 2020-05-20:
Can you map event # to button press?

jampem, 2020-05-20:
Installed some realtime logging add-on, is this ok? here is two times pushed left button:

2020-05-21 01:51:00 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401050001ff0001c09e9affc3a7bbffff09057c114f0701010d00'
2020-05-21 01:51:00 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=5 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=154>, 255, -61, 0xbba7, 255, 255, b'\x05|\x11O\x07\x01\x01\r\x00']
2020-05-21 01:51:00 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=79 command_id=7>
2020-05-21 01:51:00 WARNING (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] Unknown cluster-specific command 7
2020-05-21 01:51:00 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL request 0x0007: b'\x01\x01\r\x00'
2020-05-21 01:51:00 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] No handler for cluster command 7

2020-05-21 01:52:39 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401050001ff0001c09e9bf6c0a7bbffff09057c11500701010d00'
2020-05-21 01:52:39 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=5 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=155>, 246, -64, 0xbba7, 255, 255, b'\x05|\x11P\x07\x01\x01\r\x00']
2020-05-21 01:52:39 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=80 command_id=7>
2020-05-21 01:52:39 WARNING (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] Unknown cluster-specific command 7
2020-05-21 01:52:39 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL request 0x0007: b'\x01\x01\r\x00'
2020-05-21 01:52:39 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] No handler for cluster command 7
Connected DDisconnectSEnable Autoscroll⌘KClear Console

jampem, 2020-05-20:
and 2 times right button:

2020-05-21 01:55:43 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401050001ff0001c09e9cffc5a7bbffff09057c11510700010d00'
2020-05-21 01:55:43 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=5 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=156>, 255, -59, 0xbba7, 255, 255, b'\x05|\x11Q\x07\x00\x01\r\x00']
2020-05-21 01:55:43 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=81 command_id=7>
2020-05-21 01:55:43 WARNING (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] Unknown cluster-specific command 7
2020-05-21 01:55:43 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL request 0x0007: b'\x00\x01\r\x00'
2020-05-21 01:55:43 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] No handler for cluster command 7
2020-05-21 01:55:44 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2020-05-21 01:55:44 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2020-05-21 01:55:46 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401050001ff0001c09e9dffc3a7bbffff09057c11520700010d00'
2020-05-21 01:55:46 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=5 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=157>, 255, -61, 0xbba7, 255, 255, b'\x05|\x11R\x07\x00\x01\r\x00']
2020-05-21 01:55:46 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=82 command_id=7>
2020-05-21 01:55:46 WARNING (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] Unknown cluster-specific command 7
2020-05-21 01:55:46 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL request 0x0007: b'\x00\x01\r\x00'
2020-05-21 01:55:46 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] No handler for cluster command 7
Connected DDisconnectSEnable Autoscroll⌘KClear Console

jampem, 2020-05-20:
1 long left:

2020-05-21 02:05:05 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2020-05-21 02:05:05 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2020-05-21 02:05:15 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2020-05-21 02:05:15 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2020-05-21 02:05:16 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401050001ff0001c09ec6ffc0a7bbffff08057c117b08010d00'
2020-05-21 02:05:16 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=5 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=198>, 255, -64, 0xbba7, 255, 255, b'\x05|\x11{\x08\x01\r\x00']
2020-05-21 02:05:16 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=123 command_id=8>
2020-05-21 02:05:16 WARNING (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] Unknown cluster-specific command 8
2020-05-21 02:05:16 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL request 0x0008: b'\x01\r\x00'
2020-05-21 02:05:16 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] No handler for cluster command 8
2020-05-21 02:05:17 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401050001ff0001c09ec7ffc0a7bbffff07057c117c094305'
2020-05-21 02:05:17 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=5 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=199>, 255, -64, 0xbba7, 255, 255, b'\x05|\x11|\tC\x05']
2020-05-21 02:05:17 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=124 command_id=9>
2020-05-21 02:05:17 WARNING (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] Unknown cluster-specific command 9
2020-05-21 02:05:17 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL request 0x0009: b'C\x05'
2020-05-21 02:05:17 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] No handler for cluster command 9
Connected DDisconnectSEnable Autoscroll⌘KClear Console

jampem, 2020-05-20:
1 long right

2020-05-21 02:06:16 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2020-05-21 02:06:16 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2020-05-21 02:06:16 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401050001ff0001c09eccffb2a7bbffff08057c110108000d00'
2020-05-21 02:06:16 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=5 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=204>, 255, -78, 0xbba7, 255, 255, b'\x05|\x11\x01\x08\x00\r\x00']
2020-05-21 02:06:16 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=1 command_id=8>
2020-05-21 02:06:16 WARNING (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] Unknown cluster-specific command 8
2020-05-21 02:06:16 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL request 0x0008: b'\x00\r\x00'
2020-05-21 02:06:16 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] No handler for cluster command 8
2020-05-21 02:06:17 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'020401050001ff0001c09ecdffc4a7bbffff07057c1102093d05'
2020-05-21 02:06:17 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_MULTICAST: 2>, <EmberApsFrame profileId=260 clusterId=5 sourceEndpoint=1 destinationEndpoint=255 options=256 groupId=40640 sequence=205>, 255, -60, 0xbba7, 255, 255, b'\x05|\x11\x02\t=\x05']
2020-05-21 02:06:17 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=True is_reply=False disable_default_response=False> manufacturer=4476 tsn=2 command_id=9>
2020-05-21 02:06:17 WARNING (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] Unknown cluster-specific command 9
2020-05-21 02:06:17 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] ZCL request 0x0009: b'=\x05'
2020-05-21 02:06:17 DEBUG (MainThread) [zigpy.zcl] [0xbba7:1:0x0005] No handler for cluster command 9
Connected DDisconnectSEnable Autoscroll⌘KClear Console

Adminiuga, 2020-05-21:
pushed some updates. reinstall
```
docker exec -it homeassistant pip3 uninstall zha-quirks
docker exec -it homeassistant pip3 install git+https://github.com/Adminiuga/zha-device-handlers.git@ac/ikea-5btn-remote-2
```

and give it a go

chmielowiec, 2020-05-21:
Maybe you have an idea about my problem https://github.com/zigpy/zha-device-handlers/pull/318? My remote is sending `[1, 84]` args instead of `[1, 83]` for long pressing left/right (e1524, newest firmware)

jampem, 2020-05-21:
Yes, now it seems that everything is working, I will continue testing but for now everything looks excellent, thanks. 

Tried to remove Trådfri before testing, to add it afterwards for a test, but deleting it from Zigbee Integration did not delete it? After boot it was still there but now its working.

jampem, 2020-05-21:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0008",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step_with_on_off",
        "args": [
            0,
            43,
            5
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-21T12:46:18.729660+00:00",
    "context": {
        "id": "cb7234ee8c8a4862bb80f51caccb2912",
        "parent_id": null,
        "user_id": null
    }
}
Event 40 fired 3:46 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0008",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "move",
        "args": [
            1,
            83
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-21T12:46:16.428643+00:00",
    "context": {
        "id": "f4733464a4a34af1b60de60de364a49a",
        "parent_id": null,
        "user_id": null
    }
}
Event 39 fired 3:46 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0008",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step",
        "args": [
            1,
            43,
            5
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-21T12:46:15.098224+00:00",
    "context": {
        "id": "cccfb402cbee4708a962b13306810d86",
        "parent_id": null,
        "user_id": null
    }
}
Event 38 fired 3:46 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0005",
        "endpoint_id": 1,
        "cluster_id": 5,
        "command": "release",
        "args": [
            583
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-21T12:46:13.666095+00:00",
    "context": {
        "id": "59d613218c5f4bce8d208f6fc9c0f8ac",
        "parent_id": null,
        "user_id": null
    }
}
Event 37 fired 3:46 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0005",
        "endpoint_id": 1,
        "cluster_id": 5,
        "command": "hold",
        "args": [
            3328,
            0
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-21T12:46:13.094896+00:00",
    "context": {
        "id": "0417659cc1f843afbc4da09873b69bf6",
        "parent_id": null,
        "user_id": null
    }
}
Event 36 fired 3:46 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0005",
        "endpoint_id": 1,
        "cluster_id": 5,
        "command": "press",
        "args": [
            256,
            13,
            0
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-21T12:46:12.448053+00:00",
    "context": {
        "id": "59c65e414024429ea3addf3f8d0d42a6",
        "parent_id": null,
        "user_id": null
    }
}
Event 35 fired 3:46 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0005",
        "endpoint_id": 1,
        "cluster_id": 5,
        "command": "release",
        "args": [
            706
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-21T12:46:10.439265+00:00",
    "context": {
        "id": "fd2b09bf11414ef1bf791e7660910f73",
        "parent_id": null,
        "user_id": null
    }
}
Event 34 fired 3:46 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0005",
        "endpoint_id": 1,
        "cluster_id": 5,
        "command": "hold",
        "args": [
            3329,
            0
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-21T12:46:09.735356+00:00",
    "context": {
        "id": "b37d32c9207140c4be3778915f899fca",
        "parent_id": null,
        "user_id": null
    }
}
Event 33 fired 3:46 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0005",
        "endpoint_id": 1,
        "cluster_id": 5,
        "command": "press",
        "args": [
            257,
            13,
            0
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-21T12:46:08.348247+00:00",
    "context": {
        "id": "90fd04ccee864afbaa39f5ef73920732",
        "parent_id": null,
        "user_id": null
    }
}
Event 32 fired 3:46 PM:
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "00:0b:57:ff:fe:38:30:99",
        "unique_id": "00:0b:57:ff:fe:38:30:99:1:0x0008",
        "endpoint_id": 1,
        "cluster_id": 8,
        "command": "step_with_on_off",
        "args": [
            0,
            43,
            5
        ]
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-21T12:46:06.317054+00:00",
    "context": {
        "id": "00bdd0984d9641fda1c28e7e9eddd30b",
        "parent_id": null,
        "user_id": null
    }
}

closed 2020-05-21 by Adminiuga

Adminiuga, 2020-05-21:
@chmielowiec is it the same signature as existing  one? I have E1524 with 2.0.314 and it doesn't send anything, because it sends broadcasts. 

If it has the same signature as the existing ones, but sends different args, then we need a solution to handle both cases. 
But  probably easiest to just to modify automation to match raw zha_event and not use device automation.

Adminiuga, 2020-05-21:
@jampem can you please read the `software_build_id` of the `basic` cluster of your remote and post it here?
eg
[image]

jampem, 2020-05-21:
[image]

Adminiuga, 2020-05-21:
can you read zcl version too please?

jampem, 2020-05-21:
[image]