zigpy/zha-device-handlers #356: [Device Support Request] Eurotronic Spirit Zigbee - Battery-operated Thermostat Radiator Valve (TRV)

Hedda, 2020-05-18:
Request ZHA quirk for Eurotronic Spirit Zigbee which is a popular TRV (Thermostat Radiator Valve)

https://eurotronic.org/produkte/zigbee-heizkoerperthermostat/spirit-zigbee/

Looks like this official PDF documentation might contain all or most Zigbee attributes?

https://eurotronic.org/wp-content/uploads/2019/11/Spirit_ZigBee_BAL_web_EN_November-2019.pdf

Dependent on ZHA climate support making it into to HA https://github.com/home-assistant/core/pull/35682

There is a quirk discussion about it for deCONZ here which their gateway now have supported for:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1098

FYI, Eurotronic also makes a Z-Wave version of this TRV (Thermostat Radiator Valve):

https://eurotronic.org/produkte/z-wave-heizkoerperthermostat/spirit-z-wave-plus/

https://eurotronic.org/wp-content/uploads/2018/08/Spirit_Z-Wave_BAL_web_DE_view_V5.pdf

That same Z-Wave version is also being sold as OEM and rebranded to Aeotec Radiator Thermostat

https://aeotec.com/z-wave-home-automation/thermostat-radiator-valve-trv/

PS: I already have the Aeotec Z-Wave version of this TVR myself and that works very well for me.

Hedda, 2020-05-18:
FYI, Adminiuga has now submitted a pull request for "ZHA Climate" (beta) with thermostats support.

It will probably not make it into Home Assistant 0.110 but it might hopefully make it into 0.111 +

Dependent on ZHA climate support making it into to HA https://github.com/home-assistant/core/pull/35682 by @Adminiuga

docgalaxyblock, 2020-05-28:
@Hedda I purchased one of these without checking if they are supported... ^^
I logged the pairing process here (HA 110):
https://pastebin.com/PCbqqP1M

Maybee it is usefull or not

Greetings from Germany ^^

Adminiuga, 2020-05-28:
you should try the dev build of HA

docgalaxyblock, 2020-05-28:
ok will setup a VM for testing

docgalaxyblock, 2020-05-29:
I try it at a dev VM and it works fine.
I think a quirk needs to be created.

docgalaxyblock, 2020-06-05:
Ok, I updated my HA to the 111beta and the temperature control, reportaing and battery level work out of the box. The device could not be turned off (you can select OFF but the TRV will stay at his temperature) So the device needs some investigation. 
@dmulcahey Do you hove some time to investigate and could made a quirk please
If your time allows this it would be famous if it will be at the next beta
Greetings ^^

vigonotion, 2020-06-09:
I worked on that one already: https://github.com/vigonotion/zha-device-handlers/tree/vi/spzb?files=1

It's based on an older dev version of HA 109 or so, I'm gonna update it in the following days

Hedda, 2020-06-09:
FYI, for reference, "ZHA Climate" (beta) made it into Home Assistant 0.111 Beta release candidate:

- https://rc.home-assistant.io/blog/2020/06/01/release-111/
  - https://rc.home-assistant.io/integrations/zha/

nerdosity, 2020-06-11:
> @nerdosity Do you have a quirk? And does the "off" work?
> I do not have one... (HA 0.111)

[image]

There is no quirk, I think the device support is standard at the moment. Maybe they will add a quirk when more attributes will be in the device, I don't really know how quirks works.

Turning it off does not work @Adminiuga , I think that in zigbee2mqtt it was mapped to window mode = ON.

docgalaxyblock, 2020-06-11:
@nerdosity 
> I can confirm them working.

This post was a bit unclear ^^

@Adminiuga At my opinion this issue should stay open, only the basic temperature controll is working ATM.
If connected to a Z2M Setup:
[image]
I like the three settings at Z2M to choose from much more:

- Off - isn't working with ZHA yet

- Automatic - standard heating (if using ZHA it's labeld "heating")

- Heating - "boost mode" from the TRV

When selecting "off" at Z2M it seems for me that also the temperture will be set to 5°C and at "heating" to 30°C (I'm not sure if this is correct)

A implementation to controll the childprotection would be nice
(Something I like to test is if it is possible to bin the TRV directly to an window sensor. I ordered an aquara sensor for this test)

nerdosity, 2020-06-11:
> @nerdosity
> 
> > I can confirm them working.
> 
> This post was a bit unclear ^^
> 
Sorry, I confirmed them working just after joining, having a battery+thermostat on my HA, and seeing that temperature changing was stable (i.e. changing from buttons -> change value in HA, and changing values in HA -> change value on the LCD). But I did not test the off, my bad.

Hedda, 2020-06-12:
@vigonotion Thanks for submitting a quirk to add/fix "OFF" and "Bidirectional temperature" functions!

https://github.com/zigpy/zha-device-handlers/pull/373

docgalaxyblock, 2020-06-13:
This is not at the 0.112.0.dev20200613 right?
Because there it is not working...mine spirit did not get a quirk there

dmulcahey, 2020-06-13:
I’ll get a release pushed to dev at some point today

docgalaxyblock, 2020-06-13:
Thx @dmulcahey Could you ping me when it is pushed please

dmulcahey, 2020-06-15:
@docgalaxyblock it has been published to dev in HA

docgalaxyblock, 2020-06-15:
[image]
The "off" works fine thx @vigonotion (in my opinion even better as Z2M )
The "boost mode" is missing (I know it was not at the PR)
Would be nice if the actual "heating" will be renamed to "automatic" (like Z2M has it) 
> [image]

or adding a new option named "boost"

vigonotion, 2020-06-15:
"Auto" would be the wrong mode for that, Z2M is wrong there. See the climate documentation: https://developers.home-assistant.io/docs/core/entity/climate#hvac-modes
HEAT = Heat to a specified target temperature
AUTO = some kind of schedule, AI...

There's the option to add presets (BOOST preset), I'll look into that

Adminiuga, 2020-06-15:
Yeah, the closest thing in HA architecture would be a preset mode. How do you enable boost mode?

docgalaxyblock, 2020-06-15:
OK
@Adminiuga By pressing the middle button (manual page 5 😉)

vigonotion, 2020-06-15:
There is a manufacturer specific cluster, and using bit flags the boost mode can be enabled (like I did with the OFF mode, same cluster and attribute, and the boost mode const is already in the quirk)

https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/eurotronic/__init__.py#L132-L146

Adminiuga, 2020-06-15:
Ok, how do you enable the boost mode via Zigbee? What value to what attribute do you write?
Otherwise, per manual the boost mode is there. Just press the button :)

docgalaxyblock, 2020-06-15:
https://www.zigbee2mqtt.io/devices/SPZB0001.html

docgalaxyblock, 2020-06-15:
I found a litte bug

- Setting the TRV to OFF

- Restarting HA

- And at HA the TRVs state reporting HEATING 5°C

Hedda, 2020-06-22:
@docgalaxyblock suggest post issue to HA core as probably not related to zha-device-handlers quirk

- https://github.com/home-assistant/core/issues

Perhaps ask if @vigonotion can replicate since he posted patches and has the device too himself.

vigonotion, 2020-06-22:
The problem is that the host flags attribute (where the OFF state is stored) is not reportable ("not sending information back to HA"). We'd need to poll the host flags once on HA boot (and maybe regularly because you could disable the OFF mode on the thermostat) AND implement a listener in the quirk on read attributes to update the system_mode correctly. I'm unsure how to do the polling.

I'll take a look at this after I implemented the ON mode, but I'm having issues with the newest HA dev and zha so it's really hard to test at the moment

Adminiuga, 2020-06-22:
Have you tried configuring attribute reporting for host flags attribute, by modifying the HVAC channel source file?

stefanroelofs, 2020-07-02:
I'm still having trouble communicating with this device. 

These are the states of the eurotronic thermostat:
```
hvac_modes:
  - 'off'
min_temp: 7
max_temp: 35
current_temperature: 19.5
temperature: 15
hvac_action: idle
system_mode: '[4]/heat'
occupied_heating_setpoint: 1500
pi_heating_demand: 0
unoccupied_cooling_setpoint: 1600
friendly_name: bathroom_thermostatic_radiator_valve thermostat
supported_features: 1
```
So I can only choose 'Off' as mode, not 'Heat'.
Receiving info seems ok, when I change set temperature on the device, I see this updated in HA.
Setting temperature, or mode, from HA does not work.

Here are all log messages that reference my device (0xb4ab):
```
2020-07-02 21:31:52 DEBUG (MainThread) [zigpy.device] [0xb4ab] Delivery error for seq # 0x01, on endpoint id 1 cluster 0x0201: message send failure
2020-07-02 21:31:52 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xb4ab:1:0x0201]: async_initialize: retryable request #1 failed: [0xb4ab:1:0x0201]: Message send failure. Retrying in 0.9s
2020-07-02 21:31:53 DEBUG (MainThread) [zigpy.device] [0xb4ab] Extending timeout for 0x03 request
2020-07-02 21:31:53 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Error while sending 4 req id frame: TXStatus.MAC_NO_ACK
2020-07-02 21:31:53 DEBUG (MainThread) [zigpy.device] [0xb4ab] Delivery error for seq # 0x03, on endpoint id 1 cluster 0x0201: message send failure
2020-07-02 21:31:53 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xb4ab:1:0x0201]: async_initialize: retryable request #2 failed: [0xb4ab:1:0x0201]: Message send failure. Retrying in 0.8s
2020-07-02 21:31:53 DEBUG (MainThread) [zigpy.device] [0xb4ab] Extending timeout for 0x05 request
2020-07-02 21:31:53 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 5 under 6 request id, data: b'0005000300040005000600'
2020-07-02 21:31:53 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (26, 6, 0, <DeconzAddressEndpoint address_mode=2 address=0xb4ab endpoint=1>, 260, 513, 1, b'\x00\x05\x00\x03\x00\x04\x00\x05\x00\x06\x00', 2, 0)
2020-07-02 21:32:04 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x06 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xb4ab endpoint=1>, status: 0xd0
2020-07-02 21:32:04 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Error while sending 6 req id frame: TXStatus.NWK_ROUTE_DISCOVERY_FAILED
2020-07-02 21:32:04 DEBUG (MainThread) [zigpy.device] [0xb4ab] Delivery error for seq # 0x05, on endpoint id 1 cluster 0x0201: message send failure
2020-07-02 21:32:04 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xb4ab:1:0x0201]: async_initialize: retryable request #3 failed: [0xb4ab:1:0x0201]: Message send failure. Retrying in 3.1s
2020-07-02 21:32:07 DEBUG (MainThread) [zigpy.device] [0xb4ab] Extending timeout for 0x07 request
2020-07-02 21:32:07 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 7 under 8 request id, data: b'0007000300040005000600'
2020-07-02 21:32:07 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (26, 8, 0, <DeconzAddressEndpoint address_mode=2 address=0xb4ab endpoint=1>, 260, 513, 1, b'\x00\x07\x00\x03\x00\x04\x00\x05\x00\x06\x00', 2, 0)
2020-07-02 21:32:17 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x08 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xb4ab endpoint=1>, status: 0xd0
2020-07-02 21:32:17 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Error while sending 8 req id frame: TXStatus.NWK_ROUTE_DISCOVERY_FAILED
2020-07-02 21:32:17 DEBUG (MainThread) [zigpy.device] [0xb4ab] Delivery error for seq # 0x07, on endpoint id 1 cluster 0x0201: message send failure
2020-07-02 21:32:17 WARNING (MainThread) [homeassistant.components.zha.core.channels.base] [0xb4ab:1:0x0201]: async_initialize: all attempts have failed: [DeliveryError('[0xb4ab:1:0x0201]: Message send failure'), DeliveryError('[0xb4ab:1:0x0201]: Message send failure'), DeliveryError('[0xb4ab:1:0x0201]: Message send failure'), DeliveryError('[0xb4ab:1:0x0201]: Message send failure')]
2020-07-02 21:32:17 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xb4ab:1:0x0201]: 'async_initialize' stage succeeded
2020-07-02 21:32:17 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xb4ab:1:0x0000]: 'async_initialize' stage succeeded
2020-07-02 21:32:17 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xb4ab:1:0x0001]: 'async_initialize' stage succeeded
2020-07-02 21:32:17 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xb4ab:1:0x0019]: 'async_initialize' stage succeeded
2020-07-02 21:32:17 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xb4ab](SPZB0001): power source: Battery or Unknown
2020-07-02 21:32:17 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xb4ab](SPZB0001): completed initialization
2020-07-02 21:34:34 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xb4ab>, 1, 260, 513, b'\x1c7\x10\xc3\n\x03@)\xaa\x05', 0, 175, 255, 154, 21, 50, 0, -57]
[... cut ...]
2020-07-02 21:39:49 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.bathroom_thermostatic_radiator_valve_thermostat: Attribute 'pi_heating_demand' = 0 update
2020-07-02 21:39:49 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xb4ab>, ep: 1, profile: 0x0104, cluster_id: 0x0201, data: b'18bd0a08002000'

```
I don't quite know how to troubleshoot this any further. The device worked ok under zigbee2mqtt.
Thanks!

vigonotion, 2020-07-02:
Did you restart Home Assistant after adding the thermostat? The manufacturer specific attribute containing the system mode information is not correctly sent on initialization i think

stefanroelofs, 2020-07-02:
Yes, correct. I added device to HA. It worked ok at first. I could set states Heat, and Off. Then I restarted HA, and it stopped working.

stefanroelofs, 2020-07-02:
I might have restarted HA while the device was in 'Off' mode actually, I don't remember for sure.

Thanks for your response anyway.

nerdosity, 2020-07-11:
pi_heating_demand is working now. Now I can monitor my valve status:

[image]

I'm using this state template for turning on the burner when valves are open enough (or, when the delta between valve set temperature and the ambient temperature is large enough):

`{{ (state_attr('sensor.climate.eurotronic_spzb0001_a3d6ff01_thermostat', 'pi_heating_demand')|int+state_attr('sensor.climate.eurotronic_spzb0001_a975c403_thermostat', 'pi_heating_demand')|int+state_attr('sensor.climate.eurotronic_spzb0001_bad6ff01_thermostat', 'pi_heating_demand')|int+state_attr('sensor.climate.eurotronic_spzb0001_6d75c403_thermostat', 'pi_heating_demand')|int )+state_attr('climate.calorifero_bagnetto', 'valve')|int > 99 }}`

This worked last year with zigbee2mqtt, should work next winter I hope ;)

Lovely-Maisonette, 2020-08-04:
Is there a regression?
Version: Home Assistant 0.113.3
Docker: homeassistant/home-assistant:stable

Main problem: Battery always 0%
Secondary: HVAC modes get sometimes lost or something, and then it cannot be turned on (or off)

[image]
[image]
[image]
[image]
[image]
There now HVAC has only 1 option (off) This happens intermittently, and it is unrelated to the battery problem, which is always 0% no matter what.
[image]

nerdosity, 2020-10-12:
Hi, actually pi_heating_demand is 0-255 (I saw it jumping to 150 and I read in the PDF from Eurotronic).
It should report percentage, can you please convert it mapping from 0-255 to 0-100%? 
It makes much more sense to use it without more templates.

Hedda, 2020-10-13:
@nerdosity this this issue is closed I think that you need to open a new issue for any new problems discovered like that.

sygys, 2022-01-11:
the problem i have with this valve is that it doesnt hold its valve position in TRV_mode=1 

It seems to pop back to mode 2 on its own. Do more people have this problem?