zigpy/zha-device-handlers #357: Compatibility list of Zigbee Thermostats and HVACs units compatible with ZHA?

Hedda, 2020-05-19:
Compatibility list of Zigbee Thermostats and HVACs units compatible with ZHA?

@Adminiuga has just submitted a first pull request for "ZHA Climate" (beta) with thermostats support were he mentions "tested with Sinope and Zen Zigbee thermostats". 

- https://github.com/home-assistant/core/pull/35682

As I do not yet own any Zigbee Thermostat and HVAC units I was wondering what other Thermostat and HVAC units are based on Zigbee devices and are or could be made compatible with ZHA climate feature?

Some are full HVAC controller (temperature, fan, heating/cooling system control) while others are only temperature (heating with no cooling), some are TVR (Thermostat Radiator Valve), etc.

Few other projects, like ex. Zigbee2mqtt and deCONZ, has list of "supported devices" which kind-of also acts as list of recommended devices in regards with what features should at least work with ZHA.

* https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/information/supported_devices.md
* https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices

Quick google search found these devices, some of which are central thermostats / HVAC control units and others are thermostats with different feature set. Zigbee2mqtt pages does offer feature-sets.

Centralite Pearl Thermostat (3157100) - HVAC, fan, and A/C Controller
- https://centralite.com/products/pearl-thermostat
  - https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/3157100.md

Zen Thermostat (Zen-01 / Zen-01-W) - Zigbee Edition - HVAC controller (temperature, heating/cooling system control)
- https://store.zenthermostat.com/products/zen-thermostat

Sinope TH1123ZB - Smart thermostat for electric heating 3000 W / 4000 W – Zigbee
- https://www.sinopetech.com/us/product/smart-thermostat-for-electric-heating-3000-w-zigbee/

Sinope TH1124ZB
- https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/TH1124ZB.md

Sinope TH1300ZB - Smart floor heating thermostat 3600 W – Zigbee
- https://www.sinopetech.com/us/product/smart-floor-heating-thermostat-3600-w-zigbee/

Sinope TH1400ZB - Low voltage smart thermostat 24 Vac – Zigbee
- https://www.sinopetech.com/us/product/low-voltage-smart-thermostat-24-vac-zigbee/

Sinope TH1500ZB
- https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/TH1500ZB.md

Stelpro Ki Convector (ST218) line-voltage thermostat

Stelpro Ki (STZB402) line-voltage thermostat
- https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/STZB402.md

Stelpro Maestro (SMT402 & MT402AD) line-voltage  line-voltage thermostats
- https://www.stelpro.com/en-CA/maestro-smart-thermostat
  - https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/SMT402.md
  - https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/SMT402AD.md

Schneider Electric Wiser Thermostat Radiator Valve (WV704R0A0902) - Temperature, battery, keypad lock, heating demand
- https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/WV704R0A0902.md

Eurotronic Spirit Zigbee - Thermostat Radiator Valve (TRV)
 - https://eurotronic.org/produkte/zigbee-heizkoerperthermostat/spirit-zigbee/
  - https://eurotronic.org/wp-content/uploads/2019/11/Spirit_ZigBee_BAL_web_EN_November-2019.pdf
    - https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/SPZB0001.md

Siterwell GS361A-H04 - TRV (Thermostat Radiator Valve) - Same as Viessmann ZK03840 & Essentials Smart 120112
- https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/GS361A-H04.md

Viessmann ViCare (ZK03840) - TRV (Thermostat Radiator Valve) - Same as Siterwell GS361A-H04
-https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/ZK03840.md

Essentials Smart (120112) - TRV (Thermostat Radiator Valve) - Same as Siterwell GS361A-H04
- https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/GS361A-H04.md

Leviton Wireless Omnistat2 (RC-2000WH / RC-2000WHZB) Multistage & Heat Pump Thermostat
- https://www.leviton.com/en/products/RC-2000WHZB
  - https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/RC-2000WH.md

eCozy Smart Thermostat (1TST-EU) - TVR (Thermostat Radiator Valve)
- http://www.ecozy.de/en/support/thermostat

Moes Radiator valve (HY369RT) - TVR (Thermostat Radiator Valve)
- https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/HY369RT.md

Bitron Video Wireless thermostat AV2010/32 902010/32
- https://bvsite.smabit.eu/index.php/produkte/smart-home-produkte/funkthermostat/
  - https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/AV2010_32.md

Hive Heating thermostat (SLR1b)  (thermostat, occupied heating, weekly schedule)
- https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/SLR1b.md

Hive Heating thermostat remote control (WPT1) (none, communicate via Hive Heating thermostat SLR1b)
- https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/WPT1.md

saz, 2020-07-24:
@Hedda https://zigbee.blakadder.com/zha.html seems like a good source.

pipiche38, 2021-05-07:
> @Hedda https://zigbee.blakadder.com/zha.html seems like a good source.

I think that is the best an easy one. 
I would however suggest some enhancement like firmware update available URL.

I also feel that there is a possible miss-match between the Zigbee key and the application underneath which is some time making the real support. If for instance I take HA, sometime it works out of the box, but some other time you need a particular quirk. Having such information would avoid confusion and make it more usual for everybody

MattWestb, 2021-05-07:
I have trying getting all tuya TRV in ZHA in one matrix but its not easy getting feedback from users for verifying functions and also one photo of the real TRV for verifying the model to the ID but its better now as ti was for 6 mouths ago.

I think all good info is great putting in Wiki so user and devs can finding it then needing it.

closed 2021-11-10 by github-actions[bot]

Creepios, 2023-04-06:
Hey there, I want to use the "Viessmann ViCare (ZK03840) - TRV (Thermostat Radiator Valve)" mentioned above, but it doesn't seem working currently. I tried paring it multiple times, but it just not work.
Does anyone have a solution for this?