zigpy/zha-device-handlers #362: Xiaomi Mijia Smart Switch - Long Press/ Hold

Kanishkaz, 2020-05-26:
I am using a Xiaomi Mijia Smart Switch (Round) with a Conbee II. I recently moved from deconz to ZHA and noticed that these switches do not suppot the long press/ hold click type in ZHA. It appears Zigbee2Mqtt support this functionality too.

ZHA should support the additional click type Long Press/ Hold, similar to Deconz.

Adminiuga, 2020-05-26:
AFAIK button does not send long presses and those features were implemented in software.

Kanishkaz, 2020-05-26:
OK. I had this button working correctly first with the Xiaomi Gateway and then Deconz. Both allowed me to use the long press click type in Home Assistant. From what I can see, Zigbee2Mqtt also support this functionality (Xiaomi MiJia wireless switch (single, double, triple, quadruple, many, long, long_release click))

Hedda, 2020-05-28:
> AFAIK button does not send long presses and those features were implemented in software.

FYI, ptvo Zigbee Configurable Firmware for TI CC2530 and CC2531 should feature long press support

- https://ptvo.info/zigbee-configurable-firmware-features/
  - https://ptvo.info/zigbee-switch-configurable-firmware-v2-210/
    - https://ptvo.info/zigbee-switch-configurable-firmware-router-199/

If you happen to have a Sonoff BasicZBR3 then here are settings for flashing it with sensor support

- https://ptvo.info/sonoff-basic-zbr3-with-the-configurable-firmware-283/

Hedda, 2020-06-02:
Sonoff SNZB-01 is a new Zigbee wireless switch from ITead which claim to feature "long press" support

- https://www.itead.cc/sonoff-snzb-01-zigbee-wireless-switch.html

"button switch is customizable with three control options - single, double press and long press"

dmulcahey, 2020-06-29:
> After a bit of investigating myself and getting upto speed, I understood what you meant. I too cannot see a separate event for the long press. Zigbee2Mqtt appear to get around this by supporting checking the time between the on and off events of a single click and report it as a long press if the gap is more than 1 second.
> Do zha-device-handlers support implementing timed delays?

Not at the moment. If we get good utility classes for this I wouldn't mind adding support. I just don't want 50 different devices doing it 50 different ways

Kanishkaz, 2020-07-05:
Fair enough. Would be good to get this functionality at some point.

closed 2021-08-13 by github-actions[bot]