zigpy/zha-device-handlers #364: [Device Support Request] EigenStone Thermostat S2 (KTWKQ03ES)

slydiman, 2020-05-30:
Could you add support for Aqara / EigenStone Thermostat S2 (KTWKQ03ES), please?
Id is lumi.airrtc.tcpecn02
http://docs.opencloud.aqara.com/en/guideline/product-discription/#thermostat

Adminiuga, 2020-05-30:
where did you get it?

slydiman, 2020-05-30:
https://www.aliexpress.com/item/32992549710.html

slydiman, 2020-05-30:
Note also 
https://homekitnews.com/2020/05/27/aqara-hub-firmware-update-adds-homekit-functionality-for-more-devices/

slydiman, 2020-05-30:
According to Chinese manual it can be connected to a remote temperature/humidity sensor, but I have no idea how. I have extended the internal thermal sensor using a coaxial cable.
This device has 5 relays (each 2A max) to control heater, cooler and fan (3 speeds). It can be used to control valves. I'm using additionally 2ch 16A Aqara relay to control an electric floor heater and monitor power consumption.

Look at this thread
https://github.com/Koenkk/zigbee2mqtt/issues/3306

Adminiuga, 2020-05-31:
Try HA dev version. It supports generic Zigbee thermostat, so question is how closely xiaomi bis following Zigbee specs (rhetorical question)

slydiman, 2020-05-31:
I'm using hass.io in docker on Raspberry PI4. 
What is the easiest way to try HA dev version? 
I have joined to beta channel in supervisor, but I don't see any updates.

Adminiuga, 2020-05-31:
See on docker hub tag for the latest dev build, then in hassio cli something like `ha core update --version home-assistant.dev20200530` or whatever tag was on docker hub

Adminiuga, 2020-05-31:
Or wait till Wednesday, I think beta is cut to on Wednesday

closed 2021-08-13 by github-actions[bot]