zigpy/zha-device-handlers #366: [BUG] Plaid humidity sensor do not report battery

ocalvo, 2020-06-02:
**Describe the bug**
Spruce humidity sensor Gen1 and Gen3 correctly report humidity and temperature but do not report battery correctly. 
The reported battery is always 0.

Gen3 model: PLAID SYSTEMS PS-SPRZMS-SLP3

**To Reproduce**
Steps to reproduce the behavior:
1. Pair a Spruce Sensor Gen1 or Gen3 with home assistant using ZHA integration.
2. Notice the battery sensor

**Expected behavior**
Battery reported should not be 0

**Additional context**
```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x0000](EZSP): started initialization
[0x0000:ZDO](EZSP): 'async_initialize' stage succeeded
[0x0000](EZSP): power source: Mains
[0x0000](EZSP): completed initialization
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0xe440 (00:0d:6f:00:0c:1c:b4:d7) joined the network
[0xe440:zdo] ZDO request ZDOCmd.Device_annce: [0xe440, 00:0d:6f:00:0c:1c:b4:d7, 128]
[0xe440] Requesting 'Node Descriptor'
Tries remaining: 2
[0xe440] Extending timeout for 0x05 request
[0xe440] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4098 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0xe440] Discovering endpoints
Tries remaining: 3
[0xe440] Extending timeout for 0x07 request
[0xe440] Discovered endpoints: [1]
[0xe440:1] Discovering endpoint information
Tries remaining: 3
[0xe440] Extending timeout for 0x09 request
[0xe440:1] Discovered endpoint information: <Optional endpoint=1 profile=260 device_type=1029 device_version=0 input_clusters=[0, 1, 3, 1026, 1029] output_clusters=[3, 25]>
[0xe440] Extending timeout for 0x0b request
[0xe440:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=11 command_id=Command.Read_Attributes_rsp>
[0xe440:1] Manufacturer: PLAID SYSTEMS
[0xe440:1] Model: PS-SPRZMS-SLP3
Checking quirks for PLAID SYSTEMS PS-SPRZMS-SLP3 (00:0d:6f:00:0c:1c:b4:d7)
Considering <class 'zhaquirks.plaid.soil.SoilMoisture'>
Found custom device replacement for 00:0d:6f:00:0c:1c:b4:d7: <class 'zhaquirks.plaid.soil.SoilMoisture'>
device - 0xe440:00:0d:6f:00:0c:1c:b4:d7 entering async_device_initialized - is_new_join: True
device - 0xe440:00:0d:6f:00:0c:1c:b4:d7 has joined the ZHA zigbee network
[0xe440](PS-SPRZMS-SLP3): started configuration
[0xe440:ZDO](PS-SPRZMS-SLP3): 'async_configure' stage succeeded
[0xe440] Extending timeout for 0x0d request
[0xe440] Extending timeout for 0x0f request
[0xe440] Extending timeout for 0x11 request
[0xe440:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0xe440] Extending timeout for 0x13 request
[0xe440:1:0x0402]: bound 'temperature' cluster: Status.SUCCESS
[0xe440] Extending timeout for 0x15 request
[0xe440:1:0x0405]: bound 'humidity' cluster: Status.SUCCESS
[0xe440] Extending timeout for 0x17 request
[0xe440:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=19 command_id=Command.Configure_Reporting_rsp>
[0xe440:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0xe440] Extending timeout for 0x19 request
[0xe440:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=21 command_id=Command.Configure_Reporting_rsp>
[0xe440:1:0x0402]: reporting 'measured_value' attr on 'temperature' cluster: 30/900/50: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0xe440:1:0x0402]: finished channel configuration
[0xe440] Extending timeout for 0x1b request
[0xe440:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=23 command_id=Command.Configure_Reporting_rsp>
[0xe440:1:0x0405]: reporting 'measured_value' attr on 'humidity' cluster: 30/900/50: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0xe440:1:0x0405]: finished channel configuration
[0xe440] Extending timeout for 0x1d request
[0xe440:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=25 command_id=Command.Configure_Reporting_rsp>
[0xe440:1:0x0001]: reporting 'battery_percentage_remaining' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.INSUFFICIENT_SPACE direction=ReportingDirection.SendReports attrid=33>]]'
[0xe440:1:0x0001]: finished channel configuration
[... 23 lines cut ...]
[0xe440] Extending timeout for 0x27 request
[0xe440:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=35 command_id=Command.Read_Attributes_rsp>
[0xe440:1:0x0001]: initializing channel: from_cache: False
[0xe440] Extending timeout for 0x29 request
[0xe440:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=37 command_id=Command.Read_Attributes_rsp>
[0xe440:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=39 command_id=Command.Read_Attributes_rsp>
[0xe440] Extending timeout for 0x2b request
[0xe440:1:0x0019]: initializing channel: from_cache: False
[0xe440:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=41 command_id=Command.Read_Attributes_rsp>
[0xe440:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=43 command_id=Command.Read_Attributes_rsp>
[0xe440:1:0x0000]: initializing channel: from_cache: False
[0xe440:1:0x0001]: 'async_initialize' stage succeeded
[0xe440:1:0x0402]: 'async_initialize' stage succeeded
[0xe440:1:0x0405]: 'async_initialize' stage succeeded
[0xe440:1:0x0000]: 'async_initialize' stage succeeded
[0xe440:1:0x0019]: 'async_initialize' stage succeeded
[0xe440](PS-SPRZMS-SLP3): power source: Battery or Unknown
[0xe440](PS-SPRZMS-SLP3): completed initialization
[0xe440:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=114 command_id=Command.Report_Attributes>
[0xe440:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=2218>>]]
[0xe440:1:0x0402] Attribute report received: measured_value=2218
[0xe440:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=115 command_id=Command.Report_Attributes>
[0xe440:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
[0xe440:1:0x0405] Attribute report received: measured_value=0
```

ocalvo, 2020-06-02:
I needed to reactivate [this bug](https://github.com/zigpy/zha-device-handlers/issues/323)

ocalvo, 2020-06-02:
```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x0000](EZSP): started initialization
[0x0000:ZDO](EZSP): 'async_initialize' stage succeeded
[0x0000](EZSP): power source: Mains
[0x0000](EZSP): completed initialization
[0xe440](PS-SPRZMS-SLP3): Attempting to checkin with device - missed checkins: 2
[0xe440] Extending timeout for 0x05 request
Device 0x84f9 (00:0d:6f:00:0c:1c:b4:d7) joined the network
Device 00:0d:6f:00:0c:1c:b4:d7 changed id (0xe440 => 0x84f9)
[0x84f9:zdo] ZDO request ZDOCmd.Device_annce: [0x84f9, 00:0d:6f:00:0c:1c:b4:d7, 128]
device - 34041:00:0d:6f:00:0c:1c:b4:d7 entering async_device_initialized - is_new_join: False
device - 34041:00:0d:6f:00:0c:1c:b4:d7 has been reset and re-added or its nwk address changed
skipping discovery for previously discovered device - 34041:00:0d:6f:00:0c:1c:b4:d7
[34041](PS-SPRZMS-SLP3): started configuration
[34041:ZDO](PS-SPRZMS-SLP3): 'async_configure' stage succeeded
[0x84f9] Extending timeout for 0x07 request
[0x84f9] Extending timeout for 0x09 request
[0x84f9] Extending timeout for 0x0b request
[34041](PS-SPRZMS-SLP3): started initialization
[34041:ZDO](PS-SPRZMS-SLP3): 'async_initialize' stage succeeded
[34041:1:0x0405]: bound 'humidity' cluster: Status.SUCCESS
[0x84f9] Extending timeout for 0x0d request
[34041:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0x84f9] Extending timeout for 0x0f request
[34041:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[34041:1:0x0000]: finished channel configuration
[0x84f9] Extending timeout for 0x11 request
[0x84f9:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=13 command_id=Command.Configure_Reporting_rsp>
[34041:1:0x0405]: reporting 'measured_value' attr on 'humidity' cluster: 30/900/50: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[34041:1:0x0405]: finished channel configuration
[0x84f9] Extending timeout for 0x13 request
[0x84f9:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=15 command_id=Command.Configure_Reporting_rsp>
[0x84f9:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=17 command_id=Command.Read_Attributes_rsp>
[34041:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0x84f9] Extending timeout for 0x15 request
[34041:1:0x0000]: initializing channel: from_cache: False
[0x84f9] Extending timeout for 0x17 request
[34041:1:0x0402]: bound 'temperature' cluster: Status.TABLE_FULL
[0x84f9] Extending timeout for 0x19 request
[0x84f9:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=21 command_id=Command.Configure_Reporting_rsp>
[34041:1:0x0001]: reporting 'battery_percentage_remaining' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[34041:1:0x0001]: finished channel configuration
[34041:1:0x0405]: initializing channel: from_cache: False
[0x84f9] Extending timeout for 0x1b request
[34041:1:0x0019]: bound 'ota' cluster: Status.TABLE_FULL
[34041:1:0x0019]: finished channel configuration
[0x84f9] Extending timeout for 0x1d request
[0x84f9:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=25 command_id=Command.Configure_Reporting_rsp>
[34041:1:0x0402]: reporting 'measured_value' attr on 'temperature' cluster: 30/900/50: Result: '[[<ConfigureReportingResponseRecord status=Status.INSUFFICIENT_SPACE direction=ReportingDirection.SendReports attrid=0>]]'
[34041:1:0x0402]: finished channel configuration
[0x84f9] Extending timeout for 0x1f request
[34041:1:0x0405]: 'async_configure' stage succeeded
[34041:1:0x0001]: 'async_configure' stage succeeded
[34041:1:0x0000]: 'async_configure' stage succeeded
[34041:1:0x0402]: 'async_configure' stage succeeded
[34041:1:0x0019]: 'async_configure' stage succeeded
[34041](PS-SPRZMS-SLP3): completed configuration
[34041](PS-SPRZMS-SLP3): stored in registry: ZhaDeviceEntry(name='PLAID SYSTEMS PS-SPRZMS-SLP3', ieee='00:0d:6f:00:0c:1c:b4:d7', last_seen=1591120231.0971005)
[0x84f9] Extending timeout for 0x21 request
[0x84f9:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=27 command_id=Command.Read_Attributes_rsp>
[0x84f9:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=29 command_id=Command.Read_Attributes_rsp>
[34041:1:0x0001]: initializing channel: from_cache: False
[0x84f9] Extending timeout for 0x23 request
[34041:1:0x0402]: initializing channel: from_cache: False
[0x84f9] Extending timeout for 0x25 request
[... cut ...]
[0x84f9] Delivery error for seq # 0x05, on endpoint id 1 cluster 0x0000: message send failure
[0x84f9:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=4 command_id=Command.Report_Attributes>
[0x84f9:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=945>>]]
[0x84f9:1:0x0405] Attribute report received: measured_value=945
```

ocalvo, 2020-06-02:
Device info:
```
{
  "node_descriptor": "<NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4098 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0405",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0405"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "PLAID SYSTEMS",
  "model": "PS-SPRZMS-SLP3",
  "class": "zhaquirks.plaid.soil.SoilMoisture"
}
```

Adminiuga, 2020-06-02:
Seems like there's already device quirk for it and battery percent remaining attribute was configured successfuly for the reporting. 
How long device has been on the network?

ocalvo, 2020-06-02:
Months

Adminiuga, 2020-06-02:
Do you have a second zigbee stick for this isolated environment? And is it venv, hassio or docker? in other words how easily can you modify the source files to test a few things.

I'm concerned that binding table is limited on this device, as it keeps throwing 
```
[0xe440:1:0x0000]: bound 'basic' cluster: Status.TABLE_FULL
[0xe440:1:0x0000]: finished channel configuration
[0xe440:1:0x0019]: bound 'ota' cluster: Status.TABLE_FULL
[0xe440:1:0x0019]: finished channel configuration
```
in one log, and in the other
```
[34041:1:0x0402]: bound 'temperature' cluster: Status.TABLE_FULL
[34041:1:0x0019]: bound 'ota' cluster: Status.TABLE_FULL
[34041:1:0x0019]: finished channel configuration
```

on different clusters, depending on configuration order.

ocalvo, 2020-06-02:
Yes, I have a second ZigBee stick running hass on a developer environment using windows Linux subststem so I can tweak very easyly.
Let me know what you want to change.

tube0013, 2020-06-02:
I have a plaid sensor not yet in production and a test box (the one with the zzh stick) I can mess with.  this sensor reports the battery voltage on the mains attribute for some reason.

Adminiuga, 2020-06-02:
@tube0013 can you provide full debug logs of a such report?

tube0013, 2020-06-02:
sure, let me move i'll move it over to my test box.  my memory was the voltage was not sent very often even when triggering it.

ocalvo, 2020-06-02:
How do I activate "full debug logs"?

Adminiuga, 2020-06-03:
https://www.home-assistant.io/integrations/zha/#debug-logging

tube0013, 2020-06-03:
I had issues getting this pair on my test box oh well I'll deal with those another day....  put it back on my prod box had to pull the battery and re-insert to get a voltage. 
 
this is immediately after re-inserting the battery:

```
2020-06-02 21:12:15 DEBUG (MainThread) [zigpy.zdo] [0x648a:zdo] ZDO request ZDOCmd.NWK_addr_req: [00:0d:6f:00:0a:ff:73:23, 0, 0]
2020-06-02 21:12:15 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=2 command_id=Command.Report_Attributes>
2020-06-02 21:12:15 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0001] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=2997>>]]
2020-06-02 21:12:15 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0001] Attribute report received: mains_voltage=2997
2020-06-02 21:12:15 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=1 command_id=Command.Report_Attributes>
2020-06-02 21:12:15 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
2020-06-02 21:12:15 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] Attribute report received: measured_value=0
2020-06-02 21:12:16 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=1 command_id=Command.Report_Attributes>
2020-06-02 21:12:16 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
2020-06-02 21:12:16 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] Attribute report received: measured_value=0
```

tube0013, 2020-06-03:
Spruce support were generally pretty good they even took a look at the quirk at the time I had a ticket open with them, the rep I worked with did not know why it reports voltage on the mains attribute....

[image]

Adminiuga, 2020-06-03:
Impressive. Kudos to their support for at least trying to be helpful

tube0013, 2020-06-03:
they mentioned that they gotten another ticket from a HA user at the same time,  so demand was showing lol.

ocalvo, 2020-06-03:
```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x0000](EZSP): started initialization
[0x0000:ZDO](EZSP): 'async_initialize' stage succeeded
[0x0000](EZSP): power source: Mains
[0x0000](EZSP): completed initialization
Device 0xc187 (00:0d:6f:00:0c:1c:c7:3c) joined the network
[0xc187:zdo] ZDO request ZDOCmd.Device_annce: [0xc187, 00:0d:6f:00:0c:1c:c7:3c, 128]
[0xc187] Requesting 'Node Descriptor'
Tries remaining: 2
[0xc187] Extending timeout for 0x09 request
[0xc187] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4098 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0xc187] Discovering endpoints
Tries remaining: 3
[0xc187] Extending timeout for 0x0b request
[0xc187] Discovered endpoints: [1]
[0xc187:1] Discovering endpoint information
Tries remaining: 3
[0xc187] Extending timeout for 0x0d request
[0xc187:1] Discovered endpoint information: <Optional endpoint=1 profile=260 device_type=1029 device_version=0 input_clusters=[0, 1, 3, 1026, 1029] output_clusters=[3, 25]>
[0xc187] Extending timeout for 0x0f request
[0xc187:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=15 command_id=Command.Read_Attributes_rsp>
[0xc187:1] Manufacturer: PLAID SYSTEMS
[0xc187:1] Model: PS-SPRZMS-SLP3
Checking quirks for PLAID SYSTEMS PS-SPRZMS-SLP3 (00:0d:6f:00:0c:1c:c7:3c)
Considering <class 'zhaquirks.plaid.soil.SoilMoisture'>
Found custom device replacement for 00:0d:6f:00:0c:1c:c7:3c: <class 'zhaquirks.plaid.soil.SoilMoisture'>
device - 0xc187:00:0d:6f:00:0c:1c:c7:3c entering async_device_initialized - is_new_join: True
device - 0xc187:00:0d:6f:00:0c:1c:c7:3c has joined the ZHA zigbee network
[0xc187](PS-SPRZMS-SLP3): started configuration
[0xc187:ZDO](PS-SPRZMS-SLP3): 'async_configure' stage succeeded
[0xc187] Extending timeout for 0x11 request
[0xc187] Extending timeout for 0x13 request
[0xc187] Extending timeout for 0x15 request
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_NWK_Update_rsp: [<Status.SUCCESS: 0>, <Channels.ALL_CHANNELS: 134215680>, 21, 10, [209, 212, 180, 199, 231, 196, 176, 184, 221, 196, 213, 175, 221, 203, 206, 194]]
[0x0000:zdo] Unsupported ZDO request:ZDOCmd.Mgmt_NWK_Update_rsp
[0xc187] Delivery error for seq # 0x11, on endpoint id 0 cluster 0x0021: message send failure
[0xc187:1:0x0405]: Failed to bind 'humidity' cluster: [0xc187:0:0x0021]: Message send failure
[0xc187] Extending timeout for 0x17 request
[0xc187] Delivery error for seq # 0x13, on endpoint id 0 cluster 0x0021: message send failure
[0xc187:1:0x0001]: Failed to bind 'power' cluster: [0xc187:0:0x0021]: Message send failure
[0xc187] Extending timeout for 0x19 request
[0xc187] Delivery error for seq # 0x15, on endpoint id 0 cluster 0x0021: message send failure
[0xc187:1:0x0000]: Failed to bind 'basic' cluster: [0xc187:0:0x0021]: Message send failure
[0xc187:1:0x0000]: finished channel configuration
[0xc187] Extending timeout for 0x1b request
[0xc187] Delivery error for seq # 0x17, on endpoint id 1 cluster 0x0405: message send failure
[0xc187:1:0x0405]: failed to set reporting for 'measured_value' attr on 'humidity' cluster: [0xc187:1:0x0405]: Message send failure
[0xc187:1:0x0405]: finished channel configuration
[0xc187] Extending timeout for 0x1d request
[0xc187] Delivery error for seq # 0x19, on endpoint id 1 cluster 0x0001: message send failure
[0xc187:1:0x0001]: failed to set reporting for 'battery_voltage' attr on 'power' cluster: [0xc187:1:0x0001]: Message send failure
[0xc187] Extending timeout for 0x1f request
[0xc187] Delivery error for seq # 0x1b, on endpoint id 1 cluster 0x0000: message send failure
[0xc187:1:0x0000]: initializing channel: from_cache: False
[0xc187] Extending timeout for 0x21 request
[0xc187:1:0x0402]: bound 'temperature' cluster: Status.SUCCESS
[0xc187] Extending timeout for 0x23 request
[0xc187:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=31 command_id=Command.Configure_Reporting_rsp>
[0xc187:1:0x0001]: reporting 'battery_percentage_remaining' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.INSUFFICIENT_SPACE direction=ReportingDirection.SendReports attrid=33>]]'
[0xc187:1:0x0001]: finished channel configuration
[0xc187:1:0x0019]: bound 'ota' cluster: Status.TABLE_FULL
[0xc187:1:0x0019]: finished channel configuration
[0xc187:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=35 command_id=Command.Configure_Reporting_rsp>
[0xc187:1:0x0402]: reporting 'measured_value' attr on 'temperature' cluster: 30/900/50: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0xc187:1:0x0402]: finished channel configuration
[0xc187:1:0x0405]: 'async_configure' stage succeeded
[0xc187:1:0x0001]: 'async_configure' stage succeeded
[0xc187:1:0x0000]: 'async_configure' stage succeeded
[0xc187:1:0x0402]: 'async_configure' stage succeeded
[... 21 lines cut ...]
[... cut ...]
[0xc187:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=2289>>]]
[0xc187:1:0x0402] Attribute report received: measured_value=2289
[0xc187:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=6 command_id=Command.Report_Attributes>
[0xc187:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=358>>]]
[0xc187:1:0x0405] Attribute report received: measured_value=358
```

ocalvo, 2020-06-03:
What does it mean that it uses the "Power endpoint with Main"?

tube0013, 2020-06-03:
It basically means the sensor reports its battery voltage on the zigbee attribute meant for mains voltage in the zigbee spec.  In the zigbee spec there are defined clusters and attributes, when device follow them everything should just work, however some device like this one deviate from the spec in this case sending the battery voltage on the mains voltage attribute.

ocalvo, 2020-06-03:
Ahh, and the mains voltage attribute is meant for NON battery operated devices?

ocalvo, 2020-06-03:
How do I verify/test this in my dev environment?

Adminiuga, 2020-06-03:
https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf PDF page 110

There's already quirk which translates mains voltage into battery ones, although IIRC regular battery voltage is in 100mV (0.1V) and plaid seems to report in mV, so it should divide the actual value by `/100` prior sending it as a battery voltage 🤷 

@tube0013 do you get battery voltage?

tube0013, 2020-06-03:
not reliably, I got it above after pulling and re-inserting the battery. In the past (my original issue for this) I was able to query it but I have yet been able to do that again. It is super "shy" about sending voltage.

tube0013, 2020-06-03:
Does the current quirk take into count the 3volt battery as well, as I don't think battery size is sent

Adminiuga, 2020-06-03:
and that's wrong attribute. try reading these attrs:

[image]
[image]
[image]

Adminiuga, 2020-06-03:
> Does the current quirk take into count the 3volt battery as well, as I don't think battery size is sent

It doesn't, but  I thought `battery_percentage_remaining` was reported correctly. Is it not the case?

tube0013, 2020-06-03:
I'll try and double check,  I've only ever seen it report voltage.  .. here is it's log since 00:00 today, it reports every hour and on changes i believe.   
```
2020-06-03 00:23:16 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=21 command_id=Command.Report_Attributes>
2020-06-03 00:23:16 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
2020-06-03 00:23:16 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] Attribute report received: measured_value=0
2020-06-03 01:23:16 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=23 command_id=Command.Report_Attributes>
2020-06-03 01:23:16 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
2020-06-03 01:23:16 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] Attribute report received: measured_value=0
2020-06-03 02:12:31 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=25 command_id=Command.Report_Attributes>
2020-06-03 02:12:31 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0001] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=3049>>]]
2020-06-03 02:12:31 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0001] Attribute report received: mains_voltage=3049
2020-06-03 02:23:17 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=26 command_id=Command.Report_Attributes>
2020-06-03 02:23:17 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
2020-06-03 02:23:17 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] Attribute report received: measured_value=0
2020-06-03 02:23:19 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=26 command_id=Command.Report_Attributes>
2020-06-03 02:23:19 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
2020-06-03 02:23:19 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] Attribute report received: measured_value=0
2020-06-03 03:07:28 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=27 command_id=1>
2020-06-03 03:07:28 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0019] ZCL request 0x0001: [1, 4098, 0, 1, 3300]
2020-06-03 03:07:28 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0019] OTA query_next_image handler for 'PLAID SYSTEMS PS-SPRZMS-SLP3': field_control=1, manufacture_id=4098, image_type=0, current_file_version=1, hardware_version=3300
2020-06-03 03:07:28 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0019] No OTA image is available
2020-06-03 03:23:18 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=29 command_id=Command.Report_Attributes>
2020-06-03 03:23:18 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
2020-06-03 03:23:18 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] Attribute report received: measured_value=0
2020-06-03 03:23:23 DEBUG (MainThread) [zigpy.device] [0x648a] Delivery error for seq # 0x1d, on endpoint id 1 cluster 0x0405: message send failure
2020-06-03 04:23:19 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=31 command_id=Command.Report_Attributes>
2020-06-03 04:23:19 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
2020-06-03 04:23:19 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] Attribute report received: measured_value=0
2020-06-03 05:23:20 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=33 command_id=Command.Report_Attributes>
2020-06-03 05:23:20 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
2020-06-03 05:23:20 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] Attribute report received: measured_value=0
[... cut ...]
2020-06-03 13:23:33 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=51 command_id=Command.Report_Attributes>
2020-06-03 13:23:33 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=0>>]]
2020-06-03 13:23:33 DEBUG (MainThread) [zigpy.zcl] [0x648a:1:0x0405] Attribute report received: measured_value=0
```

Adminiuga, 2020-06-03:
> but I have yet been able to do that again. It is super "shy" about sending voltage.

question is: does this require attribute reporting configured on mains_voltage, which I'm not positive is reportable. It could be rarely reporting it, because it changes rarely too

tube0013, 2020-06-03:
I think that is the case. I think in my above log it changes as it's reading settled a bit after I pulled and re-inserted the battery a few hours earlier. there were 2 voltage reports over 13 hours

Adminiuga, 2020-06-03:
@ocalvo assuming you've been running the debug logs for the past 24 hours, find NWK of the plaid sensor (from zha device card) and run
`grep PLAID_SENSOR_NWKADDRESS:1:0x0001 home-assistant.log`

Adminiuga, 2020-06-03:
but I also have a suspicion, that in current implementation, right after startup, it would get 0 upon 1st any report from the device

tube0013, 2020-06-03:
I have a second sensor in prod, outside monitoring soil, and I don't have a single voltage report in my log.

ocalvo, 2020-06-03:
I got a report of -1.0
[image]

Trying to repair the device again

ocalvo, 2020-06-03:
```
2020-06-03 11:03:55 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x9a7b:1:0x0001]: failed to set reporting for 'battery_voltage' attr on 'power' cluster: [0x9a7b:1:0x0001]: Message send failure
```

Adminiuga, 2020-06-03:
remove the battery, wait a few minutes, put the battery in and wait till you get some reports. then run the grep again to see if it sends anything on that cluster

@tube0013 the sensor seems to have limited binding table, so depending on which cluster get bound first, some clusters may fail binding and it if happens to be the power configuration cluster, then you might not get the reports

ocalvo, 2020-06-03:
Here is the output when querying battery voltage:
```
2020-06-03 11:14:08 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.plaid_systems_ps_sprzms_slp3_3cc71c0c_power, old_state=<state sensor.plaid_systems_ps_sprzms_slp3_3cc71c0c_power=0; unit_of_measurement=%, friendly_name=PLAID SYSTEMS PS-SPRZMS-SLP3 3cc71c0c power, device_class=battery @ 2020-06-03T11:05:17.790955-07:00>, new_state=<state sensor.plaid_systems_ps_sprzms_slp3_3cc71c0c_power=0; battery_voltage=0.0, unit_of_measurement=%, friendly_name=PLAID SYSTEMS PS-SPRZMS-SLP3 3cc71c0c power, device_class=battery @ 2020-06-03T11:05:17.790955-07:00>>
2020-06-03 11:14:08 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.plaid_systems_ps_sprzms_slp3_3cc71c0c_power, old_state=<state sensor.plaid_systems_ps_sprzms_slp3_3cc71c0c_power=0; battery_voltage=0.0, unit_of_measurement=%, friendly_name=PLAID SYSTEMS PS-SPRZMS-SLP3 3cc71c0c power, device_class=battery @ 2020-06-03T11:05:17.790955-07:00>, new_state=<state sensor.plaid_systems_ps_sprzms_slp3_3cc71c0c_power=-1; battery_voltage=0.0, unit_of_measurement=%, friendly_name=PLAID SYSTEMS PS-SPRZMS-SLP3 3cc71c0c power, device_class=battery @ 2020-06-03T11:14:08.310522-07:00>>
```

Adminiuga, 2020-06-03:
well, the real fix  here (maybe not the most feasible one) is to contact their support and ask them to provide battery voltage over correct attributes

ocalvo, 2020-06-03:
What do you mean "bound the power configuration cluster"?

Here is the output after inserting the battery:
```
2020-06-03 11:21:19 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'0004010504010140010000b2ffca7b9affff0808000a0000218900'
2020-06-03 11:21:19 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=1029 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=178>, 255, -54, 0x9a7b, 255, 255, b'\x08\x00\n\x00\x00!\x89\x00']
2020-06-03 11:21:19 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=Command.Report_Attributes>
2020-06-03 11:21:19 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=137>>]]
2020-06-03 11:21:19 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] Attribute report received: measured_value=137
2020-06-03 11:21:19 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x9a7b, <EmberApsFrame profileId=260 clusterId=1029 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=0>, 58, b'\x18\x00\x0b\n\x00')
2020-06-03 11:21:19 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'0098'

2020-06-03 11:21:20 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'007b9a04010504010140010000983a0000'
2020-06-03 11:21:20 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 39547, <EmberApsFrame profileId=260 clusterId=1029 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=152>, 58, <EmberStatus.SUCCESS: 0>, b'']
2020-06-03 11:21:27 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2020-06-03 11:21:27 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2020-06-03 11:21:35 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'0400000600000008010000b3ffcc7b9affff0901fdff040101190000'
2020-06-03 11:21:35 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST: 4>, <EmberApsFrame profileId=0 clusterId=6 sourceEndpoint=0 destinationEndpoint=0 options=264 groupId=0 sequence=179>, 255, -52, 0x9a7b, 255, 255, b'\x01\xfd\xff\x04\x01\x01\x19\x00\x00']
2020-06-03 11:21:35 DEBUG (MainThread) [zigpy.zdo] [0x9a7b:zdo] ZDO request ZDOCmd.Match_Desc_req: [0xfffd, 260, [25], []]
2020-06-03 11:21:35 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x9a7b, <EmberApsFrame profileId=0 clusterId=32774 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=1>, 59, b'\x01\x00\x00\x00\x01\x01')
2020-06-03 11:21:35 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'0099'
2020-06-03 11:21:36 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'007b9a00000680000040010000993b0000'
2020-06-03 11:21:36 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 39547, <EmberApsFrame profileId=0 clusterId=32774 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=153>, 59, <EmberStatus.SUCCESS: 0>, b'']
2020-06-03 11:21:37 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2020-06-03 11:21:37 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2020-06-03 11:21:37 DEBUG (MainThread) [bellows.ezsp] Application frame 69 (incomingMessageHandler) received: b'0004011900010140000000b4ffcc7b9affff0e010101010210000001000000e40c'
2020-06-03 11:21:37 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=25 sourceEndpoint=1 destinationEndpoint=1 options=64 groupId=0 sequence=180>, 255, -52, 0x9a7b, 255, 255, b'\x01\x01\x01\x01\x02\x10\x00\x00\x01\x00\x00\x00\xe4\x0c']
2020-06-03 11:21:37 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=1 command_id=1>
2020-06-03 11:21:37 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0019] ZCL request 0x0001: [1, 4098, 0, 1, 3300]
2020-06-03 11:21:37 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0019] OTA query_next_image handler for 'PLAID SYSTEMS PS-SPRZMS-SLP3': field_control=1, manufacture_id=4098, image_type=0, current_file_version=1, hardware_version=3300
2020-06-03 11:21:37 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0019] No OTA image is available
2020-06-03 11:21:37 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x9a7b, <EmberApsFrame profileId=260 clusterId=25 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=1>, 60, b'\x19\x01\x02\x98')
2020-06-03 11:21:37 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'009a'
2020-06-03 11:21:38 DEBUG (MainThread) [bellows.ezsp] Application frame 63 (messageSentHandler) received: b'007b9a040119000101400100009a3c0000'
2020-06-03 11:21:38 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 39547, <EmberApsFrame profileId=260 clusterId=25 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=154>, 60, <EmberStatus.SUCCESS: 0>, b'']
```

ocalvo, 2020-06-03:
Power is reading again at `-1 %`

Adminiuga, 2020-06-03:
`[0xe440:1:0x0001]: bound 'power' cluster: Status.SUCCESS`

ocalvo, 2020-06-03:
I don't know, I did not save the logs.

Adminiuga, 2020-06-03:
>Power is reading again at -1 %

 logs of what it did report? in the logs when you see `[0xe440:1:0x0001]` the first `0xe440` is the NWK address of the device, may change from time to time if you rejoin it. the `:1` is endpoint id and `0x0001` is the cluster id where cluster id of `0x0001` is the power configuration cluster

if you filter the logs by `NWK:endpoint_id:cluster_id` then you could see all the communication.
Without the logs it is only guess, but my  bet -1 is what was read for the `battery_percentage_remaining`

Adminiuga, 2020-06-03:
that's already homeassistant core. so the report or response to attribute read should have been prior to that

ocalvo, 2020-06-03:
When asking for the battery voltage and waking up the device with the magnet:
```
2020-06-03 11:41:59 DEBUG (MainThread) [zigpy.device] [0x9a7b] Extending timeout for 0x04 request
2020-06-03 11:41:59 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 00:0d:6f:00:0c:1c:c7:3c/0x9a7b
2020-06-03 11:41:59 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x9a7b, <EmberApsFrame profileId=260 clusterId=1 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=4>, 5, b'\x00\x04\x00 \x00')
2020-06-03 11:42:06 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteErrorHandler frame with [<EmberStatus.MAC_INDIRECT_TIMEOUT: 66>, 0x9a7b]
2020-06-03 11:42:06 DEBUG (MainThread) [bellows.zigbee.application] Processing route error: status=EmberStatus.MAC_INDIRECT_TIMEOUT, nwk=0x9a7b
2020-06-03 11:42:09 DEBUG (MainThread) [zigpy.device] [0x9a7b] Extending timeout for 0x06 request
2020-06-03 11:42:09 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 00:0d:6f:00:0c:1c:c7:3c/0x9a7b
2020-06-03 11:42:09 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x9a7b, <EmberApsFrame profileId=260 clusterId=1 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=6>, 7, b'\x00\x06\x00 \x00')
2020-06-03 11:42:10 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=1 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=186>, 255, -51, 0x9a7b, 255, 255, b'\x08\x04\x01 \x00\x00 \x00']
2020-06-03 11:42:10 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=4 command_id=Command.Read_Attributes_rsp>
2020-06-03 11:42:10 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=1 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=187>, 255, -51, 0x9a7b, 255, 255, b'\x08\x06\x01 \x00\x00 \x00']
2020-06-03 11:42:10 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=6 command_id=Command.Read_Attributes_rsp>
2020-06-03 11:42:10 DEBUG (MainThread) [zigpy.device] [0x9a7b] Extending timeout for 0x08 request
2020-06-03 11:42:10 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 00:0d:6f:00:0c:1c:c7:3c/0x9a7b
2020-06-03 11:42:10 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x9a7b, <EmberApsFrame profileId=260 clusterId=1 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=8>, 9, b'\x00\x08\x00 \x00')
2020-06-03 11:42:11 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=1 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=188>, 255, -53, 0x9a7b, 255, 255, b'\x08\x08\x01 \x00\x00 \x00']
2020-06-03 11:42:11 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=8 command_id=Command.Read_Attributes_rsp>
2020-06-03 11:42:12 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=1029 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=189>, 255, -52, 0x9a7b, 255, 255, b'\x08\x04\n\x00\x00!\x89\x00']
2020-06-03 11:42:12 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=4 command_id=Command.Report_Attributes>
2020-06-03 11:42:12 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=137>>]]
2020-06-03 11:42:12 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] Attribute report received: measured_value=137
2020-06-03 11:42:12 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x9a7b, <EmberApsFrame profileId=260 clusterId=1029 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=4>, 10, b'\x18\x04\x0b\n\x00')
```

Adminiuga, 2020-06-03:
Per logs, reading battery voltage return 0, essentially what is seen in the state.

ocalvo, 2020-06-03:
```
2020-06-03 13:53:36 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=1029 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=14>, 255, -49, 0x9a7b, 255, 255, b'\x08\x03\n\x00\x00!\x89\x00']
2020-06-03 13:53:36 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=3 command_id=Command.Report_Attributes>
2020-06-03 13:53:36 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=137>>]]
2020-06-03 13:53:36 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] Attribute report received: measured_value=137
2020-06-03 13:53:36 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x9a7b, <EmberApsFrame profileId=260 clusterId=1029 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=3>, 15, b'\x18\x03\x0b\n\x00')
2020-06-03 14:53:32 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=1 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=15>, 255, -48, 0x9a7b, 255, 255, b'\x08\x04\n \x00 \x00!\x00 \x00']
2020-06-03 14:53:32 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=4 command_id=Command.Report_Attributes>
2020-06-03 14:53:32 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0001] ZCL request 0x000a: [[<Attribute attrid=32 value=<TypeValue type=uint8_t, value=0>>, <Attribute attrid=33 value=<TypeValue type=uint8_t, value=0>>]]
2020-06-03 14:53:32 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0001] Attribute report received: battery_voltage=0, battery_percentage_remaining=0
2020-06-03 14:53:32 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x9a7b, <EmberApsFrame profileId=260 clusterId=1 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=4>, 16, b'\x18\x04\x0b\n\x00')
2020-06-03 14:53:37 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=1029 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=16>, 255, -48, 0x9a7b, 255, 255, b'\x08\x05\n\x00\x00!\x89\x00']
2020-06-03 14:53:37 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=5 command_id=Command.Report_Attributes>
2020-06-03 14:53:37 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=137>>]]
2020-06-03 14:53:37 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] Attribute report received: measured_value=137
2020-06-03 14:53:37 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x9a7b, <EmberApsFrame profileId=260 clusterId=1029 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=5>, 17, b'\x18\x05\x0b\n\x00')
2020-06-03 15:53:38 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=1029 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=17>, 255, -49, 0x9a7b, 255, 255, b'\x08\x06\n\x00\x00!\x89\x00']
2020-06-03 15:53:38 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=6 command_id=Command.Report_Attributes>
2020-06-03 15:53:38 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=137>>]]
2020-06-03 15:53:38 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0405] Attribute report received: measured_value=137
2020-06-03 15:53:38 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x9a7b, <EmberApsFrame profileId=260 clusterId=1029 sourceEndpoint=1 destinationEndpoint=1 options=320 groupId=0 sequence=6>, 18, b'\x18\x06\x0b\n\x00')
```

Adminiuga, 2020-06-04:
well, you sensor does report battery % remaining  and  battery_voltage, but both are 0
```
2020-06-03 14:53:32 DEBUG (MainThread) [zigpy.zcl] [0x9a7b:1:0x0001] Attribute report received: battery_voltage=0, battery_percentage_remaining=0
```

I wonder if your device version is different from  Tube's . if you navigate to HA Configuration -> Devices -> pick soil sensor,  what firmware number if any do you get?
[image]

2nd: give it more time and then filter by `[0x9a7b:1:0x0001]` I wanna see if it ever reports mains voltage.

Adminiuga, 2020-06-04:
by give it more time i mean like 24 hours. if it reports it, it won't report often

tube0013, 2020-06-04:
There are different versions, I have the gen3 sensor which has the magnet sensor on the top.  I believe the gen2 sensor has it on the side. As described here 

https://support.spruceirrigation.com/knowledge-base/re-join-spruce-sensor-to-the-network/

tube0013, 2020-06-04:
here is what I get:

[image]

I also queried most of the attributes in the basic and beyond the model/manufacturer it reported the date_code:
[image]

device seems to be chatty right now lol... battery-percent remaining:

[image]

still not sending  mains-voltage when requesting.

Adminiuga, 2020-06-04:
as it was mentioned: run debug logs for 24 hours. Then filter  by `0x9a7b:1:0x0001`  and maybe  gen 2 does not report at all, who knows

BTW, did you try to get `mains_voltage` attribute?

tube0013, 2020-06-04:
yes don't get anything when I request that attribute

Adminiuga, 2020-06-04:
Do you  get `None` ???

ocalvo, 2020-06-04:
> There are different versions, I have the gen3 sensor which has the magnet sensor on the top. I believe the gen2 sensor has it on the side. As described here
> 
> https://support.spruceirrigation.com/knowledge-base/re-join-spruce-sensor-to-the-network/

I have one Gen1 and 2 Gen3. The Gen1 and Gen3 are in "production".
The other Gen3 is in my test environment.

Adminiuga, 2020-06-13:
I'm running of solutions. Configured attribute reporting for the mains_voltage attribute, but it still won't report it. It was reporting the batter_voltage regularly though, the only problem it was reporting 0V

tube0013, 2020-06-13:
I can set pull in the changed quirk tonight and test. Did you pick one of these up? I've found the battery reporting to be non-existent sometimes and yet other times it has reported it frequently.

Adminiuga, 2020-06-13:
yep, got one.  need to automate watering. 
Try this version https://github.com/zigpy/zha-device-handlers/pull/377

This substitutes requests for `battery_voltage` attribue. Was 4 hours since joined the sensor, but it did not report it yet. It was reporting battery voltage like once in 4 hours. Gonna wait till tomorrow

tube0013, 2020-06-14:
Pulled the battery, waited about 5 secs put it back in, waited a few seconds, and hit it with the magnet.....
[image]

ocalvo, 2020-06-14:
> Pulled the battery, waited about 5 secs put it back in, waited a few seconds, and hit it with the magnet.....
> [image]

Can you share the version of soil.py that you are using? 
I would like to test this in my end.

tube0013, 2020-06-14:
it is the version #377 linked above I also updated __init__.py from #376

[9 more replies]

closed 2020-06-28 by Adminiuga