zigpy/zha-device-handlers #368: Aqara Temps - Wrong type detected

martynstar, 2020-06-03:
Hi there

It seems that something has slightly changed in the hardware on the temp sensor.
Previously this was detected and shows - Temperature, Pressure, Humidity and Battery life.
Zigbee information
{
  "node_descriptor": "<Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4151 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=0 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x5f01",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0403",
        "0x0405",
        "0xffff"
      ],
      "out_clusters": [
        "0x0000",
        "0x0004",
        "0xffff"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.weather",
  "class": "zhaquirks.xiaomi.aqara.weather.Weather"
}

Now the device i purchased recently is showing only the Temperature, Pressure and Humidity.
Zigbee info.

{
  "node_descriptor": "<Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4151 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=0 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0402",
        "0x0403",
        "0x0405",
        "0xffff"
      ],
      "out_clusters": [
        "0x0000",
        "0x0004",
        "0xffff"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.weather",
  "class": "zigpy.device.Device"
}

If we could get this added to the quirks it would be great! what more info can i grab or do we need some debug logs?

Thanks

The slight hardware change, everything else looks the same on the kit but maybe a firmware?

caylor93, 2020-06-26:
Is there a timeline for this update?

Adminiuga, 2020-06-26:
if you can make changes to your environment, then try manually updating `zhaquirks/xiaomi/aqara/weather.py` file with changes from https://github.com/zigpy/zha-device-handlers/pull/386/files

dmulcahey, 2020-06-28:
Fixed by #386

dmulcahey, 2020-06-29:
Closing. @caylor93 or @martynstar reopen this if this doesn't fix it.

closed 2020-06-29 by dmulcahey

caylor93, 2020-06-29:
Thanks!  Do you have to remove and re-add the devices for the battery state to show up?

dmulcahey, 2020-06-29:
They report the lifeline report about every 60 or so minutes. If you apply the change and restart HA it should just work.