zigpy/zha-device-handlers #369: [Device Support Request] Support for WXKG02LM (Aqara Double Switch)

homietw, 2020-06-05:
The device pairs in zha homeassistant, but only unknown sensors are shown.

Device is shown as quirk: zhaquirks.xiaomi.aqara.remote_b286acn01.RemoteB286ACN01
ZHA Version:

`
homeassistant                    0.110.2      /usr/src/homeassistant
homeassistant-pyozw              0.1.10
pyflic-homeassistant             0.4.dev0
pylgnetcast-homeassistant        0.2.0.dev0
zha-quirks                       0.0.39
zigpy                            0.20.4
zigpy-cc                         0.4.2
zigpy-deconz                     0.9.2
zigpy-xbee                       0.12.1
zigpy-zigate                     0.6.1
`

dmulcahey, 2020-06-28:
The device fires events. The development sensors will be removed in HA.

closed 2020-06-28 by dmulcahey

homietw, 2020-06-29:
@dmulcahey could you please explain your comment? Will the switch be supported anytime?

dmulcahey, 2020-06-29:
It’s already supported. It sends zha_event to HA. There are no entities. The ones that were there were from development. They are removed completely in 112.

homietw, 2020-06-29:
I wasn't aware of that. Now that I'm using the events, everything works out for me. :)
Thanks a lot for your effort!