zigpy/zha-device-handlers #372: [BUG] Xiaomi Magic Cube battery and multistate_input errors

nerdosity, 2020-06-11:
**Describe the bug**
1.  battery info is reported with negative values. As battery cannot have negative values, I think that we should get always ABSOLUTE value of battery report. We can create a template sensor, but it's not worth it since negative values are just wrong, in any case
2. multistate_input stay "UNKNOWN". I can see "analog_input" updating with degrees of rotation, but not multistate_input

**Screenshots**
[image]

**Additional context**
[image]

Adminiuga, 2020-06-11:
do you record battery levels? Is it always negative or is it jumping?
If jumping, how often does it go negative?

@dmulcahey is there any data reported on the multistate input cluster/

dmulcahey, 2020-06-11:
The actions are... rotate, slide, knock etc. Should be mapped to text sensors in HA

Adminiuga, 2020-06-11:
i'm getting  all the `zha_events` 
I guess the question is: should we really keep state and have entities for it? to my taste cube is more of a stateless remote type

Adminiuga, 2020-06-11:
@nerdosity there won't be any meaningful data in the `multistate_input`,  as  the attribute updates ~~always comes with data of `0x0103`~~ depends on the action 
To catch the  useful data, subscribe or base your automations based on zha_events

Adminiuga, 2020-06-12:
The decision was made to drop those completely.  Use `zha_events` instead

dmulcahey, 2020-06-12:
Or device actions... they’re much more user friendly

Adminiuga, 2020-06-12:
And I can't reproduce the battery problem
[image]

Does  it still report a negative value? if it does and it always report a negative value, would need a debug log with attribute report from the device in question. Enable debug log as specified in integration documentation and wait till it updates. You could search for `zhaquirks.xiaomi` and around it you should see something like:
```
2020-06-12 00:04:52 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMe
ssageType.INCOMING_UNICAST: 0>, <EmberApsFrame profileId=260 clusterId=0 sourceEndpoint=1 destinationEndpoint=1 options=256 grou
pId=0 sequence=238>, 255, -51, 0x458e, 255, 255, b'\x18\x9d\n\x01\xffB%\x01!\xbd\x0b\x04!\xa8\x13\x05!\xea\x00\x06$\x01\x00\x00\
x00\x00d)\xfc\x07e!\xdd\x1af+\x9c\x88\x01\x00\n!\xd1\xa7']
2020-06-12 00:04:52 DEBUG (MainThread) [zigpy.zcl] [0x458e:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl fra
me_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=157 comma
nd_id=Command.Report_Attributes>
2020-06-12 00:04:52 DEBUG (MainThread) [zigpy.zcl] [0x458e:1:0x0000] ZCL request 0x000a: [[<Attribute attrid=65281 value=<TypeVa
lue type=CharacterString, value=^A!�^K^D!�^S^E!�>>]]
2020-06-12 00:04:52 DEBUG (MainThread) [zigpy.zcl] [0x458e:1:0x0000] Attribute report received: 65281=^A!�^K^D!�^S^E!�
2020-06-12 00:04:52 DEBUG (MainThread) [zhaquirks.xiaomi] 00:15:8d:00:02:2c:b7:75 - Attribute report. attribute_id: [65281] value: [{'battery_voltage_mV': 3005, 'X-attrib-4': 5032, 'X-attrib-5': 234, 'X-attrib-6': 1, 'temperature_measurement': 2044, 'humidity_measurement': 6877, 'pressure_measurement': 100508, 'path': 42961, 'battery_level': 200}]
```

need to see the raw data, parsed result and state changed event

nerdosity, 2020-06-12:
> And I can't reproduce the battery problem
> [image]
> 
> Does it still report a negative value? 

Nope, I gave a look this morning.
Every sensor and device that had negative values yesterday now reports correct battery value, it seemed just an initial issue. After around one hour, battery value went > 0.

nerdosity, 2020-06-12:
[image]
I don't know if this is something quirk related or something else.

closed 2020-06-12 by dmulcahey

Adminiuga, 2020-06-12:
Dunno. What page is it from? Seems like either UI or translation error.

nerdosity, 2020-06-12:
> Dunno. What page is it from? Seems like either UI or translation error.

It's the Settings -> Devices -> Magic cube device page -> clicking "+" sign on automations panel -> popup

Adminiuga, 2020-06-12:
then must have been missing translations for device triggers/actions