zigpy/zha-device-handlers #379: [Device Support Request] Hive Active Heating Receiver (Computime) SLR2

peterguy04, 2020-06-16:
**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
Climate support 

**Device signature**
`[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0xc6e3 (00:1e:5e:09:02:03:c6:c2) joined the network
[0xc6e3:zdo] ZDO request ZDOCmd.Device_annce: [0xc6e3, 00:1e:5e:09:02:03:c6:c2, 142]
[0xc6e3] Requesting 'Node Descriptor'
Tries remaining: 2
[0xc6e3] Extending timeout for 0x2b request
Ignoring message (b'0000000100') on cluster 10: unknown endpoint or cluster id: 5
[0xc6e3] Node Descriptor: <Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4153 maximum_buffer_size=82 maximum_incoming_transfer_size=255 server_mask=0 maximum_outgoing_transfer_size=255 descriptor_capability_field=0>
[0xc6e3] Discovering endpoints
Tries remaining: 3
[0xc6e3] Discovered endpoints: [5, 6, 7, 8]
[0xc6e3:5] Discovering endpoint information
Tries remaining: 3
[0xc6e3:5] Discovered endpoint information: <Optional endpoint=5 profile=260 device_type=769 device_version=1 input_clusters=[0, 3, 9, 10, 513, 64768] output_clusters=[10, 1026, 25]>
[0xc6e3:5:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=49 command_id=Command.Read_Attributes_rsp>
[0xc6e3:5] Manufacturer: Computime
[0xc6e3:5] Model: SLR2
[0xc6e3:6] Discovering endpoint information
Tries remaining: 3
[0xc6e3:6] Discovered endpoint information: <Optional endpoint=6 profile=260 device_type=769 device_version=1 input_clusters=[513] output_clusters=[]>
[0xc6e3:7] Discovering endpoint information
Tries remaining: 3
[0xc6e3:7] Discovered endpoint information: <Optional endpoint=7 profile=260 device_type=769 device_version=1 input_clusters=[1026] output_clusters=[]>
[0xc6e3:8] Discovering endpoint information
Tries remaining: 3
[0xc6e3:8] Discovered endpoint information: <Optional endpoint=8 profile=260 device_type=769 device_version=1 input_clusters=[1026] output_clusters=[]>
Checking quirks for Computime SLR2 (00:1e:5e:09:02:03:c6:c2)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {8, 5, 6, 7}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {8, 5, 6, 7}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {8, 5, 6, 7}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {8, 5, 6, 7}
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {8, 5, 6, 7}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} {8, 5, 6, 7}
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {8, 5, 6, 7}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {8, 5, 6, 7}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {8, 5, 6, 7}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {8, 5, 6, 7}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {8, 5, 6, 7}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {8, 5, 6, 7}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because endpoint list mismatch: {1} {8, 5, 6, 7}
device - 0xc6e3:00:1e:5e:09:02:03:c6:c2 entering async_device_initialized - is_new_join: True
device - 0xc6e3:00:1e:5e:09:02:03:c6:c2 has joined the ZHA zigbee network
[0xc6e3](SLR2): started configuration
[0xc6e3:ZDO](SLR2): 'async_configure' stage succeeded
[0xc6e3:5:0x0201]: bound 'thermostat' cluster: Status.SUCCESS
[0xc6e3:5:0xfd00]: bound 'manufacturer_specific' cluster: Status.SUCCESS
[0xc6e3:5:0xfd00]: finished channel configuration
[0xc6e3:5:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0xc6e3:5:0x0000]: finished channel configuration
[0xc6e3:5:0x0201] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=63 command_id=Command.Configure_Reporting_rsp>
[... 302 lines cut ...]
        "0x0201"
      ],
      "out_clusters": []
    },
    "7": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0402"
      ],
      "out_clusters": []
    },
    "8": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0402"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "Computime",
  "model": "SLR2",
  "class": "zigpy.device.Device"
}`

Adminiuga, 2020-06-16:
Can you  get debug log, including `homeassistant.core: debug` component debug, after you trying  to  change the temperature setting + 60s after the attempt

peterguy04, 2020-06-16:
ok so i hope i have the right info for you,  so i ran what you asked on both entities though i know the hot water one doesnt need me to set a temp as thats just on and off, but i dont know which is which from the entities.

20:36 - 20:37 is the first entity and 20:38 - 20:39 is the second

` 'payload': {'devices': {'states': {'climate.computime_slr2_c2c60302_thermostat': {'online': True,
2020-06-16 20:36:31 DEBUG (MainThread) [zigpy.zcl] [0xe3cf:5:0x0019] OTA query_next_image handler for 'Computime SLR2': field_control=0, manufacture_id=4153, image_type=514, current_file_version=269043264, hardware_version=None
2020-06-16 20:36:54 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.computime_slr2_c2c60302_thermostat: Attribute 'system_mode' = SystemMode.Heat update
2020-06-16 20:37:43 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.computime_slr2_c2c60302_thermostat_2: Attribute 'system_mode' = SystemMode.Heat update
2020-06-16 20:37:52 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2937895568] Received {'type': 'call_service', 'domain': 'climate', 'service': 'set_temperature', 'service_data': {'temperature': 25.5, 'entity_id': 'climate.computime_slr2_c2c60302_thermostat'}, 'id': 65}
2020-06-16 20:37:52 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=climate, service=set_temperature, service_data=temperature=25.5, entity_id=climate.computime_slr2_c2c60302_thermostat>
2020-06-16 20:37:52 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=climate.computime_slr2_c2c60302_thermostat, old_state=<state climate.computime_slr2_c2c60302_thermostat=heat; hvac_modes=('off', 'heat'), min_temp=5.0, max_temp=32.0, current_temperature=21.0, temperature=1.0, hvac_action=off, system_mode=[4]/heat, occupied_cooling_setpoint=2600, occupied_heating_setpoint=100, friendly_name=Computime SLR2 c2c60302 thermostat, supported_features=1 @ 2020-06-16T20:15:53.849994+01:00>, new_state=<state climate.computime_slr2_c2c60302_thermostat=heat; hvac_modes=('off', 'heat'), min_temp=5.0, max_temp=32.0, current_temperature=21.0, temperature=25.5, hvac_action=off, system_mode=[4]/heat, occupied_cooling_setpoint=2600, occupied_heating_setpoint=2550, friendly_name=Computime SLR2 c2c60302 thermostat, supported_features=1 @ 2020-06-16T20:15:53.849994+01:00>>
2020-06-16 20:37:52 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=climate.computime_slr2_c2c60302_thermostat, old_state=<state climate.computime_slr2_c2c60302_thermostat=heat; hvac_modes=('off', 'heat'), min_temp=5.0, max_temp=32.0, current_temperature=21.0, temperature=1.0, hvac_action=off, system_mode=[4]/heat, occupied_cooling_setpoint=2600, occupied_heating_setpoint=100, friendly_name=Computime SLR2 c2c60302 thermostat, supported_features=1 @ 2020-06-16T20:15:53.849994+01:00>, new_state=<state climate.computime_slr2_c2c60302_thermostat=heat; hvac_modes=('off', 'heat'), min_temp=5.0, max_temp=32.0, current_temperature=21.0, temperature=25.5, hvac_action=off, system_mode=[4]/heat, occupied_cooling_setpoint=2600, occupied_heating_setpoint=2550, friendly_name=Computime SLR2 c2c60302 thermostat, supported_features=1 @ 2020-06-16T20:15:53.849994+01:00>>
2020-06-16 20:37:52 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=climate.computime_slr2_c2c60302_thermostat, old_state=<state climate.computime_slr2_c2c60302_thermostat=heat; hvac_modes=('off', 'heat'), min_temp=5.0, max_temp=32.0, current_temperature=21.0, temperature=1.0, hvac_action=off, system_mode=[4]/heat, occupied_cooling_setpoint=2600, occupied_heating_setpoint=100, friendly_name=Computime SLR2 c2c60302 thermostat, supported_features=1 @ 2020-06-16T20:15:53.849994+01:00>, new_state=<state climate.computime_slr2_c2c60302_thermostat=heat; hvac_modes=('off', 'heat'), min_temp=5.0, max_temp=32.0, current_temperature=21.0, temperature=25.5, hvac_action=off, system_mode=[4]/heat, occupied_cooling_setpoint=2600, occupied_heating_setpoint=2550, friendly_name=Computime SLR2 c2c60302 thermostat, supported_features=1 @ 2020-06-16T20:15:53.849994+01:00>>
2020-06-16 20:37:52 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=climate.computime_slr2_c2c60302_thermostat, old_state=<state climate.computime_slr2_c2c60302_thermostat=heat; hvac_modes=('off', 'heat'), min_temp=5.0, max_temp=32.0, current_temperature=21.0, temperature=1.0, hvac_action=off, system_mode=[4]/heat, occupied_cooling_setpoint=2600, occupied_heating_setpoint=100, friendly_name=Computime SLR2 c2c60302 thermostat, supported_features=1 @ 2020-06-16T20:15:53.849994+01:00>, new_state=<state climate.computime_slr2_c2c60302_thermostat=heat; hvac_modes=('off', 'heat'), min_temp=5.0, max_temp=32.0, current_temperature=21.0, temperature=25.5, hvac_action=off, system_mode=[4]/heat, occupied_cooling_setpoint=2600, occupied_heating_setpoint=2550, friendly_name=Computime SLR2 c2c60302 thermostat, supported_features=1 @ 2020-06-16T20:15:53.849994+01:00>>
2020-06-16 20:37:52 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=climate.computime_slr2_c2c60302_thermostat, old_state=<state climate.computime_slr2_c2c60302_thermostat=heat; hvac_modes=('off', 'heat'), min_temp=5.0, max_temp=32.0, current_temperature=21.0, temperature=1.0, hvac_action=off, system_mode=[4]/heat, occupied_cooling_setpoint=2600, occ [... cut ...]
2020-06-16 20:39:13 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.2937895568] Received {'type': 'call_service', 'domain': 'climate', 'service': 'set_temperature', 'service_data': {'temperature': 25.5, 'entity_id': 'climate.computime_slr2_c2c60302_thermostat_2'}, 'id': 66}
2020-06-16 20:39:13 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=climate, service=set_temperature, service_data=temperature=25.5, entity_id=climate.computime_slr2_c2c60302_thermostat_2>`

Adminiuga, 2020-06-16:
Hrm,
1. I meant I need `homeassistant.core` debug in addition to all zigpy and radio debug setup per https://www.home-assistant.io/integrations/zha/#debug-logging
2. when posting logs, post it between opening and closing triple ticks -- \`\`\`

peterguy04, 2020-06-16:
ok my bad, pretty new to all this so learning as i go, i hope i have got it right this time

```
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e61000700aa00
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1761000800010001
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17610027002000220200000102795b09040102070900084b0a00042a52000000afff07d42000cc
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [32, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x5b79>, 9, 260, 1794, b'\x08K\n\x00\x04*R\x00\x00', 0, 175, 255, 7, 212, 32, 0, -52]
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy.zcl] [0x5b79:9:0x0702] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=75 command_id=Command.Report_Attributes>
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy.zcl] [0x5b79:9:0x0702] ZCL request 0x000a: [[<Attribute attrid=1024 value=<TypeValue type=int24s, value=82>>]]
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy.zcl] [0x5b79:9:0x0702] Attribute report received: instantaneous_demand=82
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 75 under 119 request id, data: b'184b0b0a00'
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x5b79>, ep: 9, profile: 0x0104, cluster_id: 0x0702, data: b'084b0a00042a520000'
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 119, 0, <DeconzAddressEndpoint address_mode=2 address=0x5b79 endpoint=9>, 260, 1794, 1, b'\x18K\x0b\n\x00', 2, 0)
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1262001b001400770002795b0904010207010500184b0b0a000200
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x126200090002002277
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 119]
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e63000700a600
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04630007000000
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04630013000c00227702795b09010000000000
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 119: 00
2020-06-16 22:26:01 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x77 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x5b79 endpoint=9>, status: 0x00
2020-06-16 22:26:02 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=climate.computime_slr2_c2c60302_thermostat, old_state=<state climate.computime_slr2_c2c60302_thermostat=heat; hvac_modes=['off', 'heat'], min_temp=5, max_temp=32, current_temperature=21, temperature=2, hvac_action=off, system_mode=[4]/heat, occupied_cooling_setpoint=2600, occupied_heating_setpoint=100, friendly_name=Computime SLR2 c2c60302 thermostat, supported_features=1 @ 2020-06-16T22:19:48.630013+01:00>, new_state=<state climate.computime_slr2_c2c60302_thermostat=heat; hvac_modes=['off', 'heat'], min_temp=5, max_temp=32, current_temperature=21, temperature=25, hvac_action=off, system_mode=[4]/heat, occupied_cooling_setpoint=2600, occupied_heating_setpoint=100, friendly_name=Computime SLR2 c2c60302 thermostat, supported_features=1 @ 2020-06-16T22:19:48.630013+01:00>>
2020-06-16 22:26:05 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e64000700aa00
2020-06-16 22:26:05 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-06-16 22:26:05 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-06-16 22:26:05 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1764000800010001
2020-06-16 22:26:05 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x1764002a002300220200000102fb2201040119000c00014e0100681101010200481100afd707d42000bd
2020-06-16 22:26:05 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [35, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x22fb>, 1, 260, 25, b'\x01N\x01\x00h\x11\x01\x01\x02\x00H\x11', 0, 175, 215, 7, 212, 32, 0, -67]
[... cut ...]
2020-06-16 22:28:02 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04ca0007000000
2020-06-16 22:28:02 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04ca0013000c00229702795b09010000000000
2020-06-16 22:28:02 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 151: 00
2020-06-16 22:28:02 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x97 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x5b79 endpoint=9>, status: 0x00
```

closed 2020-07-17 by peterguy04