zigpy/zha-device-handlers #380: [Device Support Request] Add support for updated Legrand Dimmer switch w/o neutral

pdecat, 2020-06-17:
**Is your feature request related to a problem? Please describe.**

I've updated the firmware of my Legrand Dimmer switch w/o neutral for which support was added in https://github.com/zigpy/zha-device-handlers/issues/299

Before OTA upgrade:
- app_version: 0
- hw_version: 1
- stack_version: 64
- sw_build_id: 01a (26)
- zcl_version: 2
- Firmware: 0x03401a00

After OTA upgrade (2020-06-08):
- app_version: 0
- hw_version: 6
- stack_version: 66
- sw_build_id: 02b (43)
- zcl_version: 2
- Firmware: 0x002b4203

And now it reports a new `GreenPowerProxy` endpoint with id 242:

```
{
  "node_descriptor": "<NodeDescriptor byte1=17 byte2=64 mac_capability_flags=142 manufacturer_code=4129 maximum_buffer_size=89 maximum_incoming_transfer_size=63 server_mask=10752 maximum_outgoing_transfer_size=63 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000f",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0000",
        "0x0019",
        "0xfc01"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Dimmer switch w/o neutral",
  "class": "zigpy.device.Device"
}
```

The issue is that prevents the quirk from matching:

```
2020-06-17 06:45:05 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for  Legrand  Dimmer switch w/o neutral (00:04:74:00:00:8b:0e:a2)
2020-06-17 06:45:05 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.legrand.dimmer.DimmerWithoutNeutral'>
2020-06-17 06:45:05 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 242}
```

**Describe the solution you'd like**

Could the quirk be updated to also support new firmwares?

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

TODO

**Additional context**

[image]

dmulcahey, 2020-06-29:
@pdecat wanna test #392

closed 2020-06-29 by Adminiuga

pdecat, 2020-06-29:
Thanks @dmulcahey, it works!

```
2020-06-29 23:01:59 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for  Legrand  Dimmer switch w/o neutral (00:04:74:00:00:8b:0e:a2)
2020-06-29 23:01:59 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.legrand.dimmer.DimmerWithoutNeutral'>
2020-06-29 23:01:59 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 242}
2020-06-29 23:01:59 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.legrand.dimmer.DimmerWithoutNeutral2'>
2020-06-29 23:01:59 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 00:04:74:00:00:8b:0e:a2: <class 'zhaquirks.legrand.dimmer.DimmerWithoutNeutral2'>

```

PR #392 (2020-06-29): Alternate legrand dimmer implementation
Fixes #380