zigpy/zha-device-handlers #383: [BUG] Heiman Vibration Sensor Issues

blair287, 2020-06-19:
Sorry if this isnt a bug but a device support request im not sure which as it works just with a weird affect.

It detects vibrations perfectly but the weird bit is upon detecting it reported detected which is correct but then it goes:

Clear - no vibration
Detected - Vibrations applied to sensor
Clear
Detected
Clear
Detected 
Clear

for some reason is reported detected 3 times before reporting clear properly when no vobration is detected. 

The zigbee info:

`{
  "node_descriptor": "<NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4619 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "HEIMAN",
  "model": "Vibration-EF_3.0",
  "class": "zigpy.device.Device"
}`

dmulcahey, 2020-06-19:
Have a link to the device on Amazon or something similar?

blair287, 2020-06-19:
> Have a link to the device on Amazon or something similar?

https://www.aliexpress.com/item/32953268039.html?spm=a2g0o.productlist.0.0.10284155NBVvyM&algo_pvid=8ddad9d6-497f-4a81-8563-3b0383e01409&algo_expid=8ddad9d6-497f-4a81-8563-3b0383e01409-4&btsid=0b0a050b15925951043055741e68df&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

bought their temp and humidity zigbee sensor too that works ok zigbee wise but its reporting a 2c higher temp than all my other sensors and also very slow updates takes ages from temp change to update thinking their stuff might just be crap.

closed 2021-08-13 by github-actions[bot]