zigpy/zha-device-handlers #384: [Device Support Request] open/close remote for IKEA FYRTUR not pairing correctly

APayerl, 2020-06-23:
**Is your feature request related to a problem? Please describe.**
I'm unable to create automations using the button that accompanies the IKEA Fyrtur blind. No buttons seems to exist when paired using ZHA.
[image]

**Describe the solution you'd like**
In deconz this device works. It's then possible to create automations using single and long clicks on open and close buttons.
[image]

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

> [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
> Device 0xa037 (ec:1b:bd:ff:fe:b4:66:63) joined the network
> [0xa037:zdo] ZDO request ZDOCmd.Device_annce: [0xa037, ec:1b:bd:ff:fe:b4:66:63, 128]
> [0xa037] Requesting 'Node Descriptor'
> Tries remaining: 2
> [0xa037] Extending timeout for 0x08 request
> [0xa037] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4476 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
> [0xa037] Discovering endpoints
> Tries remaining: 3
> [0xa037] Extending timeout for 0x0a request
> [0xa037] Discovered endpoints: [1]
> [0xa037:1] Discovering endpoint information
> Tries remaining: 3
> [0xa037] Extending timeout for 0x0c request
> [0xa037:1] Discovered endpoint information: <Optional endpoint=1 profile=260 device_type=515 device_version=1 input_clusters=[0, 1, 3, 9, 32, 4096, 64636] output_clusters=[3, 4, 6, 8, 25, 258, 4096]>
> [0xa037] Extending timeout for 0x0e request
> [0xa037:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=14 command_id=Command.Read_Attributes_rsp>
> [0xa037:1] Manufacturer: KE
> [0xa037:1] Model: TRADFRI open/close remote
> Checking quirks for KE TRADFRI open/close remote (ec:1b:bd:ff:fe:b4:66:63)
> Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
> Fail because device_type mismatch on at least one endpoint
> Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
> Fail because endpoint list mismatch: {11, 13} {1}
> Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
> Fail because profile_id mismatch on at least one endpoint
> Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
> Fail because device_type mismatch on at least one endpoint
> Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
> Fail because endpoint list mismatch: {3} {1}
> Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
> Fail because device_type mismatch on at least one endpoint
> Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
> Fail because endpoint list mismatch: {1, 2} {1}
> Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
> Fail because endpoint list mismatch: {1, 2} {1}
> Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
> Fail because device_type mismatch on at least one endpoint
> Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
> Fail because device_type mismatch on at least one endpoint
> Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
> Fail because endpoint list mismatch: {232, 230} {1}
> Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
> Fail because endpoint list mismatch: {232, 230} {1}
> Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
> Fail because device_type mismatch on at least one endpoint
> device - 0xa037:ec:1b:bd:ff:fe:b4:66:63 entering async_device_initialized - is_new_join: True
> device - 0xa037:ec:1b:bd:ff:fe:b4:66:63 has joined the ZHA zigbee network
> [0xa037](TRADFRI open/close remote): started configuration
> [0xa037:ZDO](TRADFRI open/close remote): 'async_configure' stage succeeded
> [0xa037] Extending timeout for 0x10 request
> [0xa037] Extending timeout for 0x12 request
> [0xa037] Extending timeout for 0x14 request
> [0xa037:1:0x0003] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=0 command_id=1>
> [0xa037:1:0x0003] ZCL request 0x0001: []
> [0xa037:1:0x0003] No handler for cluster command 1
> [0xa037:1:0x0003]: received 'identify_query' command with [] args on cluster_id '3' tsn '0'
> [0xa037:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
> [0xa037:1:0x0000]: finished channel configuration
> [0xa037] Extending timeout for 0x16 request
[... 57 lines cut ...]
> [0xa037:1:0x0008]: initializing channel: from_cache: False
> [0xa037:1:0x0006]: initializing channel: from_cache: False
> [0xa037:1:0x0019]: initializing channel: from_cache: False
> [0xa037:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=46 command_id=Command.Read_Attributes_rsp>
> [0xa037:1:0x0001]: initializing channel: from_cache: False
> [0xa037] Extending timeout for 0x32 request
> [0xa037:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=48 command_id=Command.Read_Attributes_rsp>
> [0xa037:1:0x0006]: initializing channel: from_cache: False
> [0xa037] Extending timeout for 0x34 request
> [0xa037:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=50 command_id=Command.Read_Attributes_rsp>
> [0xa037:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=52 command_id=Command.Read_Attributes_rsp>
> [0xa037:1:0x0020]: 'async_initialize' stage succeeded
> [0xa037:1:0x0000]: 'async_initialize' stage succeeded
> [0xa037:1:0x1000]: 'async_initialize' stage succeeded
> [0xa037:1:0x0001]: 'async_initialize' stage succeeded
> [0xa037:1:0x0006]: 'async_initialize' stage succeeded
> [0xa037:1:0x0102]: 'async_initialize' stage succeeded
> [0xa037:1:0x0008]: 'async_initialize' stage succeeded
> [0xa037:1:0x0006]: 'async_initialize' stage succeeded
> [0xa037:1:0x0019]: 'async_initialize' stage succeeded
> [0xa037](TRADFRI open/close remote): power source: Battery or Unknown
> [0xa037](TRADFRI open/close remote): completed initialization

**Additional context**
I'm using Hassio 0.111.4 and ZHA quirks 0.0.40 since HA 0.111.3

dmulcahey, 2020-06-29:
It is definitely related to #269 Can you get me the events for all 4 actions? subscribe to zha_event in the event tool and get me press and hold events for both buttons and I'll finish up this PR to get support added.

APayerl, 2020-06-29:
I can fix it Thursday earliest but I'll get it done! Thanks! 🙂

APayerl, 2020-07-03:
Okay I have done as you asked.
The "down fast click" zha event:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "68:0a:e2:ff:fe:7a:b3:16",
        "unique_id": "68:0a:e2:ff:fe:7a:b3:16:1:0x0102",
        "endpoint_id": 1,
        "cluster_id": 258,
        "command": "down_close",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-07-03T09:06:14.309052+00:00",
    "context": {
        "id": "f32fc0e0f4bc446798e954fab47adf92",
        "parent_id": null,
        "user_id": null
    }
}
```
The "up fast click" zha event:
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "68:0a:e2:ff:fe:7a:b3:16",
        "unique_id": "68:0a:e2:ff:fe:7a:b3:16:1:0x0102",
        "endpoint_id": 1,
        "cluster_id": 258,
        "command": "up_open",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-07-03T09:06:15.773233+00:00",
    "context": {
        "id": "751e004b27aa4dd5aa4c5a177e72bcbf",
        "parent_id": null,
        "user_id": null
    }
}
```
The "long click" actions actually send two actions, one on down press and one on release:

The "down long click" zha event(s):
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "68:0a:e2:ff:fe:7a:b3:16",
        "unique_id": "68:0a:e2:ff:fe:7a:b3:16:1:0x0102",
        "endpoint_id": 1,
        "cluster_id": 258,
        "command": "down_close",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-07-03T09:06:16.606375+00:00",
    "context": {
        "id": "4f3b8ea0643a4db191285dc841653ec2",
        "parent_id": null,
        "user_id": null
    }
}
```
and then 
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "68:0a:e2:ff:fe:7a:b3:16",
        "unique_id": "68:0a:e2:ff:fe:7a:b3:16:1:0x0102",
        "endpoint_id": 1,
        "cluster_id": 258,
        "command": "stop",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-07-03T09:06:18.343554+00:00",
    "context": {
        "id": "d71427cc226d40b9af32df413e54aae3",
        "parent_id": null,
        "user_id": null
    }
}
```
The "up long click" zha event(s):
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "68:0a:e2:ff:fe:7a:b3:16",
        "unique_id": "68:0a:e2:ff:fe:7a:b3:16:1:0x0102",
        "endpoint_id": 1,
        "cluster_id": 258,
        "command": "up_open",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-07-03T09:06:19.655935+00:00",
    "context": {
        "id": "991b65d3e04e42eb813708e4bcff9771",
        "parent_id": null,
        "user_id": null
    }
}
```
then
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "68:0a:e2:ff:fe:7a:b3:16",
        "unique_id": "68:0a:e2:ff:fe:7a:b3:16:1:0x0102",
        "endpoint_id": 1,
        "cluster_id": 258,
        "command": "stop",
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-07-03T09:06:21.622012+00:00",
    "context": {
        "id": "3b3f934b85364423b874d248b8e4acca",
        "parent_id": null,
        "user_id": null
    }
}
```

Hope this helps! :)

dmulcahey, 2020-07-03:
OK, I fixed the formatting on #269 and merged it. The device triggers will need custom code to handle long press events since both buttons fire the same event. I don't have this remote to test it. I'll see if I can pick one up.

APayerl, 2020-07-03:
okay! If you want (and its feasible) I can test for you, but maybe the development speed would be too slow?

APayerl, 2020-08-10:
You spoke of custom code, custom in the core of zha-handlers or custom for the custom device?
I might be able to try to code and test some if you could give me a push, like is there already some devices that has a similar custom code? What kind of change would be needed?

APayerl, 2020-08-10:
@dmulcahey 
You spoke of custom code, custom in the core of zha-handlers or custom for the custom device?
I might be able to try to code and test some if you could give me a push, like is there already some devices that has a similar custom code? What kind of change would be needed?

APayerl, 2020-08-11:
The remote that accompanies the blind is **Type: E1766 Open/Close remote**. It's the same for both Kadrilj and Fyrtur.

Adminiuga, 2020-10-02:
@APayerl if it doesn't work as expected in the latest release, please re-open this issue.

closed 2020-10-02 by Adminiuga

APayerl, 2020-10-30:
@Adminiuga Do I have to do something else then test with latest version of home assistant? Or do I need to change the instances zha-device-handler code for ikea devices?...

Adminiuga, 2020-10-30:
Use these triggers
[image]

Different integration, different trigger names 🤷

Adminiuga, 2020-10-30:
or with yaml you could use triggers based on `zha_event` , which may offer more flexibility.

APayerl, 2020-10-31:
but how would one differentiate between long click or "fast" click?

APayerl, 2020-11-02:
@Adminiuga I have tried to create a automation based on what you wrote without success.

APayerl, 2021-01-22:
@dmulcahey @Adminiuga This issue is still not resolved but I'm unable to reopen the issue (as adminiuga wrote) for some reason...

APayerl, 2021-02-04:
I tested this once again today and someone has done something great because now it works!