zigpy/zha-device-handlers #387: [Device Support Request] Philips Hue Outdoor Sensor SML002

voskuh123, 2020-06-27:
**Is your feature request related to a problem? Please describe.**
The sensor is working fine. Values for motion detection, temperature, battery status, illuminance are updated and shown as expected. But some settings are missing. There is a sensitivity setting which can't be set. 

**Describe the solution you'd like**
A quirk that implements the sensitivity setting? The sensitivy setting is located on cluster 0x0406 attribute 0x0030. For the outdoor sensor the sensitivy can be set between 0 and 4. There is also a sensititivy max setting (cluster 0x0406 attribute 0x0031) which is readonly. That represents the maximum sensitivy level (for the outdoor 4). 
for more information see https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Philips-Hue-sniffing#hue-motion-sensor

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be `  {
"node_descriptor": "<NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4107 maximum_buffer_size=89 maximum_incoming_transfer_size=63 server_mask=0 maximum_outgoing_transfer_size=63 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 49246,
      "device_type": "0x0850",
      "in_clusters": [
        "0x0000"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0400",
        "0x0402",
        "0x0406"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Philips",
  "model": "SML002",
  "class": "zigpy.device.Device"
}
`

**Additional context**
Add any other context or screenshots about the feature request here.

dmulcahey, 2020-06-29:
Please provide the other required information and I'll get this implemented. I need the device signature. There is a reason we ask for this stuff in the templates :) Once you provide this I'll get a quirk up for the device!

dmulcahey, 2020-06-29:
Wanna test #391 when you have a moment? Sorry for the bad comment above. Hadn't had my coffee yet and completely missed the signature!

closed 2020-06-29 by Adminiuga

voskuh123, 2020-06-29:
Thank you for your fast response and sorry about the inconvenience about the code layout. 
I tested the quirk but had some trouble to get it to work. 
At first the quirk wouldn't load properly. 
`2020-06-29 22:29:08 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry ConBee II, s/n: DE2216114 - dresden elektronik ingenieurtechnik GmbH for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 217, in async_setup
    hass, self
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 97, in async_setup_entry
    import zhaquirks  # noqa: F401 pylint: disable=unused-import, import-outside-toplevel, import-error
  File "/usr/local/lib/python3.7/site-packages/zhaquirks/__init__.py", line 188, in <module>
    importlib.import_module(modname)
  File "/usr/local/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/usr/local/lib/python3.7/site-packages/zhaquirks/philips/sml002.py", line 22, in <module>
    from . import PHILLIPS
ImportError: cannot import name 'PHILLIPS' from 'zhaquirks.philips' (/usr/local/lib/python3.7/site-packages/zhaquirks/philips/__init__.py)`

After removing line 22, there was another problem with line 45. I had to remove that line as well. 
Then, i could pair my device. When I want to read the sensitivity and sensitivity_max settings it doesn't work out of the box. I had to override the manufacturer to 4107 (0x100B). Then I can read and change the settings. 

`2020-06-29 22:18:53 DEBUG (MainThread) [homeassistant.components.zha.api] Set attribute for: cluster_id: [1030] cluster_type: [in] endpoint_id: [2] attribute: [48] value: [0] manufacturer: [4107] response: [[[<WriteAttributesStatusRecord status=Status.SUCCESS>]]]`

In addition, maybe it's good to mention that in the quirk there are some typo's. It is philips instead of phillips. Don't know if thats a problem.

dmulcahey, 2020-06-29:
how did you apply this... aside from the spelling it should work. Did you miss the constant in init.py in the same directory?

voskuh123, 2020-06-29:
Thank you for your response. I'm sorry, my bad. I missed the changes in init.py. Now, it loads as expected and no problems with pairing and set up. 

To read out the settings and change them I have to override the manufacturer to 4107 (0x100B). When leaving this out I can't read  and set the attribute.

dmulcahey, 2020-06-29:
That's expected as it is a mfg specific attribute. I believe we are contemplating a solution for this

voskuh123, 2020-06-29:
Okay, great. Thanks for your help.

Adminiuga, 2020-06-29:
Have a solution in mind and was planning it for the next version, but then it is going to be 0.113 and maybe shouldn't do big changes :D

dmulcahey, 2020-06-29:
Because of 13? Or am I missing something

Adminiuga, 2020-06-29:
yeah, had way too many issues with `zigpy==0.13.0`

PR #391 (2020-06-29): Add quirk for Phillips SML002
Fixes #387