zigpy/zha-device-handlers #394: [Device Support Request] Xiaomi Aqara D1 double key wireless wall switch WXKG07LM

WarMachine228, 2020-06-30:
Please provide support for the enw wireless switch.  It can pair but there is no quirk.
Z2m page: [https://www.zigbee2mqtt.io/devices/WXKG07LM.html]
Thanks

Adminiuga, 2020-06-30:
Do you get any zha_event ? 
Can you provide information requested in the issue template?

Adminiuga, 2020-06-30:
Per se having a quirk is a not strict requirement for device to work with ZHA

WarMachine228, 2020-07-01:
No zha event.
No triggers when creating automation in gui.
[image]
I
 will post the log file later

WarMachine228, 2020-07-01:
This is the log of the pairing
[zha.txt](https://github.com/zigpy/zha-device-handlers/files/4856672/zha.txt)

dmulcahey, 2020-07-05:
Have one coming on the slow boat... will see what’s up when I get it.

egelmex, 2020-08-06:
I have just tried to pair a d1 three way switch, 

`{
  "node_descriptor": "<Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4447 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=11264 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.n3acn3",
  "class": "zigpy.device.Device"
}`

sshaikh, 2020-09-09:
I have a similar request re QBKG25LM aka lumi.switch.l3acn3. It's supported by z2m, so I presume it's a quirk issue. One button works, the others don't.

```
{
  "node_descriptor": "<Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4447 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=11264 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "out_clusters": []
    },
    "41": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "42": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "43": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "51": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "52": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "53": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.l3acn3",
  "class": "zigpy.device.Device"
}
```

Adminiuga, 2020-09-09:
the problem is with `zigpy-cc`. If your stick is running ZStack 3.x, then switch to zigpy-znp in the new HA beta and it should take care of it.

sshaikh, 2020-09-09:
Thanks for the pointer. 

I'm using the usb stick as per https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html. So presume I'm running an old stack.

There are details here on the 3.0 stack: https://github.com/Koenkk/zigbee2mqtt/issues/1445

sshaikh, 2020-09-09:
@Adminiuga just to clarify, do I need zstack 3 to use this device at all, or to most easily use this device in HA natively?

I ask as it's unclear whether the cc2531 with v3 will work with zigpy-znp so I'm trying to figure out the best way to proceed.

Adminiuga, 2020-09-09:
CC2531 with ZStack 3.x would work with `zigpy-znp`

sshaikh, 2020-09-10:
Updating to ZStack 3, switching to beta and readding all my devices did the trick.

dmulcahey, 2020-10-10:
closing as these are supported

closed 2020-10-10 by dmulcahey