zigpy/zha-device-handlers #397: [Device Support Request] power measurement / consumption for lumi.plug

Draluak, 2020-07-06:
When adding chinese versions of xiaomi smart socket (lumi.plug) no sensors for current/total consumption are created.
In the cluster i can see the endpoints and read the attribute present_value, so it should be there generally.

Signature:
`{
  "node_descriptor": "<Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4447 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=0 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0010"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x0004",
        "0x000c"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0053",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x000c"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.plug",
  "class": "zigpy.device.Device"
}`

Pairing Log:
`[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x36cf:1:0x0b04]: async_update
[0xac0f:1:0x0b04]: async_update
[0x1396:1:0x0b04]: async_update
[0x11ef:1:0x0b04]: async_update
[0x0f94:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=142 command_id=Command.Read_Attributes>
[0x0f94:1:0x000a] ZCL request 0x0000: [[0]]
Device 0x288b (00:15:8d:00:03:93:89:8a) joined the network
Device 00:15:8d:00:03:93:89:8a changed id (0xb867 => 0x288b)
[0x288b:zdo] ZDO request ZDOCmd.Device_annce: [0x288b, 00:15:8d:00:03:93:89:8a, 142]
Checking quirks for LUMI lumi.plug (00:15:8d:00:03:93:89:8a)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 2, 3}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 2, 3}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
[... 52 lines cut ...]
[0x288b:1:0x0006] Attribute report received: on_off=0, 61440=50331392
[0x288b:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=15 command_id=Command.Read_Attributes>
[0x288b:1:0x000a] ZCL request 0x0000: [[0]]
[10379:1:0x0019]: Failed to bind 'ota' cluster: 
[10379:1:0x0019]: finished channel configuration
[10379:1:0x0006]: 'async_configure' stage succeeded
[10379:1:0x0000]: 'async_configure' stage succeeded
[10379:1:0x0019]: 'async_configure' stage succeeded
[10379](lumi.plug): completed configuration
[10379](lumi.plug): stored in registry: ZhaDeviceEntry(name='LUMI lumi.plug', ieee='00:15:8d:00:03:93:89:8a', last_seen=1594052245.6488373)
[0x288b:1:0x0003] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=171 command_id=Command.Default_Response>
[10379:1:0x0003]: executed 'trigger_effect' command with args: '(2, 0)' kwargs: '{}' result: [64, <Status.UNSUP_CLUSTER_COMMAND: 129>]
[10379](lumi.plug): started initialization
[10379:ZDO](lumi.plug): 'async_initialize' stage succeeded
[10379:1:0x0019]: initializing channel: from_cache: False
[0x288b:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=173 command_id=Command.Read_Attributes_rsp>
[10379:1:0x0006]: initializing channel: from_cache: False
[0x288b:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=175 command_id=Command.Read_Attributes_rsp>
[10379:1:0x0000]: initializing channel: from_cache: False
[0x288b:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=177 command_id=Command.Read_Attributes_rsp>
[10379:1:0x0006]: 'async_initialize' stage succeeded
[10379:1:0x0000]: 'async_initialize' stage succeeded
[10379:1:0x0019]: 'async_initialize' stage succeeded
[10379](lumi.plug): power source: Mains
[10379](lumi.plug): completed initialization`

Adminiuga, 2020-07-06:
I'm loosing all hope to understand why Xiaomi just won't use the standard clusters. 
This would need something like https://github.com/zigpy/zha-device-handlers/pull/283

Kanishkaz, 2020-07-27:
Hi,
I have this same issue with my Xiaomi Zigbee plugs since HA v0.111. Has there been any progress on it? 
Thanks.

dmulcahey, 2020-08-09:
Fixed by #434

closed 2020-08-09 by dmulcahey

thimic, 2020-08-09:
This might close https://github.com/home-assistant/core/issues/37411 too

Draluak, 2020-08-12:
Happy to see some progess here. Power measurement sensor is now available for me in HA 0.114. 
Nevertheless, as this feature request also adresses the (still) missing "total consumption" sensor, i would suggest to reopen it.

Hedda, 2021-10-06:
FYI, it looks like support for more Zigbee clusters attributes is now being worked on for [ZHA integration](https://www.home-assistant.io/integrations/zha) in Home Assistant core, see:

https://github.com/home-assistant/core/pull/56909

and

https://github.com/home-assistant/core/pull/56666

Also, note the last comment from Adminiuga in PR for ZHA in Home Assistant core -> https://github.com/home-assistant/core/pull/56909

"**_Support for other attributes could be extended later, as I couldn't find too many devices other than Sinopé RM3250ZB Electrical Load Controller – 50A – Zigbee https://www.amazon.com/dp/B07CGJ6XH2/ reporting anything interesting on the electrical measurement cluster._**"

So maybe now you could try to get this working with the latest Home Assistant beta or wait for the next upcoming release:

https://www.home-assistant.io/faq/release/

PS: There is much more discussion about ZHA energy cluster and attributes for smart energy metering here -> https://github.com/zigpy/zha-device-handlers/issues/605 and https://github.com/zigpy/zha-device-handlers/issues/1045