zigpy/zha-device-handlers #401: [Device Support Request] Hue Dimmer Switch RWL021

tigattack, 2020-07-08:
**Is your feature request related to a problem? Please describe.**
ZHA doesn't report the click count for Hue Dimmer Switch button presses.

**Describe the solution you'd like**
I would like the click count to be reported as part of the event data sent when a button is pressed on the Hue Dimmer Switch.   
e.g.
```
{
    "event_type": "zha_event",
    "data": {
        "device_ieee": "ieee_here",
        "unique_id": "unique_id_here",
        "endpoint_id": 1,
        "cluster_id": 6,
        "command": "on",
        "click_count": "1",
# Like this ^
        "args": []
    },
    "origin": "LOCAL",
    "time_fired": "2020-07-08T14:27:48.530416+00:00",
    "context": {
        "id": "id_here",
        "parent_id": null,
        "user_id": null
    }
}
```
Or similar.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
{
  "node_descriptor": "<NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4107 maximum_buffer_size=71 maximum_incoming_transfer_size=45 server_mask=0 maximum_outgoing_transfer_size=45 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 49246,
      "device_type": "0x0830",
      "in_clusters": [
        "0x0000"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x000c",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0xfc00"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Philips",
  "model": "RWL021",
  "class": "zhaquirks.philips.rwl021.PhilipsRWL021"
}
```

**Additional context**
zigbee2mqtt appears to support this and it's mentioned that `click_count` is supported by this device here: https://zigbee.blakadder.com/Philips_RWL021.html

dmulcahey, 2020-07-08:
Click count is done in software not by the device. We may support this in the future but currently we do not.

tigattack, 2020-07-08:
It would be immensely useful to many people, I'm sure. Thanks 😃

closed 2020-11-06 by Adminiuga