zigpy/zha-device-handlers #403: [BUG]

tigattack, 2020-07-10:
**Describe the bug**
Hue Dimmer Switch RWL021 battery level is not reported.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair a Philips Hue Dimmer switch
2. Go to its device page
4. See the unavailable battery level entity.

**Expected behavior**
I would expect to see the battery level entity for the switch available and reporting the level.

**Screenshots**
[image]

**Additional context**
I'm not sure if it's any help but there's data in the OP [here](https://community.smartthings.com/t/alpha-hue-dimmer-switch-zha-beta-no-hue-bridge/62874) regarding reading the battery status.

RWL021 device signature:
```txt
{
  "node_descriptor": "<NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4107 maximum_buffer_size=89 maximum_incoming_transfer_size=63 server_mask=0 maximum_outgoing_transfer_size=63 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 49246,
      "device_type": "0x0830",
      "in_clusters": [
        "0x0000"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x000c",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0xfc00"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Philips",
  "model": "RWL021",
  "class": "zhaquirks.philips.rwl021.PhilipsRWL021"
}
```

dmulcahey, 2020-07-10:
Him, mine works. Can you go to the manage clusters UI, select the PowerConfiguration cluster and then select the battery_percentage_remaining attribute and then click get zigbee attribute. you will have to press a button on the remote to wake it up so it responds. here is mine: 

[image]

battery % is out of 200 so that equates to 100%:

[image]

I have the RWL020 but I think they work the same...

tigattack, 2020-07-12:
Weirdly this doesn't work for me. Whenever I attempt to get the attribute, nothing happens and within a minute a websocket timeout error is shown in the log.
[image]

dmulcahey, 2020-07-12:
Are you pressing a button on the remote to wake it up? And do you see the event from the button when you do it?

tigattack, 2020-07-12:
Yes to both.

dmulcahey, 2020-07-12:
Whelp that doesn’t make a lot of sense. Enable debug logging and post the logs of the interaction

tigattack, 2020-07-12:
This looks to be the entire ZHA-related part of the log from button-press and click of "get zigbee attribute" to the websocket timeout error - https://pastebin.com/TQK6vYic

dmulcahey, 2020-07-13:
Please follow the instructions here: https://www.home-assistant.io/integrations/zha/#debug-logging and submit all of the logs not just the zigpy_cc stuff. Make sure it includes ZHA, zigpy and zigpy_cc

tigattack, 2020-07-13:
That's what I followed 🙂 The logs appear to include output from zigpy, zigpy_cc, and homeassistant.components.zha. If there's anything missing I can do it again and paste the entire output, but it was huge before, so I removed log entries for unrelated actions (such as live updates from environment sensors and media players).

tigattack, 2020-07-20:
Anyone have any clues on this one?

tigattack, 2020-07-22:
I removed and re-added the device due to a change in Home Assistant 0.113 and saw some interesting log outputs, though understanding most of it is beyond me.  
Thought it could be of use to this issue though - https://pastebin.com/SwVgXzbQ

dmulcahey, 2020-10-10:
these are bind failures which prevents the clusters from communicating with Zigpy / ZHA. remove it and re pair it while occasionally pressing a button on it to see if it is maybe falling asleep when you pair it. I have not seen this myself w/ the Philips remotes... maybe try changing the battery too. I am going to close this because it is not a quirks related issue.

closed 2020-10-10 by dmulcahey