zigpy/zha-device-handlers #408: [Device Support Request] Ecolink 4655BC0-R Door Sensor

Gamerayers, 2020-07-13:
Hoping for some help. Just got a set of 10 Ecolink 4655BC0-R door/window sensors with temporary monitoring. Looks like the sort from the isn’t exactly correctly and Home Assistants (ZHA) isn’t polling it correctly or something. Attached is the log from it being added to the zha network. There is an area where it is trying to consider it different sensor types, and fails since it doesn’t know how to class it. Ultimately, it shows up as a door, battery and temperature sensor. The battery is a ?, the door registers as closed always and the temperature doesn’t seem to update.

Going into manage cluster, selecting IasZone and then Zone Status gets it to update if the door is open or closed, but it doesn’t seem to do this automatically. Additionally, monitoring zha events doesn’t show it sending signals. But if I pull the battery and reinsert it, I have to readd it to the network, so I assume it isn’t fully joined correctly. I emailed Ecolink and they say they only test and officially support Smartthings. I have tried to pair to my Abode hub and it reports only as a temperature sensor. Is there any way to get this sensor setup as a class so that zha recognizes?

`[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0x0000](ConBee II): started initialization
[0x0000:ZDO](ConBee II): 'async_initialize' stage succeeded
[0x0000](ConBee II): power source: Mains
[0x0000](ConBee II): completed initialization
[0x8209:1:0x0020] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=64 command_id=0>
[0x8209:1:0x0020] ZCL request 0x0000: []
[0x8209:1:0x0020] No handler for cluster command 0
[0x8209:1:0x0020]: Received 64 tsn command 'checkin': []
[0x8209] Extending timeout for 0x40 request
[0x8209:1:0x0020] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=64 command_id=Command.Default_Response>
[0x8209:1:0x0020]: executed 'checkin_response' command with args: '(True, 8)' kwargs: '{'tsn': 64}' result: [0, <Status.SUCCESS: 0>]
[0x8209] Extending timeout for 0xea request
[0x8209:1:0x0020] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=234 command_id=Command.Default_Response>
[0x8209:1:0x0020]: executed 'set_long_poll_interval' command with args: '(24,)' kwargs: '{}' result: [2, <Status.SUCCESS: 0>]
Device 0x81db (00:0d:6f:00:11:15:a8:7f) joined the network
[0x81db:zdo] ZDO request ZDOCmd.Device_annce: [0x81db, 00:0d:6f:00:11:15:a8:7f, 128]
[0x81db] Requesting 'Node Descriptor'
Tries remaining: 2
[0x81db] Extending timeout for 0xec request
[0x81db] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4335 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0x81db] Discovering endpoints
Tries remaining: 3
[0x81db] Extending timeout for 0xee request
[0x81db] Discovered endpoints: [1]
[0x81db:1] Discovering endpoint information
Tries remaining: 3
[0x81db] Extending timeout for 0xf0 request
[0x81db:1] Discovered endpoint information: <Optional endpoint=1 profile=260 device_type=1026 device_version=0 input_clusters=[0, 1, 3, 32, 1026, 1280, 2821] output_clusters=[25]>
[0x81db] Extending timeout for 0xf2 request
[0x81db:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=242 command_id=Command.Read_Attributes_rsp>
[0x81db:1] Manufacturer: Ecolink
[0x81db:1] Model: 4655BC0-R
Checking quirks for Ecolink 4655BC0-R (00:0d:6f:00:11:15:a8:7f)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because input cluster mismatch on at least one endpoint
device - 0x81db:00:0d:6f:00:11:15:a8:7f entering async_device_initialized - is_new_join: True
device - 0x81db:00:0d:6f:00:11:15:a8:7f has joined the ZHA zigbee network
[0x81db](4655BC0-R): started configuration
[0x81db:ZDO](4655BC0-R): 'async_configure' stage succeeded
[0x81db] Extending timeout for 0xf4 request
[0x81db] Extending timeout for 0xf6 request
[... 54 lines cut ...]
[0x81db:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=16 command_id=Command.Read_Attributes_rsp>
[0x81db:1:0x0001]: initializing channel: from_cache: False
[0x81db] Extending timeout for 0x16 request
[0x81db:1:0x0500] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=18 command_id=Command.Read_Attributes_rsp>
[0x81db:1:0x0500]: initializing channel: from_cache: False
[0x81db:1:0x0402]: initializing channel: from_cache: False
[0x81db] Extending timeout for 0x18 request
[0x81db:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=20 command_id=Command.Read_Attributes_rsp>
[0x81db:1:0x0000]: initializing channel: from_cache: False
[0x81db:1:0x0019]: initializing channel: from_cache: False
[0x81db:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=22 command_id=Command.Read_Attributes_rsp>
[0x81db:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=24 command_id=Command.Read_Attributes_rsp>
[0x81db:1:0x0001]: 'async_initialize' stage succeeded
[0x81db:1:0x0020]: 'async_initialize' stage succeeded
[0x81db:1:0x0500]: 'async_initialize' stage succeeded
[0x81db:1:0x0000]: 'async_initialize' stage succeeded
[0x81db:1:0x0402]: 'async_initialize' stage succeeded
[0x81db:1:0x0019]: 'async_initialize' stage succeeded
[0x81db](4655BC0-R): power source: Battery or Unknown
[0x81db](4655BC0-R): completed initialization
[0x81db:zdo] ZDO request ZDOCmd.Match_Desc_req: [0xfffd, 260, [25], []]
[0x81db:1:0x0019] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=5 command_id=1>
[0x81db:1:0x0019] ZCL request 0x0001: [1, 4335, 5, 538315041, 1]
[0x81db:1:0x0019] OTA query_next_image handler for 'Ecolink 4655BC0-R': field_control=1, manufacture_id=4335, image_type=5, current_file_version=538315041, hardware_version=1
[0x81db:1:0x0019] No OTA image is available`

Adminiuga, 2020-07-13:
Initialization log looks like any other IAS sensor, don't see anything failing.
What kind of radio?
Include all debug as per zha integration docs. I can see it getting a response to attribute read from security zone IAS cluster, but I don't see the payload  to tell if it read the device class

Gamerayers, 2020-07-14:
I am unsure how to figure out what kind of radio it actual has.  I pulled up the FCC ID information: https://fcc.report/FCC-ID/XQC-4655BC0R
The chip itself on the device is an EM357: https://www.silabs.com/documents/public/data-sheets/EM35x.pdf

I've attached the log with the debugging enabled.

[home-assistant.log](https://github.com/zigpy/zha-device-handlers/files/4919030/home-assistant.log)

walthowd, 2020-10-23:
Should be fixed by this PR in ZHA:

https://github.com/home-assistant/core/pull/42282

closed 2021-08-13 by github-actions[bot]