zigpy/zha-device-handlers #411: [Device Support Request] Legrand wireless wall switch

Crocmagnon, 2020-07-14:
**Is your feature request related to a problem? Please describe.**
Support Legrand wireless wall switch (with Netatmo) : https://www.legrand.fr/catalogue/interrupteur/interrupteur-sans-fil-supplementaire-pour-installation-connectee-celiane-with-netatmo-avec-plaque-laque-blanc

**Describe the solution you'd like**
This device should be able to control lights (on/off and dimmer).

**Additionnal context**
I first started a thread on home-assistant core (https://github.com/home-assistant/core/issues/37436) because I couldn't discover the switches using ZHA. Now I can see them but the pairing isn't complete.

**Logs**
<details>
<summary>Pairing logs</summary>

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0xc3a0 (00:04:74:00:00:ad:38:38) joined the network
[0xc3a0:zdo] ZDO request ZDOCmd.Device_annce: [0xc3a0, 00:04:74:00:00:ad:38:38, 128]
[0xc3a0] Requesting 'Node Descriptor'
Tries remaining: 2
[0xc3a0] Extending timeout for 0x6c request
[0xc3a0] Node Descriptor: <Optional byte1=18 byte2=64 mac_capability_flags=128 manufacturer_code=4129 maximum_buffer_size=89 maximum_incoming_transfer_size=63 server_mask=10752 maximum_outgoing_transfer_size=63 descriptor_capability_field=0>
[0xc3a0] Discovering endpoints
Tries remaining: 3
[0xc3a0] Extending timeout for 0x6e request
[0xc3a0] Discovered endpoints: [1]
[0xc3a0:1] Discovering endpoint information
Tries remaining: 3
[0xc3a0] Extending timeout for 0x70 request
[0xc3a0:1] Discovered endpoint information: <Optional endpoint=1 profile=260 device_type=260 device_version=1 input_clusters=[0, 3, 15, 32, 1, 64513] output_clusters=[3, 6, 8, 0, 64513, 25]>
[0xc3a0] Extending timeout for 0x72 request
[0xc3a0:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=114 command_id=Command.Read_Attributes_rsp>
[0xc3a0:1] Manufacturer:  Legrand
[0xc3a0:1] Model:  Remote switch
Checking quirks for  Legrand  Remote switch (00:04:74:00:00:ad:38:38)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
device - 0xc3a0:00:04:74:00:00:ad:38:38 entering async_device_initialized - is_new_join: True
device - 0xc3a0:00:04:74:00:00:ad:38:38 has joined the ZHA zigbee network
[0xc3a0]( Remote switch): started configuration
[0xc3a0:ZDO]( Remote switch): 'async_configure' stage succeeded
[0xc3a0] Extending timeout for 0x74 request
[0xc3a0] Extending timeout for 0x76 request
[0xc3a0] Extending timeout for 0x78 request
[0xc3a0:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0xc3a0:1:0x0000]: finished channel configuration
[0xc3a0] Extending timeout for 0x7a request
[... 620 lines cut ...]
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x000f",
        "0x0020",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0000",
        "0x0003",
        "0x0006",
        "0x0008",
        "0x0019",
        "0xfc01"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Remote switch",
  "class": "zigpy.device.Device"
}
```

Adminiuga, 2020-07-14:
this looks fine. This looks like correct "off" command for this device type of `Dimmer_Switch`. Those are state less devices, so you won't get entitities, but only events. To see the events, go to Developers -> Event and subscribe for `zha_event` and try pressing buttons
```
2020-07-05 08:32:37 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [26, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc3a0>, 1, 260, 6, b'\x11\x00\x00', 0, 175, 255, 0, 0, 0, 0, -47]
2020-07-05 08:32:37 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xc3a0>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'110000'
```

`data: b'110000'` on "On_Off" cluster
0x11 -- frame control`
0x00 -- TSN
0x00 -- command id -- Off command

Crocmagnon, 2020-07-15:
I can see the events but they take sometimes a few seconds to appear where other wall switches (tradfri for example) are instantaneous. Also when I press the switch multiple times (not necessarily very rapidly), the events seem "retained" and are distributed over dozens of seconds.

And I can't make them join a zigbee group. The command succeeds but lights in the group don't turn on when I press the switch. The "new device joined" in the logs makes me wonder whether there could be a pairing issue because that's the only device brand I've seen this message with.

dmulcahey, 2020-07-15:
No pairing issue... remotes sleep and the spec allows devices to drop and rejoin. That’s why you see the device announcement. I’ll see if I can find any references to these devices in z2m or deconz that reference group issues.

Crocmagnon, 2020-07-15:
I just tried again and I don't seem to have my latency issue for the events. I'll try installing one and keep an eye on its behavior.

I still had no luck with the grouping though. The devices don't seem to expose a LightLink cluster.

dmulcahey, 2020-07-15:
LightLink !== groups.

dmulcahey, 2020-07-15:
Wait, did you bind this to a group or did you try to add it to a group as a member?

Crocmagnon, 2020-07-15:
Ok, I may be confused here because I'm not familiar with Zigbee.

I created a group with a lightbulb in it and tried to bind the switch to the group with some clusters:

[image]

I also like using automations or even appdaemon apps like controllerx or automoli to control my lights but these require home assistant to be up and running to work. I would have liked to have a fallback that always work in case HA is down.

dmulcahey, 2020-07-15:
When you issue that command are you pressing a button on the remote to ensure it is awake?

Crocmagnon, 2020-07-15:
No, I did not. Should I press the button on the remote before or just after issuing the command? Or maybe multiple clicks?

Crocmagnon, 2020-07-15:
Pressing the remote did the trick, thanks!

dmulcahey, 2020-07-15:
Any time!

closed 2020-07-15 by Crocmagnon