zigpy/zha-device-handlers #412: [Device Support Request] Legrand shutter switch

Crocmagnon, 2020-07-14:
**Is your feature request related to a problem? Please describe.**
Support Legrand shutter switch (with Netatmo) : https://www.legrand.fr/catalogue/interrupteur/interrupteur-pour-volet-roulant-a-cabler-pour-installation-connectee-celiane-with-netatmo-avec-plaque-laque-blanc

**Describe the solution you'd like**
This device should be able to control shutters (up/down and stop).

**Additionnal context**
I first started a thread on home-assistant core (https://github.com/home-assistant/core/issues/37436) because I couldn't discover the switches using ZHA. Now I can see them but the pairing isn't complete.

**Logs**

<details>
<summary>Pairing logs</summary>

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
Device 0xbcb7 (00:04:74:00:00:8c:91:54) joined the network
[0xbcb7:zdo] ZDO request ZDOCmd.Device_annce: [0xbcb7, 00:04:74:00:00:8c:91:54, 142]
[0xbcb7] Requesting 'Node Descriptor'
Tries remaining: 2
[0xbcb7] Extending timeout for 0x38 request
[0xbcb7] Node Descriptor: <Optional byte1=17 byte2=64 mac_capability_flags=142 manufacturer_code=4129 maximum_buffer_size=89 maximum_incoming_transfer_size=63 server_mask=10752 maximum_outgoing_transfer_size=63 descriptor_capability_field=0>
[0xbcb7] Discovering endpoints
Tries remaining: 3
[0xbcb7] Discovered endpoints: [1]
[0xbcb7:1] Discovering endpoint information
Tries remaining: 3
[0xbcb7] Delivery error for seq # 0x3c, on endpoint id 0 cluster 0x0004: message send failure
[0x2fff] Delivery error for seq # 0x34, on endpoint id 11 cluster 0x0300: message send failure
[0x2fff:11:0x0300]: failed to get attributes '['color_temperature', 'current_x', 'current_y', 'color_loop_active']' on 'light_color' cluster: [0x2fff:11:0x0300]: Message send failure
Tries remaining: 2
[0xbcb7:1] Discovered endpoint information: <Optional endpoint=1 profile=260 device_type=514 device_version=1 input_clusters=[0, 3, 258, 4, 5, 15, 64513] output_clusters=[0, 64513, 25]>
[0xbcb7:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=64 command_id=Command.Read_Attributes_rsp>
[0xbcb7:1] Manufacturer:  Legrand
[0xbcb7:1] Model:  Shutter switch with neutral
Checking quirks for  Legrand  Shutter switch with neutral (00:04:74:00:00:8c:91:54)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
device - 0xbcb7:00:04:74:00:00:8c:91:54 entering async_device_initialized - is_new_join: True
device - 0xbcb7:00:04:74:00:00:8c:91:54 has joined the ZHA zigbee network
[0xbcb7]( Shutter switch with neutral): started configuration
[0xbcb7:ZDO]( Shutter switch with neutral): 'async_configure' stage succeeded
[0xbcb7:1:0x0102]: bound 'window_covering' cluster: Status.SUCCESS
[0xbcb7:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0xbcb7:1:0x0000]: finished channel configuration
[0xbcb7:1:0x0102] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=1 command_id=Command.Configure_Reporting>
[... 476 lines cut ...]
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x000f",
        "0x0102",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0000",
        "0x0019",
        "0xfc01"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Shutter switch with neutral",
  "class": "zigpy.device.Device"
}
```

Crocmagnon, 2020-07-22:
Is there any more information I can provide to help with this integration? 😊

Adminiuga, 2020-07-23:
Pairing failed for some reason, device stopped responding, but in the debug log I can see that it was paired successfully.  You should be able to control it in HA as a cover device. Do you get an entity and can you control it?

Crocmagnon, 2020-07-23:
I see the entity in HA and I see the cover controls but when I press the buttons nothing happens.

Also, the device has a LED which is currently solid red and never changes. If I remember correctly, it should either turn green or turn off if the pairing is successful from the switch point of view.

Adminiuga, 2020-07-23:
Check zigpy-cc closed issues, someone nwas having issues pairing legend and there was a trick to pair it successfuly.

But in the 2nd debug log it looks good and even reported current position a few times.

Crocmagnon, 2020-07-24:
I don’t see any issue related to Legrand devices on the zigpy-cc repo, but I’m not very familiar with this hardware. Which « CCXXXX » should I search for?

I understand that it’s supposed to work as far as the logs can tell but it really doesn’t 😅 

Yesterday, just before posting my message, I tried to pair the devices. As I posted, I got entities in HA and saw the controls but nothing worked. I didn’t touch anything for the night and now the entities are « unavailable », as reported by the developer tools - state page. I still see them in my ZHA device but I can’t control them anymore: the buttons are grayed.

Adminiuga, 2020-07-24:
If entities become unavailable, it means they don't report any data for a few hours. 
It is a no neutral device, so it might be a Zigbee end device. Do you have any Zigbee router on the network?

Also, per https://zigbee.blakadder.com/Legrand_067771.html they are recommending to update the firmware using legrand hub

Crocmagnon, 2020-07-24:
> It is a no neutral device, so it might be a Zigbee end device.

It's a device **with** neutral.

> Do you have any Zigbee router on the network?

Yes, a dozen of light bulbs (Philips Hue) and my deConz key which is really close to 3 of the 4 switches I'm trying to pair.

> update the firmware using legrand hub

Unfortunately, I don't own the hub

Crocmagnon, 2020-08-30:
So is there anything I can do to use this switch as it is?

closed 2021-08-13 by github-actions[bot]

SuperJMN, 2024-03-24:
Hello all. I'm sorry for bringing this up, but I'm facing the very same situation. I have these shutter switches around all the house.

The firmware they report is a single number "60".

My question is, is it still impossible in 2024 to make them work with ZHA?