zigpy/zha-device-handlers #414: [Device Support Request] Legrand wirless double wall switches

Crocmagnon, 2020-07-14:
**Is your feature request related to a problem? Please describe.**
Support Legrand double wireless wall switch (with Netatmo) : https://www.legrand.fr/pro/catalogue/43071-version-dooxie-with-netatmo/commande-double-sans-fil-dooxie-with-netatmo-pour-eclairage-ou-prise-connectee-ou-micromodule-blanc
**Describe the solution you'd like**
This device should be able to control lights (on/off and dimmer).

**Additionnal context**
I first started a thread on home-assistant core (https://github.com/home-assistant/core/issues/37436) because I couldn't discover the switches using ZHA. Now I can see them but the pairing isn't complete.

**Logs**
<details>
<summary>Pairing logs</summary>

```
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0xbcb7] Delivery error for seq # 0xc7, on endpoint id 1 cluster 0x0000: message send failure
Device 0x42ee (00:04:74:00:00:92:1d:78) joined the network
[0x42ee:zdo] ZDO request ZDOCmd.Device_annce: [0x42ee, 00:04:74:00:00:92:1d:78, 128]
[0x42ee] Requesting 'Node Descriptor'
Tries remaining: 2
[0x42ee] Extending timeout for 0xcf request
[0x42ee] Node Descriptor: <Optional byte1=18 byte2=64 mac_capability_flags=128 manufacturer_code=4129 maximum_buffer_size=89 maximum_incoming_transfer_size=63 server_mask=10752 maximum_outgoing_transfer_size=63 descriptor_capability_field=0>
[0x42ee] Discovering endpoints
Tries remaining: 3
[0x42ee] Extending timeout for 0xd1 request
[0x42ee] Discovered endpoints: [1, 2]
[0x42ee:1] Discovering endpoint information
Tries remaining: 3
[0x42ee] Extending timeout for 0xd3 request
[0x42ee:1] Discovered endpoint information: <Optional endpoint=1 profile=260 device_type=260 device_version=1 input_clusters=[0, 3, 15, 32, 1, 64513] output_clusters=[3, 6, 8, 0, 64513, 25, 0]>
[0x42ee] Extending timeout for 0xd5 request
[0x42ee:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=213 command_id=Command.Read_Attributes_rsp>
[0x42ee:1] Manufacturer:  Legrand
[0x42ee:1] Model:  Double gangs remote switch
[0x42ee:2] Discovering endpoint information
Tries remaining: 3
[0x42ee] Extending timeout for 0xd7 request
[0x42ee:2] Discovered endpoint information: <Optional endpoint=2 profile=260 device_type=260 device_version=1 input_clusters=[3, 15] output_clusters=[3, 6, 8]>
Checking quirks for  Legrand  Double gangs remote switch (00:04:74:00:00:92:1d:78)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 2}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1, 2}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because endpoint list mismatch: {1} {1, 2}
device - 0x42ee:00:04:74:00:00:92:1d:78 entering async_device_initialized - is_new_join: True
device - 0x42ee:00:04:74:00:00:92:1d:78 has joined the ZHA zigbee network
[0x42ee]( Double gangs remote switch): started configuration
[0x42ee:ZDO]( Double gangs remote switch): 'async_configure' stage succeeded
[0x42ee] Extending timeout for 0xd9 request
[0x42ee] Extending timeout for 0xdb request
[... 783 lines cut ...]
        "0x0006",
        "0x0008",
        "0x0019",
        "0xfc01"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0003",
        "0x000f"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006",
        "0x0008"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Double gangs remote switch",
  "class": "zigpy.device.Device"
}
```

dmulcahey, 2020-07-14:
Same as #411 check for zha_event when interacting with the device

Crocmagnon, 2020-07-15:
I had to try to pair it once, see it fail (red light on the remote but device present in HA) and retry a pairing without resetting the device (just pressing once on the reset button) to get it to successfully pair and see the `zha_event`.

closed 2020-07-15 by Crocmagnon