zigpy/zha-device-handlers #415: [FEATURE] Better Xiaomi battery percentage calculation

puddly, 2020-07-17:
Interpreting the battery percentage as the voltage linearly decreasing between 3.0V and 2.8V results in most devices having 100% charge until they're already starting to act strangely or are effectively dead.

Here's what the heartbeat voltages for my devices look like:

[image]

Some information about the data:

 - Most of my devices came with batteries that were already at the steady voltage of ~3.05V. Only brand new batteries caused spikes up to ~3.3V.
 - Two devices are kept outside (the only two that drop below 2.9V) while the rest stay indoors. Indoor temperatures fluctuate by about 15 degrees C, which is why the voltages aren't strictly decreasing as well.
 - Some devices (like the red one) never even get close to 3.0V.

I unfortunately did not save the entire heartbeat message so I do not have any of the SoC temperatures to correlate with this data for anything but temperature sensors. It seems like we could calculate a "true" battery percentage based on the temperature of the reported temperature of the battery and the battery voltage.

Here's what the two plotted together look like for a temperature sensor kept outside (only data source I have, sorry for the discontinuities):

[image]

I think anything is better than the current method so even a simple linear compensation might work. Thoughts? You can pull all of this data from your Home Assistant log files if you have debug logging enabled for zigpy.

Adminiuga, 2020-07-17:
another problem I'm noticing with xiaomi, is the variation in the reported battery level vs actual battery level., eg battery is 3.2v, but xiaomi report 3.0v. That 200mV is quite a lot in % of battery life for xiaomi

puddly, 2020-07-17:
Are you measuring the battery voltage without a load? I believe the relatively long duration between radio activity allows the battery's voltage to recover quite a bit.

Adminiuga, 2020-07-17:
yes, but that was a fresh battery, so it shouldn't matter that much. Cause on the dead battery, radio activity would sag it even more. Per specs they should measure the battery voltage during transmissions, but since it is xiaomi...

puddly, 2020-07-18:
Hmm, that's a tough one. Their favored microcontroller has an internal voltage reference and a 10 bit ADC so it seems like this should be reliable (unless they have their own reference voltage?). Is it one specific model?

stast1, 2020-08-27:
zhaquirks.xiaomi.aqara.weather.Weather
[image]

puddly, 2020-08-27:
@stast1: 2% to -2%? That means your battery voltage had a low of `2.798V` and a high of `2.802V`. Interesting. Both are lower than the lowest voltage I've ever observed.

What do the observed outdoor temperatures look like for the same range?

stast1, 2020-08-28:
> What do the observed outdoor temperatures look like for the same range?

[image]
[image]

stast1, 2020-08-28:
The sensor works without any problems.

puddly, 2020-08-28:
That is interesting, mine become unstable by the time their voltage drops this low. Is the physical distance or the path taken by the packets to your coordinator somehow compensating for impact of the low voltage? Would you happen to still be using the battery that came with the sensor?

stast1, 2020-08-28:
Distance is about 7 meters. I will use it until it stops working. This sensor is not critical for me.

stast1, 2020-09-15:
[image]

stast1, 2020-09-18:
[image]

stast1, 2020-10-17:
[image]
Interestingly, it will come to -100? :-)

stast1, 2020-10-31:
[image]

puddly, 2020-10-31:
2.786V? The JN5169 advertises "*2.0 V to 3.6 V battery operation*". Let's see if yours can get down to -800%.

Following Energizer's chart for CR2032 batteries, your battery is either at about 12% or 25% charge level (not sure if the sensor's discharge is closer to "continuous" or "pulse").

Maybe your sensor is very close to a Zigbee router or the coordinator itself so its reducing range due to the dropping voltage may not be causing it to drop off the network?

stast1, 2020-11-01:
> Maybe your sensor is very close to a Zigbee router or the coordinator itself so its reducing range due to the dropping voltage may not be causing it to drop off the network?

This is the farthest sensor on my network. About 5 meters through the wall. There is also a second sensor inside the refrigerator.:
[image]

stast1, 2020-11-22:
[image]

MattWestb, 2020-11-22:
I have not looking on Xiaomi sensors schematics but IKEAs have one resistor in serial from the battery and the chip and some extra capacitors for killing spikes then the radio is transmitting.
Its making the measurement not being the real battery more the chip working voltage that is good and bad then calculating the % of the battery power left.
My experience of both is that they can working well with low battery level if having god radio connection but if trying repairing them its not working at all if the battery is little low (the LED blinking for reset and little then start pairing and then stopping blinking and pairing).

stast1, 2020-11-28:
Everything does not work.
[image]

SHxKM, 2021-06-12:
I've also experienced this today upon upgrading from HA 2021.1.X to 2021.6.4. Sensor (WSDCGQ11LM) reports temp/humidity just fine:

[image]

MattWestb, 2021-06-12:
The interesting is wot voltage the sensor was reporting under the same time period (and also little before).

SHxKM, 2021-06-12:
> The interesting is wot voltage the sensor was reporting under the same time period (and also little before).

Is this something I should have access to or can see?

MattWestb, 2021-06-12:
I think its not being save in the database.
I was looking in my influxDB but its not being saved only the calculated % is there :-((

I was looking on my Lumi weathers (5) and all over 75% is stable and under is jumping 3-4 % up or down little now and then.

SHxKM, 2021-06-12:
> I think its not being save in the database.
> I was looking in my influxDB but its not being saved only the calculated % is there :-((
> 
> I was looking on my Lumi weathers (5) and all over 75% is stable and under is jumping 3-4 % up or down little now and then.

Maybe I can ping the cluster attribute and see what's being reported there? I don't really know what it is though.

MattWestb, 2021-06-12:
You  is having the attribute under the graph 2.8 V

SHxKM, 2021-06-12:
OK. So I guess there's nothing to be done here?

MattWestb, 2021-06-12:
I think you was running ZHA without the last patch for Xiaomi battery calculation #720 so the before update is the old way and after is the new but the voltage is the same only how its being made.
Before its was 100% more or less until it was dead.

SHxKM, 2021-06-12:
I remember the value being 70% before upgrading HA (you can see in screenshot). 

Also, I ran pip install upgrade zha quirks and zigpy. I’m on the latest release of both.

MattWestb, 2021-06-12:
I think the old calculation of 70% is in the new nearly zero = OK.

How "old" is the battery ? Lumi weather is running pity long on one.

SHxKM, 2021-06-12:
I received this device in December 2020/January 2021. Is the battery life supposed to be so bad?
Also as said values seem to be reporting fine.

MattWestb, 2021-06-12:
I think the sensor is 100% only the batty is little week.
My oldest is have getting one new but not the newest from November is still very OK.
I think you have getting one with old batter so only putting one new in and looking in some month how is running.

They is easy running over one yer is having one good router in the near.

SHxKM, 2021-06-12:
> I think the sensor is 100% only the batty is little week.
> My oldest is have getting one new but not the newest from November is still very OK.
> I think you have getting one with old batter so only [...]

Sorry I'm not following. My Zigbee router is Very close to this sensor. Are you suggesting I switch the battery? or just wait until this one dies?

MattWestb, 2021-06-12:
As long its reporting its OK and then its stooping you can putting on new in.

SHxKM, 2021-06-13:
Well what do you know, my Xiaomi device just died. So I guess reporting is pretty darn accurate...(?)

MattWestb, 2021-06-13:
If they is have low battery and working its not possible resetting them and getting them pairing then its blinking on time but have not enough power for doing the pairing.

Hope you is having some new CR-2032 so you is getting it up and running !!

SHxKM, 2021-06-13:
> If they is have low battery and working its not possible resetting them and getting them pairing then its blinking on time but have not enough power for doing the pairing.
> 
> Hope you is having some new CR-2032 so you is getting it up and running !!

Nope. The device simply died because the battery died. Put in a new battery and it's reporting a nice 94%.

SHxKM, 2021-06-14:
Update: The new battery only last a few hours before going back to report at %0:

[image]

MattWestb, 2021-06-14:
Looks like ZHA is doing all thing OK then you is getting relevant voltage but the sensor is very hungry on batteries or is the sensor defect.

Let it  running as long as possible on the "new" battery and then putting  in one new and see what happens.

I have baying my first 2 set of them from china but the last i have baying in EU so i have 3 years reclamation right if its going broken (they have printing the serial number on the bill so cant shagging the EU and Chinese devices).

closed 2021-12-18 by github-actions[bot]