zigpy/zha-device-handlers #416: [Device Support Request] Add support for ptvo.info - ptvo.switch

dmulcahey, 2020-07-18:
**Is your feature request related to a problem? Please describe.**

Moved from HA core repo:

No objects for bmp280 Data Clusters.
Can't read value from temperature, humidity, pressure.
In logs DATA exists.

Only switch and binary sensor is is created.

**Describe the solution you'd like**
Translate the 3 readings on the analog input cluster into the appropriate ZCL clusters for temperature, humidity and pressure

<details>
  <summary>Here are the logs</summary>

```
Device 0x0830 (00:12:4b:00:1d:38:d3:c6) joined the network
[0x0830:zdo] ZDO request ZDOCmd.Device_annce: [0x0830, 00:12:4b:00:1d:38:d3:c6, 142]
[0x0830] Requesting 'Node Descriptor'
Tries remaining: 2
[0x0830] Extending timeout for 0xa0 request
[0x0830] Node Descriptor: <Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4447 maximum_buffer_size=80 maximum_incoming_transfer_size=160 server_mask=0 maximum_outgoing_transfer_size=160 descriptor_capability_field=0>
[0x0830] Discovering endpoints
Tries remaining: 3
Ignoring message (b'00000a00001000') on cluster 6: unknown endpoint or cluster id: 1
[0x0830] Discovered endpoints: [6, 4, 3, 2, 1]
[0x0830:6] Discovering endpoint information
Tries remaining: 3
Ignoring message (b'00010a00001000') on cluster 6: unknown endpoint or cluster id: 'No cluster ID 0x0006 on (00:12:4b:00:1d:38:d3:c6, 3)'
[0x0830:6] Discovered endpoint information: <Optional endpoint=6 profile=260 device_type=256 device_version=0 input_clusters=[12] output_clusters=[]>
[0x0830:4] Discovering endpoint information
Tries remaining: 3
[0x0830:4] Discovered endpoint information: <Optional endpoint=4 profile=260 device_type=256 device_version=0 input_clusters=[18] output_clusters=[]>
[0x0830:3] Discovering endpoint information
Tries remaining: 3
[0x0830:6:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=2 command_id=Command.Report_Attributes>
[0x0830:6:0x000c] ZCL request 0x000a: [[<Attribute attrid=85 value=<TypeValue type=float, value=29.670000076293945>>, <Attribute attrid=28 value=<TypeValue type=CharacterString, value=C,77>>]]
[0x0830:6:0x000c] Attribute report received: present_value=29.670000076293945, description=C,77
[0x0830:6:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=3 command_id=Command.Report_Attributes>
[0x0830:6:0x000c] ZCL request 0x000a: [[<Attribute attrid=85 value=<TypeValue type=float, value=99570.0>>, <Attribute attrid=28 value=<TypeValue type=CharacterString, value=Pa,77>>]]
[0x0830:6:0x000c] Attribute report received: present_value=99570.0, description=Pa,77
[0x0830:6:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=4 command_id=Command.Report_Attributes>
[0x0830:6:0x000c] ZCL request 0x000a: [[<Attribute attrid=85 value=<TypeValue type=float, value=147.1482391357422>>, <Attribute attrid=28 value=<TypeValue type=CharacterString, value=m,77>>]]
[0x0830:6:0x000c] Attribute report received: present_value=147.1482391357422, description=m,77
[0x0830:6:0x000c] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=5 command_id=Command.Report_Attributes>
[0x0830:6:0x000c] ZCL request 0x000a: [[<Attribute attrid=85 value=<TypeValue type=float, value=54.0830078125>>, <Attribute attrid=28 value=<TypeValue type=CharacterString, value=%,77>>]]
[0x0830:6:0x000c] Attribute report received: present_value=54.0830078125, description=%,77
[0x0830:3] Discovered endpoint information: <Optional endpoint=3 profile=260 device_type=256 device_version=0 input_clusters=[12, 6] output_clusters=[6]>
[0x0830:2] Discovering endpoint information
Tries remaining: 3
[0x0830:2] Discovered endpoint information: <Optional endpoint=2 profile=260 device_type=256 device_version=0 input_clusters=[18] output_clusters=[]>
[0x0830:1] Discovering endpoint information
Tries remaining: 3
[0x0830:1] Discovered endpoint information: <Optional endpoint=1 profile=260 device_type=256 device_version=0 input_clusters=[0, 2821, 12, 6] output_clusters=[0, 2821, 6]>
[0x0830:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=178 command_id=Command.Read_Attributes_rsp>
[0x0830:1] Manufacturer: ptvo.info
[0x0830:1] Model: ptvo.switch
Checking quirks for ptvo.info ptvo.switch (00:12:4b:00:1d:38:d3:c6)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 6}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 2, 3, 4, 6}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 6}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 2, 3, 4, 6}
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1, 2, 3, 4, 6}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
[... 146 lines cut ...]
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
      "out_clusters": []
    },
    "6": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "ptvo.info",
  "model": "ptvo.switch",
  "class": "zigpy.device.Device"
}
```

</details>

**Additional context**
https://github.com/home-assistant/core/issues/37845

MaximumSU, 2020-07-18:
I asked the developer of that firmware already. Unfortunately, he not accepted all of that wishes. 

> There are one inconvenience and one bug.
Inconvenience – all data reported in AnalogInput cluster – temperature, pressure etc, but in ZCL we have separate clusters for each values. So, some software not expect, for example, the temperature value inside AnalogInput cluster, but expect it in temperature cluster.
Bug – when the sensor, connected to the some output, printed not one value (for example, BMP280 has 3 values – temperaure, humidity and pressure), the data transmited to coordinator under the same endpoint and cluster id. It is not possible in that situation to differenciate the values. The only difference – it is “description”, what makes it almost impossible to use that values in almost all software, which not do, of course, the synaxis analize of “description” attribute.

answer was:
>1. Unfortunately, I cannot implement a separate cluster for each the value type. It requires a lot of memory.
>2. See #2. Also, I did it for identical sensors on the same endpoint (DS18B20). Zigbee2MQTT can successfully parse this data.

Comment about "See #2" he meant the ability to have more then one sensor on one bus, but with different addresses, but our problem - it is about many values for the one endpoint.

Sadly, i not sure, is it possible to make the quirk for that firmware - it not have any "signature". All components, what can be used for identification - all is variables. There are only one posibility - to introduce some selection the quirk by the hands (from GUI or YAML).

dmulcahey, 2020-07-18:
We will be able to do this in a quirk without an issue. The value that is reported has enough info to split this up correctly. I’ll get something together when I have a moment.

MaximumSU, 2020-07-18:
> The value that is reported has enough info to split this up correctly.

I mean not the quirk itself, i mean the detection of apllyment of that quirk. That is not a device with the fixed endpoints and manufacturer. They are may be any combinations. So - how to detect the need of the quirk?

dmulcahey, 2020-07-18:
Interesting, what is this then? Do you have a link to the product?

MaximumSU, 2020-07-18:
@dmulcahey  Of course, [the link](https://ptvo.info/zigbee-switch-configurable-firmware-v2-210/)

dmulcahey, 2020-07-18:
Interesting, is there a way to change the manufacturer and the model? If so, users would need to add their own custom quirks locally.

MaximumSU, 2020-07-18:
> is there a way to change the manufacturer and the model?

Yes, it is easy - some fields in the "Expert" configuration tab. There are some defaults, but it can be changed easy.

closed 2021-08-13 by github-actions[bot]

garubi, 2023-06-29:
Does anyone has come up with some example or stub of a quirk for ptvo?