zigpy/zha-device-handlers #417: [BUG] Ikea blind not working properly when command issues via google home

blair287, 2020-07-20:
I i open and close the blind via home assistant it works although with a 5 second delay on command issue to it executing, but if the command is issued via google to close the blind (linked via Nabucasa) it says sure and home assistant shows the down arrow picture so its received a command but then does absolutely nothing just sits there.

I i issue the command again for a second time then it moves.

logs:

{'handler': 'google_actions',
 'msgid': '5e509ba7-f3d0-4de3-9e07-f98852125cb5',
 'payload': {'inputs': [{'context': {'locale_country': 'GB',
                                     'locale_language': 'en'},
                         'intent': 'action.devices.EXECUTE',
                         'payload': {'commands': [{'devices': [{'customData': {'baseUrl': 'https://mylink.ui.nabu.casa',
                                                                               'httpPort': 8123,
                                                                               'httpSSL': False,
                                                                               'proxyDeviceId': 'dc25e787-65fd-4c8e-bf64-7ec6681ecb93',
                                                                               'uuid': '3a199554fb2542fe87c0ff516e651fdd',
                                                                               'webhookId': 'ce565e5a19587d7e06f9f8aed82d68d10a796dea8e200b7b414709bd35610114'},
                                                                'id': 'cover.bedroom_blind'}],
                                                   'execution': [{'command': 'action.devices.commands.OpenClose',
                                                                  'params': {'openPercent': 0}}]}]}}],
             'requestId': '11090729089018306071'}}

2020-07-20 09:40:40 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event google_assistant_command[L]: request_id=11090729089018306071, entity_id=cover.bedroom_blind, execution=command=action.devices.commands.OpenClose, params=openPercent=0, source=cloud>
2020-07-20 09:40:40 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=cover, service=close_cover, service_data=entity_id=cover.bedroom_blind>
2020-07-20 09:40:40 DEBUG (MainThread) [zigpy.device] [0x5d0b] Extending timeout for 0x91 request
2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 145 under 146 request id, data: b'019101'
2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (18, 146, 0, <DeconzAddressEndpoint address_mode=2 address=0x5d0b endpoint=1>, 260, 258, 1, b'\x01\x91\x01', 2, 0)
2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x121700190012009200020b5d01040102010103000191010200
2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x121700090002002292
2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 146]
2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e18000700a600
2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x04180007000000
2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x04180013000c002292020b5d01010000000000
2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 146: 00
2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x92 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x5d0b endpoint=1>, status: 0x00
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e19000700aa00
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x1719000800010001
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17190023001c002a02000001020b5d01040102010500088e0b010000afff233a8600ca
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [28, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x5d0b>, 1, 260, 258, b'\x08\x8e\x0b\x01\x00', 0, 175, 255, 35, 58, 134, 0, -54]
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy.zcl] [0x5d0b:1:0x0102] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=142 command_id=Command.Default_Response>
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x5d0b>, ep: 1, profile: 0x0104, cluster_id: 0x0102, data: b'088e0b0100'
2020-07-20 09:40:41 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5d0b:1:0x0102]: executed 'down_close' command with args: '()' kwargs: '{}' result: [1, <Status.SUCCESS: 0>]
2020-07-20 09:40:41 DEBUG (MainThread) [homeassistant.components.zha.cover] state=closing
2020-07-20 09:40:41 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=cover.bedroom_blind, old_state=<state cover.bedroom_blind=open; current_position=100, friendly_name=Bedroom Blind, supported_features=15 @ 2020-07-20T07:30:35.886728+01:00>, new_state=<state cover.bedroom_blind=closing; current_position=100, friendly_name=Bedroom Blind, supported_features=15 @ 2020-07-20T09:40:41.780775+01:00>>
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x171a000800010001
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x171a0023001c002202000001020b5d0104010201050008910b010000afffc0338600ca
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [28, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x5d0b>, 1, 260, 258, b'\x08\x91\x0b\x01\x00', 0, 175, 255, 192, 51, 134, 0, -54]
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy.zcl] [0x5d0b:1:0x0102] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=145 command_id=Command.Default_Response>
2020-07-20 09:40:41 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x5d0b>, ep: 1, profile: 0x0104, cluster_id: 0x0102, data: b'08910b0100'
2020-07-20 09:40:41 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5d0b:1:0x0102]: executed 'down_close' command with args: '()' kwargs: '{}' result: [1, <Status.SUCCESS: 0>]
2020-07-20 09:40:41 DEBUG (MainThread) [homeassistant.components.zha.cover] state=closing
2020-07-20 09:40:41 DEBUG (MainThread) [hass_nabucasa.iot] Publishing message:
{'msgid': '5e509ba7-f3d0-4de3-9e07-f98852125cb5',
 'payload': {'payload': {'commands': [{'ids': ['cover.bedroom_blind'],
                                       'states': {'online': True,
                                                  'openPercent': 100},
                                       'status': 'SUCCESS'}]},
             'requestId': '11090729089018306071'}}

Adminiuga, 2020-07-20:
Zigbee command completed successfuly
```
2020-07-20 09:40:41 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5d0b:1:0x0102]: executed 'down_close' command with args: '()' kwargs: '{}' result: [1, <Status.SUCCESS: 0>]
```

Nabu casa reported successfuly???
```
2020-07-20 09:40:41 DEBUG (MainThread) [hass_nabucasa.iot] Publishing message:
{'msgid': '5e509ba7-f3d0-4de3-9e07-f98852125cb5',
'payload': {'payload': {'commands': [{'ids': ['cover.bedroom_blind'],
'states': {'online': True,
'openPercent': 100},
'status': 'SUCCESS'}]},
'requestId': '11090729089018306071'}}
```

I'm not sure the problem is with ZHA...

blair287, 2020-07-21:
If it completed why did the blind not move you can see from the screen shot it says closing but it's not actually doing anything.

As far as I can see from logs it's asking blind to move to 0% but the blind reports a state of 100% so it's not completed.

Adminiuga, 2020-07-21:
🤷 I believe you are having issues, but I don't know the cause. From ZHA/Zigbee point of view
1. requests was sent out at `2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 145 under 146 request id, data: b'019101'`
2. Far end device confirmed reception of the request `2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 146: 00` within the same second, you can't ask more from an end device
3. Far end device device sent us a reply confirming action of the request at `2020-07-20 09:40:41 DEBUG (MainThread) [zigpy.zcl] [0x5d0b:1:0x0102] ZCL deserialize: <ZCLHeader frame_control= manufacturer=None tsn=145 command_id=Command.Default_Response>` the very next second. 

From zigbee point of view everything was successful. If the device does not behave correctly then 🤷 Get a zigbee sniffer and compare the packets on air? Try openening the blinds instead of closing? maybe commands are reversed?

blair287, 2020-07-21:
> 🤷 I believe you are having issues, but I don't know the cause. From ZHA/Zigbee point of view
>
> 1. requests was sent out at `2020-07-20 09:40:40 DEBUG (MainThread) [zigpy_deconz.zigbee.application] [...]

Trying to remove it to repair but the device will not go from the device list in ZHA all the opens disappear after clicking remove device but the device itself is just stuck in ZHA I've tried restarting HA still there.

blair287, 2020-07-21:
Only came down a few inches this time and then just stopped.

[image]

dmulcahey, 2020-07-21:
Do these things have any sort of calibration that has to happen?

blair287, 2020-07-22:
> Do these things have any sort of calibration that has to happen?

Once it's paired you press the down button on the blind and then again when it's at the length you need to stop it. Then you double tap the bottom button to set the max length.

That's working it I tell HA move to 50% it moves to half way of the length I set.

I've got a feeling maybe Google is messing up because of the massive delay between asking the blind to move and it actually moving as if I ask Google to close or open she waits about 6 second before saying ok then it doesn't move if I ask a second time she responds ok within 1 second and the blind moves within 1 second.

But on both tries HA receives the command ZHA issues the command but the blind doesn't seem to react or it takes up to 10 seconds after the command is issued.

But what's strange is if I press down or up in the HA app it always moves although still has the big delay times.

blair287, 2020-07-23:
> Do these things have any sort of calibration that has to happen?

Looks like I'm not the only one experiencing these delays. What could I do to try and help track down what's going wrong with these blinds on ZHA.

[image]

Adminiuga, 2020-07-23:
Battery operated devices poll for data their parents and sleep between the polls, so there's an inherent delay when communicating with Zigbee End-Devices, regardless if those are Ikea or any other vendor.

blair287, 2020-07-23:
> Battery operated devices poll for data their parents and sleep between the polls, so there's an inherent delay when communicating with Zigbee End-Devices, regardless if those are Ikea or any other vendor.

Yes however on Smartthings there isn't this delay it's moving within 1 second and thats through the cloud too.

I believe it's this delay that causes the issue with Google if I move the blind via the app then issue the command from Google within a few seconds it works most of the ti
me.

I have been testing and calling the service cover close 4/10 times also made the icon change to show the blind should be moving up or down but it doesn't actually move.

When I first tried this blind on HA back in February this issue wasn't present from what I can remember.

Adminiuga, 2020-07-23:
delay between google and HA is not ZHA's concern. I was looking only on the delay between ZHA service call and ZCL request sent to the radio and far end acknowledging the reception of the request. in the logs you provided I don't see any significant delay.

blair287, 2020-07-23:
> Battery operated devices poll for data their parents and sleep between the polls, so there's an inherent delay when communicating with Zigbee End-Devices, regardless if those are Ikea or any other vendor.

> delay between google and HA is not ZHA's concern. I was looking only on the delay between ZHA service call and ZCL request sent to the radio and far end acknowledging the reception of the request. in the logs you provided I don't see any significant delay.

Here is a video of one of the issues.

You can see when I ask Google to move the blind sometimes it will not move or it will move a little on this occasion it was moving a few cm then stopping.

However when I operate it via the app it moves perfectly fine.

https://bianco.tv/video/Pc7pZ

I will try and record the other issue where it just doesn't move too.

blair287, 2020-07-24:
> delay between google and HA is not ZHA's concern. I was looking only on the delay between ZHA service call and ZCL request sent to the radio and far end acknowledging the reception of the request. in the logs you provided I don't see any significant delay.

Is there any other info I can give you so we can try and find the issue? I really want only one ZigBee network don't want to have to get an IKEA hub just for a blind.

I will once I finish my next two work shift gets logs for it working, partially working and not working see if any clues in there.

Is there anything I can get off the Smartthings hub with ZigBee logs to see any clues too?

blair287, 2020-07-28:
> delay between google and HA is not ZHA's concern. I was looking only on the delay between ZHA service call and ZCL request sent to the radio and far end acknowledging the reception of the request. in the logs you provided I don't see any significant delay.

here is a log when i asked google to set the blind to 90% set position commands to google seems to be more reliable than asking for open or close anything here thats usefully ill also capture another log of a open or close command when it ignores it.

```2020-07-28 07:37:49 DEBUG (MainThread) [hass_nabucasa.iot] Received message:
{'handler': 'google_actions',
 'msgid': 'bb4870d8-a4e1-46b2-81a2-a8957b76e713',
 'payload': {'inputs': [{'context': {'locale_country': 'GB',
                                     'locale_language': 'en'},
                         'intent': 'action.devices.EXECUTE',
                         'payload': {'commands': [{'devices': [{'customData': {'baseUrl': 'https://jzq3suvnrfmpstssffp0ulvqy22aowwy.ui.nabu.casa',
                                                                               'httpPort': 8123,
                                                                               'httpSSL': False,
                                                                               'proxyDeviceId': 'dc25e787-65fd-4c8e-bf64-7ec6681ecb93',
                                                                               'uuid': '3a199554fb2542fe87c0ff516e651fdd',
                                                                               'webhookId': 'ce565e5a19587d7e06f9f8aed82d68d10a796dea8e200b7b414709bd35610114'},
                                                                'id': 'cover.bedroom_blind'}],
                                                   'execution': [{'command': 'action.devices.commands.OpenClose',
                                                                  'params': {'openPercent': 90}}]}]}}],
             'requestId': '5288864564052301216'}}

2020-07-28 07:37:49 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event google_assistant_command[L]: request_id=5288864564052301216, entity_id=cover.bedroom_blind, execution=command=action.devices.commands.OpenClose, params=openPercent=90, source=cloud>
2020-07-28 07:37:49 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=cover, service=set_cover_position, service_data=entity_id=cover.bedroom_blind, position=90>
2020-07-28 07:37:49 DEBUG (MainThread) [zigpy.device] [0x3b74] Extending timeout for 0x21 request
2020-07-28 07:37:49 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 33 under 34 request id, data: b'0121050a'
2020-07-28 07:37:49 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (19, 34, 0, <DeconzAddressEndpoint address_mode=2 address=0x3b74 endpoint=1>, 260, 258, 1, b'\x01!\x05\n', 2, 0)
2020-07-28 07:37:49 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 34]
2020-07-28 07:37:49 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-07-28 07:37:49 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-07-28 07:37:49 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 34: 00
2020-07-28 07:37:49 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x22 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x3b74 endpoint=1>, status: 0x00
2020-07-28 07:37:50 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-07-28 07:37:50 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-07-28 07:37:50 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [28, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 42>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3b74>, 1, 260, 258, b'\x08\x17\x0b\x05\x00', 0, 175, 255, 238, 136, 142, 0, -52]
2020-07-28 07:37:50 DEBUG (MainThread) [zigpy.zcl] [0x3b74:1:0x0102] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=23 command_id=Command.Default_Response>
2020-07-28 07:37:50 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3b74>, ep: 1, profile: 0x0104, cluster_id: 0x0102, data: b'08170b0500'
2020-07-28 07:37:50 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x3b74:1:0x0102]: executed 'go_to_lift_percentage' command with args: '(10,)' kwargs: '{}' result: [5, <Status.SUCCESS: 0>]
2020-07-28 07:37:50 DEBUG (MainThread) [homeassistant.components.zha.cover] state=closing
2020-07-28 07:37:50 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=cover.bedroom_blind, old_state=<state cover.bedroom_blind=open; current_position=100, friendly_name=Bedroom Blind, supported_features=15 @ 2020-07-27T18:53:19.244244+01:00>, new_state=<state cover.bedroom_blind=closing; current_position=100, friendly_name=Bedroom Blind, supported_features=15 @ 2020-07-28T07:37:50.975574+ [... cut ...]
2020-07-28 07:37:53 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-07-28 07:37:53 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 39: 00
2020-07-28 07:37:53 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x27 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x3b74 endpoint=1>, status: 0x00
2020-07-28 07:37:53 DEBUG (MainThread) [hass_nabucasa.google_report_state] Received message:
{'msgid': 'c1bc3582d36d431abe75ca9a27e741ae'}
```

blair287, 2020-07-28:
here is a log from an open command which it didnt move.

```{'handler': 'google_actions',
 'msgid': '6c645609-9eaf-43c3-bb95-e36c2000b9fa',
 'payload': {'inputs': [{'context': {'locale_country': 'GB',
                                     'locale_language': 'en'},
                         'intent': 'action.devices.EXECUTE',
                         'payload': {'commands': [{'devices': [{'customData': {'baseUrl': 'https://mylink.nabu.casa',
                                                                               'httpPort': 8123,
                                                                               'httpSSL': False,
                                                                               'proxyDeviceId': 'dc25e787-65fd-4c8e-bf64-7ec6681ecb93',
                                                                               'uuid': '3a199554fb2542fe87c0ff516e651fdd',
                                                                               'webhookId': 'ce565e5a19587d7e06f9f8aed82d68d10a796dea8e200b7b414709bd35610114'},
                                                                'id': 'cover.bedroom_blind'}],
                                                   'execution': [{'command': 'action.devices.commands.OpenClose',
                                                                  'params': {'openPercent': 100}}]}]}}],
             'requestId': '6186555781578573228'}}

2020-07-28 08:13:35 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event google_assistant_command[L]: request_id=6186555781578573228, entity_id=cover.bedroom_blind, execution=command=action.devices.commands.OpenClose, params=openPercent=100, source=cloud>
2020-07-28 08:13:35 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=cover, service=open_cover, service_data=entity_id=cover.bedroom_blind>
2020-07-28 08:13:35 DEBUG (MainThread) [zigpy.device] [0x3b74] Extending timeout for 0xb3 request
2020-07-28 08:13:35 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 179 under 180 request id, data: b'01b300'
2020-07-28 08:13:35 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (18, 180, 0, <DeconzAddressEndpoint address_mode=2 address=0x3b74 endpoint=1>, 260, 258, 1, b'\x01\xb3\x00', 2, 0)
2020-07-28 08:13:35 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 180]
2020-07-28 08:13:35 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-07-28 08:13:35 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-07-28 08:13:35 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 180: 00
2020-07-28 08:13:35 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xb4 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x3b74 endpoint=1>, status: 0x00
2020-07-28 08:13:39 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-07-28 08:13:39 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-07-28 08:13:39 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [28, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3b74>, 1, 260, 258, b'\x08\xb1\x0b\x00\x00', 0, 175, 255, 155, 143, 142, 0, -54]
2020-07-28 08:13:39 DEBUG (MainThread) [zigpy.zcl] [0x3b74:1:0x0102] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=177 command_id=Command.Default_Response>
2020-07-28 08:13:39 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3b74>, ep: 1, profile: 0x0104, cluster_id: 0x0102, data: b'08b10b0000'
2020-07-28 08:13:39 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x3b74:1:0x0102]: executed 'up_open' command with args: '()' kwargs: '{}' result: [0, <Status.SUCCESS: 0>]
2020-07-28 08:13:39 DEBUG (MainThread) [homeassistant.components.zha.cover] state=opening
2020-07-28 08:13:39 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=cover.bedroom_blind, old_state=<state cover.bedroom_blind=closing; current_position=90, friendly_name=Bedroom Blind, supported_features=15 @ 2020-07-28T07:57:14.849224+01:00>, new_state=<state cover.bedroom_blind=opening; current_position=90, friendly_name=Bedroom Blind, supported_features=15 @ 2020-07-28T08:13:39.527586+01:00>>
2020-07-28 08:13:39 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-07-28 08:13:39 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-07-28 08:13:39 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [28, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x3b74>, 1, 260, 258, b'\x08\xb3\x0b\x00\x00', 0, 175, 255, 1 [... cut ...]
2020-07-28 08:14:04 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 209: 00
2020-07-28 08:14:04 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xd1 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x91fb endpoint=11>, status: 0x00
2020-07-28 08:14:04 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-07-28 08:14:04 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
```

blair287, 2020-07-28:
This one i asked for close blind but its didnt move again so i reissued the command again then it moved correctly.

```{'handler': 'google_actions',
 'msgid': '39b1e5f4-44a9-480e-a4d9-1ff47186c928',
 'payload': {'inputs': [{'context': {'locale_country': 'GB',
                                     'locale_language': 'en'},
                         'intent': 'action.devices.EXECUTE',
                         'payload': {'commands': [{'devices': [{'customData': {'baseUrl': 'mylink.ui.nabu.casa',
                                                                               'httpPort': 8123,
                                                                               'httpSSL': False,
                                                                               'proxyDeviceId': 'dc25e787-65fd-4c8e-bf64-7ec6681ecb93',
                                                                               'uuid': '3a199554fb2542fe87c0ff516e651fdd',
                                                                               'webhookId': 'ce565e5a19587d7e06f9f8aed82d68d10a796dea8e200b7b414709bd35610114'},
                                                                'id': 'cover.bedroom_blind'}],
                                                   'execution': [{'command': 'action.devices.commands.OpenClose',
                                                                  'params': {'openPercent': 0}}]}]}}],
             'requestId': '15710398858588850630'}}

2020-07-28 08:44:57 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event google_assistant_command[L]: request_id=15710398858588850630, entity_id=cover.bedroom_blind, execution=command=action.devices.commands.OpenClose, params=openPercent=0, source=cloud>
2020-07-28 08:44:57 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=cover, service=close_cover, service_data=entity_id=cover.bedroom_blind>
2020-07-28 08:44:57 DEBUG (MainThread) [zigpy.device] [0x3b74] Extending timeout for 0x5a request
2020-07-28 08:44:57 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 90 under 91 request id, data: b'015a01'
2020-07-28 08:44:57 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (18, 91, 0, <DeconzAddressEndpoint address_mode=2 address=0x3b74 endpoint=1>, 260, 258, 1, b'\x01Z\x01', 2, 0)
2020-07-28 08:44:57 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 91]
2020-07-28 08:44:57 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-07-28 08:44:57 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-07-28 08:44:57 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 91: 00
2020-07-28 08:44:57 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x5b 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x3b74 endpoint=1>, status: 0x00
2020-07-28 08:44:58 DEBUG (MainThread) [homeassistant.components.zha.entity] light.hallway_light: polling current state - from cache: False
2020-07-28 08:44:58 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 92 under 93 request id, data: b'005c000000'
2020-07-28 08:44:58 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 93, 0, <DeconzAddressEndpoint address_mode=2 address=0x18ba endpoint=11>, 260, 6, 1, b'\x00\\\x00\x00\x00', 2, 0)
2020-07-28 08:44:58 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 93]
2020-07-28 08:44:59 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2020-07-28 08:44:59 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2020-07-28 08:44:59 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 93: 00
2020-07-28 08:44:59 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x5d 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x18ba endpoint=11>, status: 0x00
2020-07-28 08:44:59 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2020-07-28 08:44:59 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-07-28 08:44:59 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [31, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x18ba>, 11, 260, 6, b'\x18\\\x01\x00\x00\x00\x10\x00', 0, 175, 255, 33, 149, 142, 0, -59]
2020-07-28 08:44:59 DEBUG (MainThread) [zigpy.zcl] [0x18ba:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=92 command_id=Command.Read_Attributes_rsp>
2020-07-28 08:44:59 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x18ba>, ep: 11, profile: 0x0104, cluster_id: 0x0006, data: b'185c010000001000'
[... cut ...]
2020-07-28 08:45:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 153, 0, <DeconzAddressEndpoint address_mode=2 address=0x3b74 endpoint=1>, 260, 258, 1, b'\x18\x1a\x0b\n\x00', 2, 0)
2020-07-28 08:45:41 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 153]
2020-07-28 08:45:41 DEBUG (MainThread) [hass_nabucasa.google_report_state] Publishing message:
{'msgid': '55b26f17d6c9442aa4158eecf94b45cc',
 'payload': {'devices': {'states': {'cover.bedroom_blind': {'online': True,
                                                            'openPercent': 0}}}}}
```

blair287, 2020-07-31:
I've tried to figure this out and I believe the issue is now not with ZHA (apart from the delay issue which you can see sometimes in the logs) 

Using Nabu casa issue there every time. But with duckdns and manual Google integration the issue regarding the blind not opening or closing is not there.

Thanks for help I will close this issue and open a new one for the delay issue.

closed 2020-07-31 by blair287