zigpy/zha-device-handlers #418: [BUG] Hue Outdoor Sensor (SML002) Drops Off ZHA With Reboot of HA

townsmcp, 2020-07-20:
**Describe the bug**
For a few releases of HA (couple of weeks roughly), I have experinced weird behaviour with Hue Outdoor sensor - when rebooting HA, the sensor drops off from ZHA and is unreachable. The work around for now is to delete the device and pair it again in HA however this is getting a bit of a pain having to type in the entity ID again to match automations etc.
I have quirks enabled in configuration.yaml

**To Reproduce**
Steps to reproduce the behavior:
1. Pair SML002 to HA via ZHA
2. Reboot HA

**Expected behavior**
Device stays known after a reboot

**Screenshots**
[image]
[image]

Adminiuga, 2020-07-20:
1. what radio?
2. debug logs:
  a. debug log prior a reboot: try reading the "model" attribute of the basic cluster successfuly
  b. debug log for a reboot and an attempt to read the "model" attribute of the basic cluster

townsmcp, 2020-07-22:
Cool - I will grab the debug logs and update.
In the mean time, the radio is Conbee 2 USB stick

townsmcp, 2020-07-24:
@Adminiuga I think this is what your looking for?
Before reboot:
2020-07-24 08:19:14 DEBUG (MainThread) [homeassistant.components.zha.api] Read attribute for: cluster_id: [0] cluster_type: [in] endpoint_id: [1] attribute: [5] manufacturer: [None] response: [SML002] failure: [{}],

After a reboot it does seem to be keeping the sensor at the moment. I have upgraded to 0.113.0 though. I will keep an eye on it for the next day

townsmcp, 2020-07-24:
@Adminiuga Didnt last long. No reboot of HA. The sensor just randomly disconnected.
When clicking on Manage Clusters I get a popup with a drop down box however when I click the drop down box nothing is there

Adminiuga, 2020-07-24:
What other devices you have on the network? Are there any Zigbee routers?

townsmcp, 2020-07-25:
The outdoor motion sensor sits right below a Hue spotlight (therefore it acts as a repeater/router).
14 Hue motion sensors.
5 zigbee door sensors
3 Hue strip lights
8 zigbee plug sockets (all acting as repeater/router)
These are all scattered around the house to reinforce the Zigbee mesh.
Zigbee is using channel 20 and wifi router is set to channel 1 to avoid channel clash.

This issue with the hue outdoor sensor has been happening for roughly 2 to 3 weeks at most. Around the time it happened first time I recall seeing an updated SML002 quirk on here? I wonder if that could be the issue? If so, is there a way to roll back a quick on my local machine?

Adminiuga, 2020-07-25:
Quirk wouldn't shouldn't affect the device. Having a spotlight doesn't necessarily makes it a parent device.

Remove motion sensor.
Pair it VIA spot light.

Provide issue information as per zha docs. Enable debug logs and post debug logs

meremortals70, 2020-08-18:
Have you tried upgrading the firmware in them. I had a lot of problem with my three until I did a firmware upgrade to them.

as19git67, 2020-09-02:
I have the same problem with my hue motion outdoor sensor getting unavailable after reboot. I've read in another home assistant bug report that the issue could be that the zigbee.db file is not saved automatically after adding the device and those get lost after reboot. Interesting that it is the same sensor and not random zigbee device.

meremortals70, 2020-09-05:
> @meremortals70 according to Hue release notes page (https://www.philips-hue.com/en-gb/support/release-notes/accessories), he latest update is from 11th March 2019. I moved over to ZHA only a couple of months ago with the latest firmware at that point. Thanks for the suggestion though

Sorry I meant the Conbee II firmware.

dmulcahey, 2020-10-10:
From the screen shot at the top: the device shows unknown model / manufacturer. I fix was pushed to Zigpy for this and it is available in .116 of HA. Please remove this device from HA. update to .116 and then pair the device again. I am going to close this issue as it is not a quirks related problem.

closed 2020-10-10 by dmulcahey