zigpy/zha-device-handlers #420: [Device Support Request] Siterwell GS361 (uses the tuya platform)

saz, 2020-07-24:
**Is your feature request related to a problem? Please describe.**
No support for Siterwell GS361 or essentials ESS-HK-TRV-6103 in zha (both are the same).
It's possible to pair such a device, but there are no entities to controll it.

**Describe the solution you'd like**
Support for the above mentioned TRVs

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
[0x4a73:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=104 command_id=Command.Report_Attributes>
[0x4a73:1:0x0400] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=20001>>]]
[0x4a73:1:0x0400] Attribute report received: measured_value=20001
Device 0x2d4f (08:6b:d7:ff:fe:c8:3f:a5) joined the network
[0x2d4f:zdo] ZDO request ZDOCmd.Device_annce: [0x2d4f, 08:6b:d7:ff:fe:c8:3f:a5, 128]
[0x2d4f] Requesting 'Node Descriptor'
Tries remaining: 2
[0x2d4f] Extending timeout for 0x3a request
[0x2d4f] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=0 maximum_buffer_size=82 maximum_incoming_transfer_size=255 server_mask=11264 maximum_outgoing_transfer_size=255 descriptor_capability_field=0>
[0x2d4f] Discovering endpoints
Tries remaining: 3
[0x2d4f] Extending timeout for 0x3c request
[0x2d4f] Discovered endpoints: [1]
[0x2d4f:1] Discovering endpoint information
Tries remaining: 3
[0x2d4f] Extending timeout for 0x3e request
[0x2d4f:1] Discovered endpoint information: <Optional endpoint=1 profile=260 device_type=0 device_version=0 input_clusters=[0, 3] output_clusters=[3, 25]>
[0x2d4f] Extending timeout for 0x40 request
[0x2d4f:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=64 command_id=Command.Read_Attributes_rsp>
[0x2d4f:1] Manufacturer: _TYST11_jeaxp72v
[0x2d4f:1] Model: eaxp72v
Checking quirks for _TYST11_jeaxp72v eaxp72v (08:6b:d7:ff:fe:c8:3f:a5)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
device - 0x2d4f:08:6b:d7:ff:fe:c8:3f:a5 entering async_device_initialized - is_new_join: True
device - 0x2d4f:08:6b:d7:ff:fe:c8:3f:a5 has joined the ZHA zigbee network
[0x2d4f](eaxp72v): started configuration
[0x2d4f:ZDO](eaxp72v): 'async_configure' stage succeeded
[0x2d4f] Extending timeout for 0x42 request
[0x2d4f] Extending timeout for 0x44 request
[0x2d4f:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x2d4f:1:0x0000]: finished channel configuration
[0x2d4f] Extending timeout for 0x46 request
[0x2d4f:1:0x0019]: bound 'ota' cluster: Status.SUCCESS
[0x2d4f:1:0x0019]: finished channel configuration
Ignoring message (b'09000100350404000102') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
Ignoring message (b'09010100350404000102') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
[... 23 lines cut ...]
[0x2d4f:1:0x0019]: initializing channel: from_cache: False
Ignoring message (b'090d0100390701000100') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
Ignoring message (b'090e0100390701000100') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
Ignoring message (b'090f01003a1104000100') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
Ignoring message (b'091001003a1104000100') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
Ignoring message (b'091101003a1104000100') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
Ignoring message (b'091201003b1304000100') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
[0x2d4f:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=74 command_id=Command.Read_Attributes_rsp>
[0x2d4f:1:0x0000]: initializing channel: from_cache: False
[0x2d4f:1:0x0000]: 'async_initialize' stage succeeded
[0x2d4f:1:0x0019]: 'async_initialize' stage succeeded
[0x2d4f](eaxp72v): power source: Battery or Unknown
[0x2d4f](eaxp72v): completed initialization
Ignoring message (b'091301003b1304000100') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
Ignoring message (b'091401003b1304000100') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
Ignoring message (b'091501003c0302000400000116') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
Ignoring message (b'091601003c0302000400000116') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
Ignoring message (b'091701003c0302000400000116') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
Ignoring message (b'091801003d1502000400000064') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
Ignoring message (b'091901003d1502000400000064') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
Ignoring message (b'091a01003d1502000400000064') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (08:6b:d7:ff:fe:c8:3f:a5, 1)'
```

**Additional context**
Support is available in zigbee2mqtt, which might help to get it working, see https://github.com/Koenkk/zigbee-herdsman-converters/pull/1120

FrnchFrgg, 2020-10-24:
I might be able to translate the zigbee2mqtt support into a quirk.

MattWestb, 2020-10-24:
This devices is on my wish list then its have all the futures that i want and nice design.
Hope getting it integrated in ZHA and if possible more of the tuya TRV devices but this is my number one for the moment.

FrnchFrgg, 2020-10-24:
The quirk I would work would base off the work done in #530 so I suppose you'd have to wait until that one is merged. I can still upload a preliminary version here integrating the needed changes.

FrnchFrgg, 2020-10-24:
@MattWestb @saz can you test this branch to see if the valve seem usable ? As I do not have the device, I'm blindly translating things. In particular, I found two different places where the device is supported, and they do not agree on the meaning of two attributes: one thinks they report the states, the other thinks they enable/disable such reporting. I implemented the reporting (because otherwise I do not know how the reporting would be done).

Could you test when the "valve_state" attribute changes on the TuyaManufacturerCluster, and if writing to it does something sane or not ? Same for "window_open".

saz, 2020-10-29:
I'd love to, but I don't have a device laying around right now. :-(

MattWestb, 2020-11-01:
@FrnchFrgg I have odred on 5 pack with GW that looks the same [RC-290.zigbee](https://www.pearl.at/at-a-NX4917-3103.shtml?vid=953&curr=ATS) if its is that we knowing in some day or you is getting little more code to do ;-)))

MattWestb, 2020-11-11:
I have getting the TRVs and have starting installing them (need adapter for mounting) and using the original GW for testing. Waiting for your code being committed so being easier testing in my production system.

PR #538 (2020-11-27): Thermostat
Implement support for the Siterwell GS361 and Shojzj 378RT (fixes: #420, fixes: #541).
Also implement support for Moes HY638
Support for Revolt NX4914-944 (fixes: #581)
Also support electric heating thermostat Moes BHT-002GCLZB (fixes #560)
closed 2020-11-27 by dmulcahey