zigpy/zha-device-handlers #421: Xiaomi QBKG03LM

egelmex, 2020-07-25:
Xiaomi QBKG03LM is showing up as a router with two on_off entities, however the device state doesn't update to match the switch state.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
[0x4a34:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=106 command_id=Command.Report_Attributes>
[0x4a34:1:0x0400] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=313>>]]
[0x4a34:1:0x0400] Attribute report received: measured_value=313
[0x4a34:1:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=107 command_id=Command.Report_Attributes>
[0x4a34:1:0x0406] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=bitmap8, value=bitmap8.1>>]]
[0x4a34:1:0x0406] Attribute report received: occupancy=1
[0x4a34:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=108 command_id=Command.Report_Attributes>
[0x4a34:1:0x0400] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=321>>]]
[0x4a34:1:0x0400] Attribute report received: measured_value=321
[0x4a34:1:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=109 command_id=Command.Report_Attributes>
[0x4a34:1:0x0406] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=bitmap8, value=bitmap8.1>>]]
[0x4a34:1:0x0406] Attribute report received: occupancy=1
Ignoring message (b'18070a0000100000f02300f2a103') on cluster 6: unknown endpoint or cluster id: 3
Ignoring message (b'18080a00001000') on cluster 6: unknown endpoint or cluster id: 5
Ignoring message (b'1c5f11090af0ff4109aa1005418701011000') on cluster 0: unknown endpoint or cluster id: 1
Device 0x2a9a (00:15:8d:00:02:e2:d3:60) joined the network
Device 00:15:8d:00:02:e2:d3:60 changed id (0xa1f2 => 0x2a9a)
[0x2a9a:zdo] ZDO request ZDOCmd.Device_annce: [0x2a9a, 00:15:8d:00:02:e2:d3:60, 132]
Checking quirks for None None (00:15:8d:00:02:e2:d3:60)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} set()
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} set()
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} set()
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} set()
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} set()
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} set()
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} set()
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} set()
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} set()
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} set()
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} set()
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
[... 37 lines cut ...]
[0x2a9a](unk_model): stored in registry: ZhaDeviceEntry(name='LUMI lumi.ctrl_neutral2', ieee='00:15:8d:00:02:e2:d3:60', last_seen=1595695633.9536753)
[0x2a9a](unk_model): started initialization
[0x2a9a:ZDO](unk_model): 'async_initialize' stage succeeded
[0x2a9a](unk_model): power source: Mains
[0x2a9a](unk_model): completed initialization
Ignoring message (b'18010a0000100000f02300ffff02') on cluster 6: unknown endpoint or cluster id: 2
Ignoring message (b'18020a0000100000f02300ffff02') on cluster 6: unknown endpoint or cluster id: 3
[0x4a34:1:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=110 command_id=Command.Report_Attributes>
[0x4a34:1:0x0400] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=317>>]]
[0x4a34:1:0x0400] Attribute report received: measured_value=317
[0x4a34:1:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=111 command_id=Command.Report_Attributes>
[0x4a34:1:0x0406] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=bitmap8, value=bitmap8.1>>]]
[0x4a34:1:0x0406] Attribute report received: occupancy=1
```

**Additional context**
```
{
  "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=132 manufacturer_code=4151 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=1 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>",
  "endpoints": {},
  "manufacturer": "unk_manufacturer",
  "model": "unk_model",
  "class": "zigpy.device.Device"
}
```

Adminiuga, 2020-07-25:
Remove and re-pair. It didn't get discovered correctly.

egelmex, 2020-07-26:
Re-pair fixed it. Closing ticket

closed 2020-07-26 by egelmex