zigpy/zha-device-handlers #423: [Device Support Request] Unable to group second switch in lumi.switch.b2nacn02

ruifung, 2020-07-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

It's strange how in ZHA, the first switch can be bound to a zigbee group while the second button can not.
Even though both switches show up as separate entities in ZHA.

Basic functionality works fine though, i.e. switching on/off and state updates to ZHA.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Allow for the second button to be bound to the zigbee group.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

```json
{
  "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4447 maximum_buffer_size=127 maximum_incoming_transfer_size=100 server_mask=0 maximum_outgoing_transfer_size=100 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a",
        "0x0010"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0006",
        "0x0010"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x0004",
        "0x000c"
      ]
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0053",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": [
        "0x000c"
      ]
    },
    "5": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0010",
        "0x0012"
      ],
      "out_clusters": []
    },
    "6": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0010",
        "0x0012"
      ],
      "out_clusters": []
    },
    "7": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0010",
        "0x0012"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.b2nacn02",
  "class": "zigpy.device.Device"
}
```

**Additional context**
Add any other context or screenshots about the feature request here.

[image]
[image]

dmulcahey, 2020-08-03:
This may be a device limitation. I'm not 100% positive though. Can you capture full debug logs when you attempt the binding?

Adminiuga, 2020-08-03:
The 2nd endpoint doesn't have the group cluster, so it is not possible to add it to the grouo

iandarbey, 2021-03-02:
To clarify.... This switch (and potentially the 3 switch version) work as expected outside of this particular issue??

closed 2021-09-05 by github-actions[bot]