zigpy/zha-device-handlers #426: [Documentation] Documentation to help get devices added

jlongman, 2020-07-28:
Hey there, thanks in advance for what you do.

While trying to do #94 I had a hard time with two things:

- how to do a new device
- how to test the new files once added

The second is maybe simplest, the docs talk about using pypi as a source, (I've never done this but) do I push my code to pypi to validate whether what I'm doing works?  How would I do this?  (If it matters I'm in a hassio environment).

For the first, well I was trying to write this custom handling for the orvibo motion detector that works in zigbee2mqtt. I took an aqara device that had the same requirement to reset the state on a timer and copied a bunch off of it.  But it wasn't clear whether the MOTION AND IAS_ZONE was a choice/a feature to make integration or merely reflecting the devices zigbee reporting.

Additionally the aqara had a signature and a replacement which was referencing `cluster_id`s and `Clusters` _with_ `cluster_id`s so it was confusing. e.g.

```
    signature = {
# skip the SimpleDescriptor
        MODELS_INFO: [(LUMI, "lumi.sensor_motion.aq2")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    XIAOMI_CLUSTER_ID,
                    OccupancySensing.cluster_id, 
# etc
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    BasicCluster,
                    PowerConfigurationCluster,
                    Identify.cluster_id,
# etc
```

(Is that the point?  the first is used for matching the device and the second is what we present?)

Anyways, I think I made my point is that how to build one isn't clear.  Maybe there's some mandatory fields.  And I get that this probably isn't trivial and even harder to explain, but are there resources somewhere I'm missing maybe?

jlongman, 2020-07-28:
These are what I came up for what it's worth.  (The illuminance appears to be something I'm not sure is supported in the orvibo but maybe is in the aqara).

# orvibo/__init__.py
```
import asyncio
import logging
import math

from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
)
from zigpy.zcl.clusters.security import IasZone

from .. import Bus, LocalDataCluster
from ..const import (
    CLUSTER_COMMAND,
    OFF,
    ON,
    ZONE_STATE,
)

_LOGGER = logging.getLogger(__name__)

ORVIBO = "ORVIBO"
BATTERY_SIZE = "battery_size"
ZONE_TYPE = 0x0001
MOTION_TYPE = 0x000D

class OrviboCustomDevice(CustomDevice):
    """Custom device representing orvibo devices."""

    def __init__(self, *args, **kwargs):
        """Init."""
        self.battery_bus = Bus()
        if not hasattr(self, BATTERY_SIZE):
            self.battery_size = 10
        super().__init__(*args, **kwargs)

class MotionCluster(LocalDataCluster, IasZone):
    """Motion cluster."""

    cluster_id = IasZone.cluster_id

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self._timer_handle = None
        self.endpoint.device.motion_bus.add_listener(self)
        super()._update_attribute(ZONE_TYPE, MOTION_TYPE)

    def motion_event(self):
        """Motion event."""
        super().listener_event(CLUSTER_COMMAND, None, ZONE_STATE, [ON])

        _LOGGER.debug("%s - Received motion event message", self.endpoint.device.ieee)

        if self._timer_handle:
            self._timer_handle.cancel()

        loop = asyncio.get_event_loop()
        self._timer_handle = loop.call_later(120, self._turn_off)

    def _turn_off(self):
        _LOGGER.debug("%s - Resetting motion sensor", self.endpoint.device.ieee)
        self._timer_handle = None
        super().listener_event(CLUSTER_COMMAND, None, ZONE_STATE, [OFF])

class IlluminanceMeasurementCluster(CustomCluster, IlluminanceMeasurement):
    """Multistate input cluster."""
[... 138 lines cut ...]
                ],
            }
        },
    }

class IlluminanceMeasurementCluster(CustomCluster, IlluminanceMeasurement):
    """Multistate input cluster."""

    cluster_id = IlluminanceMeasurement.cluster_id
    ATTR_ID = 0

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self.endpoint.device.illuminance_bus.add_listener(self)

    def _update_attribute(self, attrid, value):
        if attrid == self.ATTR_ID and value > 0:
            value = 10000 * math.log10(value) + 1
        super()._update_attribute(attrid, value)

    def illuminance_reported(self, value):
        """Illuminance reported."""
        self._update_attribute(self.ATTR_ID, value)
```

timmo001, 2020-07-28:
My biggest concern is how do I test what I have added works locally before committing the changes to a PR. This may be less for those who have already contributed, but not for newbies to zha.

I have tried to point the pip add-on to my specific git repository but that doesn't seem to work

liorm0505, 2020-07-28:
Hi, 
If i want to test this code, should i put it in custom component folder?

dmulcahey, 2020-07-28:
@timmo001 are you using a venv for HA?

tube0013, 2020-07-28:
to test new/updated quirks with a supervised or hassos install:

You need to get access to the home-assistant docker container. I do this with the portainer add-on, you can get command line access in your browser within any container. 

[image]
 
I stage the updated/new quirk file(s) in a folder in my config directory. Open a console via portainer.

browse to the quirks install folder:  /usr/local/lib/python3.7/site-packages/zhaquirks/ or /usr/local/lib/python3.8/site-packages/zhaquirks/

go into directory of the manufacturer or create a new one, then copy the quirk files from /config/ to the location. , delete the __pycache__ folder, then close out the terminal and restart HA.

[image]

jlongman, 2020-07-29:
Trying this now, looks like I have more to do:
```
2020-07-28 20:56:54 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for ORVIBO 895a2d80097f4ae2b2d40500d5e03dcc (00:15:8d:00:01:bf:a5:c3)
2020-07-28 20:56:54 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.orvibo.sn10zw-1c0.MotionOrvibo'>
2020-07-28 20:56:54 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint
```
fixed that (I'll fix the above) and look at this some more.  It still seems to detecting motion constantly so it looks like more work to do.  IIRC the original groovy document had 15 seconds and the aqara example has 120.  Anyways, the device issue is separate, I'll maybe open another issue.

The addition of docker instructions is excellent, _thank you_.

But for example I don't know whether the `signature` and `replacement` structures should be different or why.  I also assume the difference between a cluster and device is some zigbee implied terminology.  Maybe a better question is, is there value in describing these?  Are there devices that can serve as models that explain this?

And once again, sorry if it's repetitive, but thanks for ZHA and the work, to all involved.

dmulcahey, 2020-07-29:
I’m gonna write a guide soon that breaks down a full existing quirk with explanations. I started mapping out what I want to cover today. Have you read what’s in the contributing section? It goes into light detail. In the simplest terms: `signature` is what the device is OOTB... `replacement` is what you want it to be.

dmulcahey, 2020-07-29:
@tube0013 any chance you wanna or that into the readme?

tube0013, 2020-07-29:
> @tube0013 any chance you wanna or that into the readme? 

Sure will work on it in the coming days

liorm0505, 2020-07-29:
> to test new/updated quirks with a supervised or hassos install:
>
> You need to get access to the home-assistant docker container. I do this with the portainer add-on, you can get command line access [...]

Thank you for the detailed explanation.
I will give it a try today.

liorm0505, 2020-07-29:
Hi @jlongman ,
Is there a way to get your latest code for testing as well ?

Thanks,

jlongman, 2020-08-01:
@liorm0505 I'm assuming your interest is the actual device support, so if you don't mind, I'll move that discussion to #94.  (If your point is more about taking that as an example for instructional purposes, then this would have been the correct place.)

I see @dmulcahey has started to [add to the contributing guide](https://github.com/zigpy/zha-device-handlers/commit/74d7845a0d5268caff0f482aa18da8f3ebf4a428) and already the replacement notes are helpful.

dmulcahey, 2020-08-03:
#431 is the first swing at a guide. LMK what you think.

PR #431 (2020-08-09): Update contributing guide
Fixes: #426
closed 2020-08-09 by dmulcahey