zigpy/zha-device-handlers #433: [Device Support Request]  ZNGZDJ11LM lumi.curtain.aq2

albalaing, 2020-08-06:
I'm doing a ZNGZDJ11LM quirks lumi.curtain.aq2
The position value comes in the AnalogOutput cluster. (0x000d) attribure 85 and in attribute 61440 it tells us if it is moving or not
The problem is that I do not know how to force the reading of the attribure 85 so that the position of the blind is updated every second. 
self.read_attributes (, [85,]) does nothing
Otherwise everything works fine and the position is updated when the movement stops.
I put a part of the code
```python
    class MyAnalogOutput (CustomCluster, AnalogOutput):

        manufacturer_attributes = {0xF000: ("moving", t.uint8_t),}
   
        def __init__(self, *args, **kwargs):
            """Init."""
            super().__init__(*args, **kwargs)
            self._timer_handle = None

        def _get_pos(self):
            self._timer_handle = None
            self.read_attributes (,[85,])
            if (self.__getitem__(61440)==64040192):
                loop = asyncio.get_event_loop()
                self._timer_handle = loop.call_later(1, self._get_pos)

        def _update_attribute(self, attrid, value):
            clus = self._endpoint.in_clusters[0x0102]
            if attrid==85:
                clus._update_attribute(8, value)
            if (attrid==61440) and (value == 64040192):    
                if self._timer_handle:
                    self._timer_handle.cancel()
                loop = asyncio.get_event_loop()
                clus._timer_handle = loop.call_later(1, self._get_pos)            
            super()._update_attribute(attrid, value)

    class MyWindowCovering (CustomCluster, WindowCovering):
   
        def _update_attribute(self, attrid, value):
            if (attrid==8): 
                if (self.__getitem__(0x0017)==1):  #reverse
                     value = 100 - value       
           super()._update_attribute(attrid, value)

```

dmulcahey, 2020-08-06:
You shouldn’t attempt to read it every second. Just use what is being sent by the device when it is sent.

albalaing, 2020-08-06:
The device does not send anything while moving. So the position would not be updated until it stopped. In the xiaomi application the position is updated while moving.

dmulcahey, 2020-08-06:
Are you sure it isn’t sending anything at all? And not just nothing on that cluster?

dmulcahey, 2020-08-06:
They also could be faking it on the front end.

albalaing, 2020-08-06:
I have put some logs
```python
    class MyAnalogOutput (CustomCluster, AnalogOutput):

        manufacturer_attributes = {0xF000: ("moving", t.uint8_t),}

        def __init__(self, *args, **kwargs):
            """Init."""
            super().__init__(*args, **kwargs)
            self._timer_handle = None

        def _get_pos(self):
            self._timer_handle = None
            _LOGGER.info("TIMER")
            self.read_attributes (self,[85,])
            if (self.__getitem__(61440)==64040192):
                loop = asyncio.get_event_loop()
                self._timer_handle = loop.call_later(1, self._get_pos)

        def _update_attribute(self, attrid, value):
            _LOGGER.info("AnalogOutput - Attrinute: %d Valuie: %d", attrid,value)
            clus = self._endpoint.in_clusters[0x0102]
            if attrid==85:
                clus._update_attribute(8, value)
            if (attrid==61440) and (value == 64040192):    
                if self._timer_handle:
                    self._timer_handle.cancel()
                loop = asyncio.get_event_loop()
                clus._timer_handle = loop.call_later(1, self._get_pos)            
            super()._update_attribute(attrid, value)

    class MyWindowCovering (CustomCluster, WindowCovering):
    
        def _update_attribute(self, attrid, value):
            if (attrid==8): 
                if (self.__getitem__(0x0017)==1):
                     value = 100 - value       
            _LOGGER.info("WindowCovering - Attrinute: %d Valuie: %d", attrid,value)
            super()._update_attribute(attrid, value)
```
this is the result of raising the blind
```
15:42:50 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] AnalogOutput - Attrinute: 85 Valuie: 2
15:42:50 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] WindowCovering - Attrinute: 8 Valuie: 2
15:42:50 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] AnalogOutput - Attrinute: 61440 Valuie: 64040192  
15:42:51 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] TIMER
15:42:52 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] TIMER
15:42:53 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] TIMER
15:42:54 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] TIMER
15:42:55 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] TIMER
15:42:56 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] TIMER
15:42:57 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] TIMER
15:42:58 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] TIMER
15:42:59 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] TIMER
15:43:00 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] TIMER
15:43:01 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] AnalogOutput - Attrinute: 85 Valuie: 22
15:43:01 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] WindowCovering - Attrinute: 8 Valuie: 22
15:43:01 INFO (MainThread) [zhaquirks.xiaomi.aqara.curtain_aq2] AnalogOutput - Attrinute: 61440 Valuie: 13708544   
```

Adminiuga, 2020-08-06:
for it to send the updates, you could try binding that cluster and configure attribute reporting for that attribute.
Although, most of the times legacy aqara was hardcoded to send updates.

albalaing, 2020-08-06:
I do not know how to do that.  Can you give me some example code?

closed 2021-08-13 by github-actions[bot]