zigpy/zha-device-handlers #435: [Device Support Request] Philips Hue Motion Sensor SML001

jvalst, 2020-08-10:
**Is your feature request related to a problem? Please describe.**
My ZHA won't recognize the Motion sensor SML001

**Describe the solution you'd like**
Support for the Hue Motion Sensor (SML001). 

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**

`light.keuken2: polling current state - from cache: False
[0x3ed7:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=145 command_id=Command.Read_Attributes_rsp>
[0x3ed7:11:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=147 command_id=Command.Read_Attributes_rsp>
[0x3ed7:11:0x0008]: received attribute: 0 update with value: 254
[0x3ed7:11:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=149 command_id=Command.Read_Attributes_rsp>
Device 0x7d79 (00:17:88:01:08:64:16:9b) joined the network
[0x7d79:zdo] ZDO request ZDOCmd.Device_annce: [0x7d79, 00:17:88:01:08:64:16:9b, 128]
[0x7d79] Requesting 'Node Descriptor'
Tries remaining: 2
[0x7d79] Extending timeout for 0x97 request
[0x7d79] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4107 maximum_buffer_size=89 maximum_incoming_transfer_size=63 server_mask=0 maximum_outgoing_transfer_size=63 descriptor_capability_field=0>
[0x7d79] Discovering endpoints
Tries remaining: 3
[0x7d79] Extending timeout for 0x99 request
[0x229e:1:0x0402] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=77 command_id=Command.Report_Attributes>
[0x229e:1:0x0402] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=2582>>]]
[0x229e:1:0x0402] Attribute report received: measured_value=2582
[0x229e:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=78 command_id=Command.Report_Attributes>
[0x229e:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=5771>>]]
[0x229e:1:0x0405] Attribute report received: measured_value=5771
[0x229e:1:0x0403] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=79 command_id=Command.Report_Attributes>
[0x229e:1:0x0403] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=int16s, value=1012>>, <Attribute attrid=20 value=<TypeValue type=int8s, value=-1>>, <Attribute attrid=16 value=<TypeValue type=int16s, value=10124>>]]
[0x229e:1:0x0403] Attribute report received: measured_value=1012, 20=-1, 16=10124
[0x7d79] Discovered endpoints: [2, 1]
[0x7d79:2] Discovering endpoint information
Tries remaining: 3
[0x7d79] Extending timeout for 0x9b request
[0x7d79:2] Discovered endpoint information: <Optional endpoint=2 profile=260 device_type=263 device_version=0 input_clusters=[0, 1, 3, 1030, 1024, 1026] output_clusters=[25]>
[0x7d79] Extending timeout for 0x9d request
[0x7d79] Extending timeout for 0x9f request
light.woonkamer: polling current state - from cache: False
[0xc749:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=161 command_id=Command.Read_Attributes_rsp>
[0xc749:11:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=163 command_id=Command.Read_Attributes_rsp>
[0xc749:11:0x0008]: received attribute: 0 update with value: 119
[0xc749:11:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=165 command_id=Command.Read_Attributes_rsp>
[0x7d79] Extending timeout for 0xa7 request
light.keuken1: polling current state - from cache: False
[0x44dc:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=169 command_id=Command.Read_Attributes_rsp>
[0x44dc:11:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=171 command_id=Command.Read_Attributes_rsp>
[0x44dc:11:0x0008]: received attribute: 0 update with value: 254
[0x44dc:11:0x0300] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=173 command_id=Command.Read_Attributes_rsp>
[0x7d79:2] Manufacturer: None
[0x7d79:2] Model: None
[0x7d79:1] Discovering endpoint information
Tries remaining: 3
[0x7d79] Extending timeout for 0xaf request
[0x7d79:1] Discovered endpoint information: <Optional endpoint=1 profile=49246 device_type=2128 device_version=2 input_clusters=[0] output_clusters=[0, 3, 4, 6, 8, 768, 5]>
[0x7d79] Extending timeout for 0xb1 request
[0x7d79:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=177 command_id=Command.Read_Attributes_rsp>
[0x7d79:1] Manufacturer: Philips
[0x7d79:1] Model: SML001
Checking quirks for Philips SML001 (00:17:88:01:08:64:16:9b)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 2}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1, 2}
[... 34 lines cut ...]
[0x7d79] Extending timeout for 0xbf request
[0x7d79:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0x7d79:1:0x0006]: finished channel configuration
[0x7d79] Extending timeout for 0xc1 request
[0x7d79:1:0x0005]: bound 'scenes' cluster: Status.SUCCESS
[0x7d79:1:0x0005]: finished channel configuration
[0x7d79] Extending timeout for 0xc3 request
[0x7d79:1:0x0300]: bound 'light_color' cluster: Status.SUCCESS
[0x7d79:1:0x0300]: finished channel configuration
[0x7d79] Extending timeout for 0xc5 request
[0x7d79:1:0x0000]: 'async_configure' stage succeeded
[0x7d79:1:0x0006]: 'async_configure' stage succeeded
[0x7d79:1:0x0008]: 'async_configure' stage succeeded
[0x7d79:1:0x0006]: 'async_configure' stage succeeded
[0x7d79:1:0x0005]: 'async_configure' stage succeeded
[0x7d79:1:0x0300]: 'async_configure' stage succeeded
[0x7d79:2:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0x7d79:2:0x0000]: finished channel configuration
[0x7d79] Extending timeout for 0xc7 request
[0x7d79:2:0x0402]: bound 'temperature' cluster: Status.SUCCESS
[0x7d79] Extending timeout for 0xc9 request
`

**Additional context**
As seen in the integrations pages (https://zigbee.blakadder.com/Philips_SML001.html), the motion sensor from Philips Hue isn't quite integrated in ZHA yet. The weird thing is, is that the outside sensor is set as integrated. Is it possible to have the indoor sensor working as well? My Bulbs from Philips are all connected flawless. Please tell me if you guys need any log or something and how to do it. Should be awesome to have that motion sensor included as well for everyone :)

Adminiuga, 2020-08-10:
What does not work?

jvalst, 2020-08-10:
Yeah I was send here from the other thread back in Home Assistant Github. You explained it there already (https://github.com/home-assistant/core/issues/38706). Weird thing now is, that after re-pairing the motion sensor, I suddenly see 5 entities showing. I don't have an outside hue motion sensor unfortunately, cause I see that one positive with ZHA on that support page (https://zigbee.blakadder.com/Philips_SML002.html). Can't try that one unfortunately, but that will work the same with the on/off you mentioned I guess. Any Idea why it now creates the 5 entities?

jvalst, 2020-08-14:
Don't see any zha_event popping up either when walking by the sensor. Hopefully there is a change this one will be completely integrated in zha. The Hue bridge is now only on for that sensor xD

Adminiuga, 2020-08-14:
what radio? 
Works fine with HUSBZB-1
[image]

Adminiuga, 2020-08-14:
per log it was configured correctly.

jvalst, 2020-08-17:
You also saw the succesfull message that a device was discovered? I didn't get that. I've a sonoff zigbee stick, CC2531.

Adminiuga, 2020-08-17:
The issue is in zigpy_cc. There are a few open issues for it. Let's continue it in https://github.com/zigpy/zigpy-cc/issues/57
Duplicate of https://github.com/zigpy/zigpy-cc/issues/57

closed 2020-08-17 by Adminiuga