zigpy/zha-device-handlers #437: [Device Support Request] Sonoff SNZB-01 wireless button

acsmth, 2020-08-13:
**Is your feature request related to a problem? Please describe.**
Sonoff SNZB-01 pairs, but is misidentified as "eWeLink WB01" and the only entity is battery level.

**Describe the solution you'd like**
Accurate identification and entities for short press, long press, and double-tap.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
`Device 0xe2bc (00:12:4b:00:1b:7c:d5:35) joined the network
[0xe2bc:zdo] ZDO request ZDOCmd.Device_annce: [0xe2bc, 00:12:4b:00:1b:7c:d5:35, 128]
[0xe2bc] Requesting 'Node Descriptor'
Tries remaining: 2
[0xe2bc] Extending timeout for 0x27 request
[0xe2bc] Node Descriptor: <Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=0 maximum_buffer_size=80 maximum_incoming_transfer_size=160 server_mask=0 maximum_outgoing_transfer_size=160 descriptor_capability_field=0>
[0xe2bc] Discovering endpoints
Tries remaining: 3
[0xe2bc] Extending timeout for 0x29 request
[0xe2bc] Discovered endpoints: [1]
[0xe2bc:1] Discovering endpoint information
Tries remaining: 3
[0xe2bc] Extending timeout for 0x2b request
[0xe2bc:1] Discovered endpoint information: <Optional endpoint=1 profile=260 device_type=0 device_version=0 input_clusters=[0, 3, 1] output_clusters=[6, 3]>
[0xe2bc] Extending timeout for 0x2d request
[0xe2bc:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=45 command_id=Command.Read_Attributes_rsp>
[0xe2bc:1] Manufacturer: eWeLink
[0xe2bc:1] Model: WB01
Checking quirks for eWeLink WB01 (00:12:4b:00:1b:7c:d5:35)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because endpoint list mismatch: {1, 2} {1}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because device_type mismatch on at least one endpoint
device - 0xe2bc:00:12:4b:00:1b:7c:d5:35 entering async_device_initialized - is_new_join: True
device - 0xe2bc:00:12:4b:00:1b:7c:d5:35 has joined the ZHA zigbee network
[0xe2bc](WB01): started configuration
[0xe2bc:ZDO](WB01): 'async_configure' stage succeeded
[0xe2bc] Extending timeout for 0x2f request
[0xe2bc] Extending timeout for 0x31 request
[0xe2bc] Extending timeout for 0x33 request
[0xe2bc:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0xe2bc:1:0x0000]: finished channel configuration
[0xe2bc] Extending timeout for 0x35 request
[0xe2bc:1:0x0001]: bound 'power' cluster: Status.SUCCESS
[0xe2bc] Extending timeout for 0x37 request
[0xe2bc:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xe2bc:1:0x0006]: finished channel configuration
[0xe2bc:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=53 command_id=Command.Read_Attributes_rsp>
[0xe2bc:1:0x0000]: initializing channel: from_cache: False
[0xe2bc:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=55 command_id=Command.Configure_Reporting_rsp>
[0xe2bc:1:0x0001]: reporting 'battery_voltage' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0xe2bc] Extending timeout for 0x39 request
[0xe2bc:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=57 command_id=Command.Configure_Reporting_rsp>
[0xe2bc:1:0x0001]: reporting 'battery_percentage_remaining' attr on 'power' cluster: 3600/10800/1: Result: '[[<ConfigureReportingResponseRecord status=Status.SUCCESS>]]'
[0xe2bc:1:0x0001]: finished channel configuration
[0xe2bc:1:0x0000]: 'async_configure' stage succeeded
[0xe2bc:1:0x0001]: 'async_configure' stage succeeded
[0xe2bc:1:0x0006]: 'async_configure' stage succeeded
[0xe2bc](WB01): completed configuration
[0xe2bc](WB01): stored in registry: ZhaDeviceEntry(name='eWeLink WB01', ieee='00:12:4b:00:1b:7c:d5:35', last_seen=1597280794.3809981)
[0xe2bc] Extending timeout for 0x3b request
[0xe2bc:1:0x0003] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=59 command_id=Command.Default_Response>
[0xe2bc:1:0x0003]: executed 'trigger_effect' command with args: '(2, 0)' kwargs: '{}' result: [64, <Status.UNSUP_CLUSTER_COMMAND: 129>]
[0xe2bc](WB01): started initialization
[0xe2bc:ZDO](WB01): 'async_initialize' stage succeeded
[0xe2bc] Extending timeout for 0x3d request
[0xe2bc] Extending timeout for 0x3f request
[0xe2bc:1:0x0006]: initializing channel: from_cache: False
[0xe2bc:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=61 command_id=Command.Read_Attributes_rsp>
[0xe2bc:1:0x0000]: initializing channel: from_cache: False
[0xe2bc:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=63 command_id=Command.Read_Attributes_rsp>
[0xe2bc:1:0x0001]: initializing channel: from_cache: False
[0xe2bc] Extending timeout for 0x41 request
[0xe2bc:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=65 command_id=Command.Read_Attributes_rsp>
[0xe2bc:1:0x0000]: 'async_initialize' stage succeeded
[0xe2bc:1:0x0001]: 'async_initialize' stage succeeded
[0xe2bc:1:0x0006]: 'async_initialize' stage succeeded
[0xe2bc](WB01): power source: Battery or Unknown
[0xe2bc](WB01): completed initialization`

**Additional context**
Device support in zigbee2mqtt is documented [here](https://www.zigbee2mqtt.io/devices/SNZB-01.html)

Adminiuga, 2020-08-13:
It is not misidentified, this is what device reports for it's model and manufacturer.
This is a switch/button and therefore it is not supposed to have any entities in HA. Those are stateless devices and they fire "zha_event" events

Adminiuga, 2020-08-13:
Go to developers -> events and listen to zha_event then press button

acsmth, 2020-08-13:
> Go to developers -> events and listen to zha_event then press button

Got it, thanks for the explanation.

closed 2020-08-13 by acsmth

docgalaxyblock, 2020-10-28:
@Adminiuga Could you please create a quirk for this device that the buttons will be supported at the automation wizzard?
Thx