zigpy/zha-device-handlers #438: [Device Support Request] Xiaomi Mija Switch

JeffResc, 2020-08-13:
**Is your feature request related to a problem? Please describe.**
I believe this device is already supported, so I didn't mark it as a bug, but perhaps it is if it's supposed to be supported. I found [this file](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xiaomi/mija/sensor_switch.py) which shows that it might be supported. I purchased the device [here on AliExpress](https://www.aliexpress.com/item/32968741338.html).

**Describe the solution you'd like**
I'd like clarification on if this device is compatiable, because I could not successfully pair it.

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
```
[0xff15] Requesting Node Descriptor failed
[0xff15] Discovering endpoints
Tries remaining: 3
[0xff15] Extending timeout for 0x17 request
Tries remaining: 2
[0xff15] Extending timeout for 0x19 request
Device 0xe6d6 (00:15:8d:00:04:01:06:a5) joined the network
Device 00:15:8d:00:04:01:06:a5 changed id (0xff15 => 0xe6d6)
Canceling old initialize call
[0xe6d6:zdo] ZDO request ZDOCmd.Device_annce: [0xe6d6, 00:15:8d:00:04:01:06:a5, 128]
[0xe6d6] Requesting 'Node Descriptor'
Tries remaining: 2
[0xe6d6] Extending timeout for 0x1b request
Ignoring message (b'18000a050042126c756d692e73656e736f725f737769746368') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18010a0100200a') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'1c3412020a02ff4c0600100021ec0b21a8012400000000002133892052') on cluster 0: unknown endpoint or cluster id: 1
Device 0xaae6 (00:15:8d:00:04:01:06:a5) joined the network
Device 00:15:8d:00:04:01:06:a5 changed id (0xe6d6 => 0xaae6)
Canceling old initialize call
[0xaae6:zdo] ZDO request ZDOCmd.Device_annce: [0xaae6, 00:15:8d:00:04:01:06:a5, 128]
[0xaae6] Requesting 'Node Descriptor'
Tries remaining: 2
[0xaae6] Extending timeout for 0x1d request
Ignoring message (b'18000a050042126c756d692e73656e736f725f737769746368') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'18010a0100200a') on cluster 0: unknown endpoint or cluster id: 1
Ignoring message (b'1c3412020a02ff4c0600100021ec0b21a8012400000000002133892053') on cluster 0: unknown endpoint or cluster id: 1
Tries remaining: 1
[0xaae6] Extending timeout for 0x1f request
[0xaae6] Requesting Node Descriptor failed
[0xaae6] Discovering endpoints
Tries remaining: 3
[0xaae6] Extending timeout for 0x21 request
```

**Additional context**
Here are the codes received when the button is interacted with:

Button pressed: `Ignoring message (b'18030a00001000') on cluster 6: unknown endpoint or cluster id: 1` (used for single press or long press)
Button released: `Ignoring message (b'18040a00001001') on cluster 6: unknown endpoint or cluster id: 1` (used for single press or long press)
Double press: `Ignoring message (b'180b0a00802002') on cluster 6: unknown endpoint or cluster id: 1`
Triple press: `Ignoring message (b'180c0a00802003') on cluster 6: unknown endpoint or cluster id: 1`
Quadrouple press: `Ignoring message (b'180d0a00802004') on cluster 6: unknown endpoint or cluster id: 1`
Five or more times press: `Ignoring message (b'180e0a00802080') on cluster 6: unknown endpoint or cluster id: 1`

fluffymadness, 2020-08-13:
From my experience with these buttons....the pairing with these is extremely tricky -.-
Press the reset button for a long time, when the pairing sequence is initiated try to do short presses of the reset button all 2-3 seconds. And press the button from time to time. Do this for 20-30 seconds, and if you are lucky there will appear some messages in the pairing log that button presses are registered and the button will pair.
If it paired only partly, remove the device from zha and try to pair again. Sometimes I needed about 2-3 tries for these buttons with the above procedure for getting them paired.

fluffymadness, 2020-08-13:
Many of the xiaomi stuff is not 100% zigbee compliant. 
You can get a good overview about these problems here
https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623

pipiche38, 2020-08-13:
@JeffResc  Could you clarify which Mija switch it is ? 
Is that the small round button ? If that is the case indeed it doesn't respond to node descriptor

JeffResc, 2020-08-13:
@pipiche38 It is the small button one. If you check the original post, I linked the original AliExpress link and the product description and images are correct on that page.

pipiche38, 2020-08-13:
Ok. So indeed that one is a very special one, as it just send it model name via a reporting attribute 0x0005 on cluster 0x0000, and then will send a report attribute on cluster 0x0006 with the following values:
one click 0x00 an then 0x01
 two clicks 0x02
three clicks 0x03 
long click 0x00, then 0x02 when release

But do not expect to make any Discovery

JeffResc, 2020-09-28:
CC2531 was in software mode, not hardware mode. Putting it in hardware mode fixed my issue.

closed 2020-09-28 by JeffResc