zigpy/zha-device-handlers #439: [Device Support Request] Lonsonho 2 Gang wall switch

scaytrase, 2020-08-13:
**Is your feature request related to a problem? Please describe.**
Device almost works good without quirk but HA cannot control second (right) switch at the moment

**Describe the solution you'd like**
Control second switch as the first one. Currently issuing any commands to it results in timout error (stactrace below)

**Device signature - this can be acquired by removing the device from ZHA and pairing it again from the add devices screen. Be sure to add the entire content of the log panel after pairing the device to a code block below this line.**
Full log
```
Device 0xa474 (bc:33:ac:ff:fe:65:5c:e2) joined the network
[0xa474:zdo] ZDO request ZDOCmd.Device_annce: [0xa474, bc:33:ac:ff:fe:65:5c:e2, 142]
[0xa474] Requesting 'Node Descriptor'
Tries remaining: 2
[0xa474] Extending timeout for 0x15 request
[0xa474] Node Descriptor: <Optional byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=0 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>
[0xa474] Discovering endpoints
Tries remaining: 3
[0xa474] Discovered endpoints: [1, 2]
[0xa474:1] Discovering endpoint information
Tries remaining: 3
[0xa474:1] Discovered endpoint information: <Optional endpoint=1 profile=260 device_type=256 device_version=0 input_clusters=[0, 3, 4, 5, 6] output_clusters=[25]>
[0xa474:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=27 command_id=Command.Read_Attributes_rsp>
[0xa474:1] Manufacturer: _TYZB01_6sadkhcy
[0xa474:1] Model: TS0002
[0xa474:2] Discovering endpoint information
Tries remaining: 3
[0xa474:2] Discovered endpoint information: <Optional endpoint=2 profile=260 device_type=256 device_version=0 input_clusters=[0, 3, 4, 5, 6] output_clusters=[25]>
Checking quirks for _TYZB01_6sadkhcy TS0002 (bc:33:ac:ff:fe:65:5c:e2)
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 2}
Considering <class 'zhaquirks.lutron.lzl4bwhl01remote.LutronLZL4BWHL01Remote2'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.osram.a19twhite.A19TunableWhite'>
Fail because endpoint list mismatch: {3} {1, 2}
Considering <class 'zhaquirks.philips.rom001.PhilipsROM001'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.philips.rwl020.PhilipsRWL020'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.philips.rwl021.PhilipsRWL021'>
Fail because profile_id mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 2}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2}
Considering <class 'zhaquirks.xiaomi.mija.smoke.MijiaHoneywellSmokeDetectorSensor'>
Fail because endpoint list mismatch: {1} {1, 2}
device - 0xa474:bc:33:ac:ff:fe:65:5c:e2 entering async_device_initialized - is_new_join: True
device - 0xa474:bc:33:ac:ff:fe:65:5c:e2 has joined the ZHA zigbee network
[0xa474](TS0002): started configuration
[0xa474:ZDO](TS0002): 'async_configure' stage succeeded
[0xa474:1:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xa474:1:0x0000]: bound 'basic' cluster: Status.SUCCESS
[0xa474:1:0x0000]: finished channel configuration
[0xa474:1:0x0019]: bound 'ota' cluster: Status.SUCCESS
[0xa474:1:0x0019]: finished channel configuration
[0xa474:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=37 command_id=Command.Default_Response>
[0xa474:1:0x0006]: reporting 'on_off' attr on 'on_off' cluster: 0/900/1: Result: '[6, <Status.UNSUP_GENERAL_COMMAND: 130>]'
[0xa474:1:0x0006]: finished channel configuration
[0xa474:1:0x0000] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=39 command_id=Command.Read_Attributes_rsp>
[0xa474:1:0x0000]: initializing channel: from_cache: False
[0xa474:2:0x0006]: bound 'on_off' cluster: Status.SUCCESS
[0xa474:1:0x0006]: 'async_configure' stage succeeded
[... 93 lines cut ...]
Logger: homeassistant.components.websocket_api.http.connection.281473257193424
Source: components/zha/core/device.py:549
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 22:04:57 (2 occurrences)
Last logged: 22:08:13

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 130, in handle_call_service
    connection.context(msg),
  File "/usr/src/homeassistant/homeassistant/core.py", line 1260, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1295, in _execute_service
    await handler.func(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 518, in admin_handler
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 925, in issue_zigbee_cluster_command
    manufacturer=manufacturer,
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 549, in issue_cluster_command
    command, *args, manufacturer=manufacturer, expect_reply=True
  File "/usr/local/lib/python3.7/site-packages/zigpy/device.py", line 210, in request
    result = await asyncio.wait_for(req.result, timeout)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 449, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```

Adminiuga, 2020-08-14:
what radio?
it can't communicate with the switch
```
 [0xa474:2:0x0006]: failed to get attributes '['on_off']' on 'on_off' cluster: 
...
[0xa474:1:0x0006]: failed to get attributes '['on_off']' on 'on_off' cluster: 
```

scaytrase, 2020-08-14:
сс2531. i've tried zigbee2mqtt (linked the config) and it works perfectly on the same radio, both endpoints.

Adminiuga, 2020-08-14:
enable debug log and submit the logs. This could be a problem with zigpy-cc