zigpy/zha-device-handlers #443: [Device Support Request] Tuya Water Sensor

meremortals70, 2020-08-17:
Could we please have support for this device added?

[image]

{
  "node_descriptor": "<Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=0 maximum_buffer_size=82 maximum_incoming_transfer_size=255 server_mask=11264 maximum_outgoing_transfer_size=255 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYST11_qq9mpfhw",
  "model": "q9mpfhw",
  "class": "zigpy.device.Device"
}

FrnchFrgg, 2020-10-24:
If you attach a debug log of what the sensor sends when it detects water and stops detecting it, I might be able to write a patch for it, that you would have to test afterwards. Of course the support wouldn't be as complete as if I had the device, but for just a simple binary sensor I might be able to pull it off. Sadly there seems to be no support for it where I looked for the siren support.

kukachik, 2021-05-03:
Start detection and stop
```
2021-05-03 17:25:06 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef01014001000071ffe4fe9effff0a09630200026604000100'
2021-05-03 17:25:06 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260,
 clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=113), 255, -28, 0x9efe, 255, 255, b'\tc\x02\x00\x02f\x04\x00\x01\x00']
2021-05-03 17:25:06 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef01014001000072ffe3fe9effff0a09640200046501000101'
2021-05-03 17:25:06 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260,
 clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=114), 255, -29, 0x9efe, 255, 255, b'\td\x02\x00\x04e\x01\x00\x01\x01']

2021-05-03 17:25:12 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef01014001000073ffe4fe9effff0a09650200026604000100'
2021-05-03 17:25:12 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260,
 clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=115), 255, -28, 0x9efe, 255, 255, b'\te\x02\x00\x02f\x04\x00\x01\x00']
2021-05-03 17:25:12 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040100ef01014001000074ffe3fe9effff0a09660200046501000100'
2021-05-03 17:25:12 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260,
 clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=116), 255, -29, 0x9efe, 255, 255, b'\tf\x02\x00\x04e\x01\x00\x01\x00']
```

kukachik, 2021-05-03:
Maybe we can add  _TYST11_qq9mpfhw by simple copy TS0207 BlitzWolf water sensor ?

closed 2021-11-11 by github-actions[bot]

pmffromspace, 2024-06-17:
I got the same sensor TS0207 from _TZ3000_k4ej3ww2 but its still not correctly detected. Any news on this?