zigpy/zha-device-handlers #444: [Device Support Request]

meremortals70, 2020-08-17:
Could we please have support for this device added please?

[image]

{
  "node_descriptor": "<NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=0 maximum_buffer_size=82 maximum_incoming_transfer_size=255 server_mask=11264 maximum_outgoing_transfer_size=255 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYST11_d0yu2xgi",
  "model": "0yu2xgi",
  "class": "zigpy.device.Device"
}

FrnchFrgg, 2020-09-19:
This alarm has support implemented there: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L5824
With PR #484 recently merged, it shouldn't be too hard to write that support.

As I recently bought that device, I'll try and write the quirks as soon as I have enough time to dive in (which sadly might not be very soon).

meremortals70, 2020-09-25:
> This alarm has support implemented there: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L5824
> With PR #484 recently merged, it shouldn't be too hard to [...]

Appreciate your help

FrnchFrgg, 2020-10-18:
For now, the device can be used by manipulating the attributes in the TuyaManufClusterSiren cluster. Their name should be straightforward.

I'll clean up the patch first (making tox happy because it seems it doesn't like my formatting).

Then I will try to find standard clusters that are relevant to that siren and map the information and commands to them.

If someone has information as to how tuya "get_data" should be used to actually get some data back, I'm all ears.

FrnchFrgg, 2020-10-23:
The liked MR makes the tuya siren available as an On/Off device, a Temperature sensor and an humidity sensor. Other attributes (melody, sound duration, automatic alarms, etc.) are available as attributes on the Tuya cluster, and are persistent.
A followup might implement IAS warning device clusters, but that would entail saving the current melody, setting it to a specific "alarm-like" melody, probably changing the maximum duration, then activating the alarm, and reverting back the settings to what they were before, which might be overkill for a zhaquirks implementation.

commit cae0d371 (2020-11-02): Support for Tuya-based NEO Siren (closes #444) (#530)

* Initial support for Tuya-based NEO Siren

Also implement a base class to map non-standard tuya data to attributes.
Custom tuya clusters can derive from that class and any defined
attribute on the cluster will be automatically mapped.

* Implement temperature and humidity clusters

* Implement the On/Off cluster

* Update documentation
closed 2020-11-02 by Adminiuga