zigpy/zha-device-handlers #445: [Device Support Request]

meremortals70, 2020-08-17:
Could we please have support for this device addd?

[image]

{
  "node_descriptor": "<NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=0 maximum_buffer_size=82 maximum_incoming_transfer_size=255 server_mask=11264 maximum_outgoing_transfer_size=255 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYST11_7hfcudw5",
  "model": "hfcudw5",
  "class": "zigpy.device.Device"
}

Adminiuga, 2020-08-17:
@dmulcahey Tuya devices are worst than Xiaomi.

meremortals70, 2020-08-17:
No problems in dumping them is they are an issue, except for the siren, I need that 😀

dmulcahey, 2020-08-17:
Let’s see what we can do... hold on to them for now!

m0bygit, 2020-08-31:
Support for this device would indeed be nice. 
Mine is a different brand, but I'm guessing the only difference between my unit and the op's is the manufacturer and model strings?
Here is a description for my Blitzwolf branded device (BW-IS3)

https://www.blitzwolf.com/Rechargeable-ZigBee-PIR-Motion-Sensor-p-442.html

```
{
  "node_descriptor": "<Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=0 maximum_buffer_size=82 maximum_incoming_transfer_size=255 server_mask=11264 maximum_outgoing_transfer_size=255 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYST11_i5j6ifxj",
  "model": "5j6ifxj",
  "class": "zigpy.device.Device"
}
```

Zigbee2mqtt and ioBroker.zigbee seems to support these devices.

Thanks for a great project.

Adminiuga, 2020-08-31:
Enable debug logging as per zha documentation, pair module and try trigger the motion and see if you get anything in the logs.

m0bygit, 2020-08-31:
Sure thing. I think it's this?

`2020-08-31 19:07:49 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]

2020-08-31 19:07:49 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)

2020-08-31 19:07:49 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [33, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xef67>, 1, 260, 61184, b'\t{\x01\x00\x1e\x03\x04\x00\x01\x02', 0, 175, 255, 206, 144, 59, 0, -46]

2020-08-31 19:07:49 DEBUG (MainThread) [zigpy.device] Ignoring message (b'097b01001e0304000102') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (ec:1b:bd:ff:fe:30:2d:74, 1)'

2020-08-31 19:07:49 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xef67>, ep: 1, profile: 0x0104, cluster_id: 0xef00, data: b'097b01001e0304000102'`

Adminiuga, 2020-08-31:
well, if it is a message for the motion detection, then they are using a manufacturer specific cluster and not a standard IAS Zone cluster. These manufacturers want to lock you in their ecosystem. Vote with your money and don't buy their products???

m0bygit, 2020-08-31:
Yes they do, but I don't want to get locked into the Blitzwolf ecosystem (obviously). That's why I'm here. Reading the description of this project (ZHA Device Handlers) lead me to believe that this was the whole point of the project, to fix custom behaviors outside the zigbee specs ("Zigbee devices that deviate from or do not fully conform to the standard specifications..." ).

pipiche38, 2020-09-01:
But that is true to most of the zigbee products manufacturers, most do something different, and that is why why we have project like zigpy, zigate, deConz....

closed 2020-09-05 by Adminiuga

m0bygit, 2020-09-05:
Excellent @Adminiuga, thanks!