zigpy/zha-device-handlers #446: [Device Support Request] Orvibo AM25 Blind Controller

meremortals70, 2020-08-17:
Could we please have support for this device added?

[image]

{
  "node_descriptor": "<Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4633 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0102"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "ORVIBO",
  "model": "2a103244da0b406fa51410c692f79ead",
  "class": "zigpy.device.Device"
}

dmulcahey, 2020-08-18:
What isn’t working?

meremortals70, 2020-08-18:
They are meant to have a number of positions available for setting other than open and closed.

iskobkarev, 2021-03-18:
I've got 6 of these blinds motors and all disconnect from the network after a day or two. Power cycle brings them back.
They work fine when connected and can be registered no problems. 

Here is the signature:
{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4633, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0102"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "ORVIBO",
  "model": "2a103244da0b406fa51410c692f79ead",
  "class": "zigpy.device.Device"
}

iskobkarev, 2021-05-04:
Hello there

The motors still disconnect after a number of days - seem to be quire random. Otherwise it is a very nice device.
The power cycle or initiation of the ZB join sequence (programming button: 4 short-1 long) returns it back.

what I've tried
 - adding light switches/routers nearby
 - moving wifi to different channels
 - upgrading the coordinator EZSP = Silicon Labs EmberZNet protocol: Elelabs, HUSBZB-1, Telegesis by ZHA , to 6.7.8

ready to bet $100 to whoever can fix it !! 

Here is the Orvibo AM25 re-join log for the HA core-2021.5.0b7

Sent 'mgmt_permit_joining_req' to 04:cf:8c:df:3c:76:22:28: [<Status.SUCCESS: 0>]
Device 0xe814 (08:6b:d7:ff:fe:1e:c1:cd) joined the network
Device 08:6b:d7:ff:fe:1e:c1:cd changed id (0xc78f => 0xe814)
[0xe814] Extending timeout for 0xdf request
Checking quirks for ORVIBO 2a103244da0b406fa51410c692f79ead (08:6b:d7:ff:fe:1e:c1:cd)
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}
Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
Fail because device_type mismatch on at least one endpoint
Device 0xe814 (08:6b:d7:ff:fe:1e:c1:cd) joined the network
Skip initialization for existing device 08:6b:d7:ff:fe:1e:c1:cd
[0xe814] Cancelling old group rescan
device - 0xe814:08:6b:d7:ff:fe:1e:c1:cd entering async_device_initialized - is_new_join: False
device - 0xe814:08:6b:d7:ff:fe:1e:c1:cd has been reset and re-added or its nwk address changed
skipping discovery for previously discovered device - 0xe814:08:6b:d7:ff:fe:1e:c1:cd
[0xe814](2a103244da0b406fa51410c692f79ead): started configuration
[0xe814:ZDO](2a103244da0b406fa51410c692f79ead): 'async_configure' stage succeeded
[0xe814] Extending timeout for 0xe1 request
[0xe814:1:0x0000]: finished channel configuration
[0xe814] Extending timeout for 0xe3 request
[0xe814] Extending timeout for 0xe5 request
[0xe814:1:0x0019]: finished channel configuration
Device 08:6b:d7:ff:fe:1e:c1:cd already exists. Updating it.
Device 0xe814 (08:6b:d7:ff:fe:1e:c1:cd) joined the network
Skip initialization for existing device 08:6b:d7:ff:fe:1e:c1:cd
[0xe814] Cancelling old group rescan
[0xe814:zdo] ZDO request ZDOCmd.Device_annce: [0xE814, 08:6b:d7:ff:fe:1e:c1:cd, 128]
[0xe814] Extending timeout for 0xe8 request
[0xe814](2a103244da0b406fa51410c692f79ead): started initialization
[0xe814:ZDO](2a103244da0b406fa51410c692f79ead): 'async_initialize' stage succeeded
[0xe814:1:0x0000]: initializing channel: from_cache: False
[0xe814:1:0x0000]: finished channel configuration
[0xe814:1:0x0001]: initializing channel: from_cache: False
[0xe814] Extending timeout for 0xea request
[0xe814:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=234 command_id=Command.Read_Attributes_rsp>
[0xe814:1:0x0004] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=225 command_id=2>
[0xe814:1:0x0004] ZCL request 0x0002: [255, []]
[0xe814:1:0x0004] No handler for cluster command 2
[0xe814] Extending timeout for 0xec request
[0xe814:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=236 command_id=Command.Read_Attributes_rsp>
[0xe814:1:0x0001]: finished channel configuration
[0xe814:1:0x0102]: initializing channel: from_cache: False
[0xe814] Extending timeout for 0xee request
[0xe814:1:0x0102] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=238 command_id=Command.Read_Attributes_rsp>
[0xe814:1:0x0102]: Attribute report 'Window Covering'[current_position_lift_percentage] = 100
[0xe814:1:0x0102]: finished channel configuration
[... 45 lines cut ...]
[0xe814:1:0x0003] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=253 command_id=Command.Default_Response>
[0xe814:1:0x0003]: executed 'trigger_effect' command with args: '(2, 0)' kwargs: '{}' result: [64, <Status.UNSUP_CLUSTER_COMMAND: 129>]
[0xe814](2a103244da0b406fa51410c692f79ead): started initialization
[0xe814:ZDO](2a103244da0b406fa51410c692f79ead): 'async_initialize' stage succeeded
[0xe814:1:0x0000]: initializing channel: from_cache: False
[0xe814:1:0x0000]: finished channel configuration
[0xe814:1:0x0001]: initializing channel: from_cache: False
[0xe814] Extending timeout for 0x00 request
[0xe814:1:0x0102]: initializing channel: from_cache: False
[0xe814] Extending timeout for 0x02 request
Duplicate 2 TSN
[... cut ...]
[0xe814:1:0x0001] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=3 command_id=Command.Read_Attributes_rsp>
[0xe814:1:0x0001]: finished channel configuration
[0xe814:1:0x0000]: 'async_initialize' stage succeeded
[0xe814:1:0x0001]: 'async_initialize' stage succeeded
[0xe814:1:0x0102]: 'async_initialize' stage succeeded
[0xe814:1:0x0019]: 'async_initialize' stage succeeded
[0xe814](2a103244da0b406fa51410c692f79ead): power source: Battery or Unknown
[0xe814](2a103244da0b406fa51410c692f79ead): completed initialization

zegelin, 2021-09-19:
@iskobkarev I too am experiencing the disconnect issue. I purchased a handful of these motors earlier this year and have been trying on-and-off for months to get the Zigbee connection working reliably.

I'm certain that the disconnect issue is not bug with zigpy, as the problem also occurs with OpenHAB's ZigBee component and zigbee2mqtt.

I've also tried two different coordinators: A Nortek HUSBZB-1 (Ember-based) and a Electrolama ZZH! (CC2652R-based).

With zigbee2mqtt I'm also experiencing slow response times and out-of-order position updates.

Sadly I think the issue is in the firmware of these devices. I contacted the AliExpress seller that I purchased these motors from and their response was essentially that these motors only work with TuYa and that I should use a TuYa hub for compatibility.

See my discussion over on zigbee2mqtt: https://github.com/Koenkk/zigbee2mqtt/discussions/8795

iskobkarev, 2021-09-19:
> See my discussion over on zigbee2mqtt: Koenkk/zigbee2mqtt#8795

Hi Adam

I've managed to get these things to work a bit more stable with the motor
staying online for a few months in a row as compared to a few days.

I upgraded the HUSBZB-1 to the latest ember code and moved it to channel
15. I had to re-register all the devices to join the new channel but they
came up with the same name and there were no hass configuration changes
needed. Also, I have at least one zigbee router (a switch or an outlet) on
every floor for better comms.

I can't tell if some of my hass upgrades cause them to drop off but I've
noticed that if I don't upgrade/reboot they stay online for longer.

I even contacted Orvibo support but their response was that only their hub
should be used and they refused to look into it. Well that is sad, these
are very functional devices.I use the curtain battery-supported motors from
Orvibo and these work flawlessly with Hass never disconnecting. So yes I
agree it is very likely a firmware issue on these blinds motors.

Also, there is a new configuration parameter under HubZ Smart Home
Controller which defines unavailability timer for battery and main operated
devices. I set it to 24 hours and I'll monitor if that makes any changes in
behaviour.

Thanks

Igor

closed 2022-03-25 by github-actions[bot]