zigpy/zha-device-handlers #450: [Device Support Request] Sercomm SZ-DWS04 Door/Window Sensor

meremortals70, 2020-08-18:
Please add support for the following device?

[image]

{
  "node_descriptor": "<NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4401 maximum_buffer_size=82 maximum_incoming_transfer_size=255 server_mask=0 maximum_outgoing_transfer_size=255 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Sercomm Corp.",
  "model": "SZ-DWS04",
  "class": "zigpy.device.Device"
}

dmulcahey, 2020-08-18:
The contact sensor works OOTB. A quirk can be added for battery.

meremortals70, 2020-08-18:
Thanks yes Battery and tamper alarm is missing

Gast0n87, 2021-05-15:
I have this contact sensor and it's missing the Battery function. Any help pointing me in the right direction?

closed 2021-11-18 by github-actions[bot]

bverkron, 2021-11-30:
> I have this contact sensor and it's missing the Battery function. Any help pointing me in the right direction?

Did you have any luck with this? I’d like to create a quirk for this but getting setup for development wasn’t as straightforward as I expected and I’ve been short on time to research further.