zigpy/zha-device-handlers #451: [Device Support Request] Visonic MP-840

meremortals70, 2020-08-18:
Please add support for this device?

[image]

{
  "node_descriptor": "<NodeDescriptor byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4113 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Visonic",
  "model": "MP-840",
  "class": "zigpy.device.Device"
}

dmulcahey, 2020-08-18:
Can you describe what isn’t working? Quirks are only necessary if there is device functionality that isn’t working.

meremortals70, 2020-08-18:
Thanks, I done get battery reporting or tamper alarm.

dmulcahey, 2020-08-18:
I don’t think we support tamper alarm anywhere yet. We can look into that.

meremortals70, 2020-08-18:
Thanks David

meremortals70, 2020-08-18:
I’m trying where possible to upgrade my sensors to professional sensors instead of the cheaper Tuya type devices.