zigpy/zha-device-handlers #452: [Device Support Request] Sercomm SZ-ESW01-AU

meremortals70, 2020-08-18:
Please add support for this device?

[image]

{
  "node_descriptor": "<NodeDescriptor byte1=1 byte2=64 mac_capability_flags=142 manufacturer_code=4401 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=0 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003",
        "0x000a",
        "0x0019",
        "0x0b05"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006"
      ]
    }
  },
  "manufacturer": "Sercomm Corp.",
  "model": "SZ-ESW01-AU",
  "class": "zigpy.device.Device"
}

dmulcahey, 2020-08-18:
I have the US version of this.. what isn’t working? These work OOTB from what I can see...

meremortals70, 2020-08-18:
Thanks, there are three entities enabled out of the box but one is not reporting. I have 11 of the. There are also other attributes I would like to get exposed and not sure if there is a cluster to set the switch as always on

dmulcahey, 2020-08-18:
Can you elaborate? Please be very specific. There are 3 entities... ok which ones? And which one isn’t reporting? Please detail the attributes that aren’t exposed that you would like exposed etc. Please provide as much detail as possible.

meremortals70, 2020-08-18:
Thanks let me reply tomorrow when I am not on an iPad where I can provide screenshots etc.

Adminiuga, 2020-08-18:
What radio?

meremortals70, 2020-08-18:
Conbee II

closed 2021-08-13 by github-actions[bot]