zigpy/zha-device-handlers #457: [BUG/DISCUSSION] IKEA RGB Tradfri Lights & Color Temperature

TheJulianJES, 2020-08-21:
**Describe the bug**
IKEA RGB lights (with color temperature support) do not display a color temperature slider when using the ZHA implementation.

**To Reproduce**
Steps to reproduce the behavior:
1. Get a CWS bulb from IEKA (RGB and color temperature)
2. See that there's no color temperature slider in Home Assistant.

**Expected behavior**
That there is a color temperature slider in the Home Assistant frontend present (like in other apps: native TRADFRI component or app).

**Additional context**
(Previous context in: https://github.com/zigpy/zha-device-handlers/pull/243#issuecomment-677910290)

Issue in Home Assistant repo was closed due to inactivity:
https://github.com/home-assistant/core/issues/16143

It was already mentioned that such a "hack" (to convert color temperature to hue/sat) would likely not be copied: https://github.com/home-assistant/core/issues/16143#issuecomment-489598830

The conversion in the native TRADFRI component in Home Assistant can be found here: https://github.com/home-assistant/core/blob/dev/homeassistant/components/tradfri/light.py#L234-L247

Afaik, zigbee2mqtt also has a conversion for this process(?).

I think there could be some discussion in regards to how and where this could be implemented. (Would like to try myself)

wafliron, 2020-11-02:
@TheJulianJES did you manage to make any progress with this? I've got three of these bulbs and it's a real frustration that their white spectrum value can't be controlled in ZHA (faked or otherwise).

TheJulianJES, 2020-11-02:
@wafliron Sorry, I didn't continue working on this. I got the the color temperature slider to show up in Home Assistant by using a constant attribute for color capabilities, but I didn't work on correctly "translating" the color temperatures to RGB for the bulb (and back).
(Just returned my RGB IKEA bulbs for now (except one for testing) and bought IKEA color temperature bulbs or Hues for color (they have nicer colors)).
I'll post an update here when/if I work on this in the future again.

wafliron, 2020-11-03:
@TheJulianJES No problem - thanks for taking the time to let me know regardless.

FWIW I discovered that the Home Assistant AppDaemon add-on "ControllerX" has a system built into it for translating colour temps to RGB values (as per https://xaviml.github.io/controllerx/start/type-configuration, _color_wheel: color_temp_wheel_). The actual translation works well, albeit introduces some quirks in other parts of the ControllerX "system". 

Just thought I'd mention, as referencing the code which powers that system may be helpful for you (or anyone else) that tries to tackle this in the future.

odinho, 2021-08-06:
It does still happen with one of my GU10 IKEA lights. It isn't set up with color. I've been trying to debug/set up an env to do it, but it has proven too much of a hurdle since it doesn't look very straight-forward to do.
I now also have a E27 IKEA RGB bulb, and a E16 (or the small one), and they both seem to work. I also bought a no-name GU10 to replace the IKEA one since HomeAssistant wasn't able to make it work. I didn't test the latest update (it's out now), but the one before I still had it in and did a full reset and last-try.

If it's expected anything changed, I can of course test it again. Maybe one day I'll have the stamina to set up a test env, and I can figure out what hack makes it work. (Aside: I looked at zigbee2mqtt's translation layer , and it seems easier to understand and read, -- there's been no effort to actually use the same compatibility layer there for ZHA instead of going it alone?)

R5fan, 2021-08-29:
My problem is different, but related, and might shed some light on the problem.. I have some ZB-CL01 bulbs by eWeLight that are RGBWW. Due to some issues with my conbee, I have them temporarily connected to an ikea tradfri hub. Using the Ikea app, I have no control over color temp, only brightness and color, and once a color is selected, its stuck in color mode, with no way to get back to white mode. In Home Assistant, using the tradfri integration, I *can* control either RGB or color temp when controlling the light directly from the frontend. When trying to save that state in a scene, I get the same problem as the ikea app, activating the (HA) scene makes it go in color mode.

Im not sure what to conclude from this.

closed 2022-03-04 by github-actions[bot]

danielholm, 2024-12-10:
It's still an issue. Should I report a new bug?