zigpy/zha-device-handlers #464: [Device Support Request]

rwoodslap, 2020-08-24:
I purchased a Konke multi-function button and connected it to Home Assistant.  The device shows up on the network, but when I press the button, the following error message shows up in the ZigBee log:

[0xaab3:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=83 command_id=Command.Report_Attributes>
Failed to parse message (b'08530a00001080') on cluster 6, because 128 is not a valid Bool

I'd like to know if someone can help me figure out how to make the button work (I purchased it for less than $10 from Amazon) and if I'll need this library to do that.

Here is the device signature:
{
  "node_descriptor": "<Optional byte1=2 byte2=64 mac_capability_flags=128 manufacturer_code=4098 maximum_buffer_size=82 maximum_incoming_transfer_size=82 server_mask=11264 maximum_outgoing_transfer_size=82 descriptor_capability_field=0>",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0002",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "Konke",
  "model": "3AFE170100510001",
  "class": "zigpy.device.Device"
}

pipiche38, 2020-08-24:
> [0xaab3:1:0x0006] ZCL deserialize: <ZCLHeader frame_control= manufacturer=None tsn=83 command_id=Command.Report_Attributes>

Indeed The Konke Multifunction Switch is a bit special and do not send a Bool on Cluster 0x0006 Attribute 0x0000 !

Basically here are the value I have identified coming from that Switch

( 0x01 , 0x80 ) # Simple Click
( 0x02, 0x81 ) # Multi Click
( 0x82 ) # Long Click
( 0xcd ) # Short Reset ( short click on the reset button )

Welcome in the nice world of Zigbee !!!!

rwoodslap, 2020-08-24:
Is there a plan to support these events from Home Assistant?  I would love to use the button and have these events trigger something, but it that is not in the works, then I'm just out the $9 I paid for the button. :)

PR #617 (2020-12-04): Add Support for the Konke Button
Thank you @dmulcahey for the assist!

fixes: #464 & #300 🎉
closed 2020-12-04 by dmulcahey