zigpy/zha-device-handlers #465: [BUG]Hue bulbs with ZHA delays or not responding in UI

blair287, 2020-08-25:
## The problem

When switching on a scene containing several hue bulbs all controlled via ZHA 5/10 times some do not turn on or they turn on but UI says they are off. If the devices are in a scene the scene will hang until the bulbs update their state or about 5 seconds pass then another scene can be run other wise nothing responds, this also occurs when they are not in a scene but appears to be less frequent but still happens a lot.

This issue used to happen very infrequently but now my lights do not work more than 50% of the time.

I have hue remotes activating a scene when pressed, the event is seen my HA for the remote button press, the lights seem to actually turn on most of the time but the UI says they are off or they turn on but there is over a 5 second delay till they do. This used to be instant.

## Environment

HA supervised inside proxmox on intel nuc

- Home Assistant Core release with the issue: 
0.114.3
- Last working Home Assistant Core release (if known): 
0.112 i do not recall it being this bad
- Operating environment (OS/Container/Supervised/Core): 
supervised inside proxmox
- Integration causing this issue: 
Possibly ZHA
- Link to integration documentation on our website: 

## Problem-relevant `configuration.yaml`

```yaml

```

## Traceback/Error logs

Here is an example the light actually turned on but it looks like something failed and the UI showed it as off, which then prevents another scene from running.

```(MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 174>, 0]
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 189: e9
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xbd 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xe60a endpoint=11>, status: 0xe9
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Error while sending 189 req id frame: TXStatus.MAC_NO_ACK
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy.device] [0xe60a] Delivery error for seq # 0xbc, on endpoint id 11 cluster 0x0006: message send failure
2020-08-25 07:54:41 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xe60a:11:0x0006]: command failed: on exception: [0xe60a:11:0x0006]: Message send failure
2020-08-25 07:54:41 DEBUG (MainThread) [homeassistant.components.zha.entity] light.office_light: turned on: {'move_to_level_with_on_off': [4, <Status.SUCCESS: 0>], 'on_off': DeliveryError('[0xe60a:11:0x0006]: Message send failure')}
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (24, 193, 0, <DeconzAddressEndpoint address_mode=2 address=0x3591 endpoint=11>, 260, 768, 1, b'\x01\xc0\x075>\n\x17\x01\x00', 2, 0)
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 174>, 0]
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 46>, 193]
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [28, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|APSDE_DATA_CONFIRM|2: 46>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0f4e>, 11, 260, 6, b'\x18\xbe\x0b\x01\x00', 0, 175, 215, 160, 125, 171, 0, -67]
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy.zcl] [0x0f4e:11:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=190 command_id=Command.Default_Response>
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0f4e>, ep: 11, profile: 0x0104, cluster_id: 0x0006, data: b'18be0b0100'
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)

```
Here is another this is happening on all bulbs

```2020-08-25 08:10:18 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.processor_use, old_state=<state sensor.processor_use=2; unit_of_measurement=%, friendly_name=Processor use, icon=mdi:memory @ 2020-08-25T08:07:18.537308+01:00>, new_state=<state sensor.processor_use=3; unit_of_measurement=%, friendly_name=Processor use, icon=mdi:memory @ 2020-08-25T08:10:18.545293+01:00>>
2020-08-25 08:10:19 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x3591:11:0x0006]: command failed: off exception: 
2020-08-25 08:10:19 DEBUG (MainThread) [homeassistant.components.zha.entity] light.office_play_2: turned off: ```
## Additional information

Adminiuga, 2020-08-25:
it wasn't successful in sending the request out:
```
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xbd 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xe60a endpoint=11>, status: 0xe9
2020-08-25 07:54:41 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Error while sending 189 req id frame: TXStatus.MAC_NO_ACK
```

Adminiuga, 2020-08-25:
not a problem with zha or device handlers

blair287, 2020-08-26:
> it wasn't successful in sending the request out:
>
> ```
> 2020-08-25 07:54:41 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xbd 'aps_data_confirm' for <DeconzAddressEndpoint [...]

how come this is happening more and more went from a nearly perfect working ZHA set up to 50% of the time lights not coming on even though in UI they say they are on.

Adminiuga, 2020-08-27:
there were no changes to zigpy_deconz for a really long time. Zigpy_deconz is what ZHA uses to communicate with ConBee. 
ConBee tells zigpy_deconz it did not sent the packet, what else ZHA is supposed to do if the problem is reported by the network?

meremortals70, 2020-08-28:
> how come this is happening more and more went from a nearly perfect working ZHA set up to 50% of the time lights not coming on even though in UI they say they are on.

What firmware are you running on your Conbee? I was having a lot of issues with ZHA before updating the firmware in my Conbee.

The only Philips lights I have any issues with now are the LWB010's which are rubbish anyway.

Adminiuga, 2020-09-05:
and what channel are you running on? Don't run on channel 11

blair287, 2020-09-05:
> and what channel are you running on? Don't run on channel 11
I am on channel 11. How come I shouldn't use it?

dmulcahey, 2020-09-06:
It really depends on what channels your WiFi runs on. Zigbee and WiFi share spectrum. Take a peek at: https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html

TheJulianJES, 2020-09-06:
Not sure if this is related, but I'm also getting:
``[zigpy_deconz.zigbee.application] Error while sending 64 req id frame: TXStatus.NWK_BT_TABLE_FULL``
and
``[zigpy_deconz.zigbee.application] Error while sending 82 req id frame: TXStatus.MAC_NO_ACK``
when activating a scene with multiple Philips Hue lights.
Most of the time, it takes about 3-4 tries to get all lights turning on and changing to the correct color.
Tried with older ZLL hue bulbs and with the newer ZHA hue bulbs.

Could be related to network connectivity - I tried simply turning off all WiFi access point (LQI is around 190 for most bulbs), but this didn't really improve the situation. (ConBee II is hooked up with a USB 2.0 2m extension cable to a USB 3.0 port (there are no USB 2.0 ports on that machine))
But I'm also not sure how and if the BT_TABLE_FULL error can be solved.
(Or what it really means)

blair287, 2020-09-06:
> It really depends on what channels your WiFi runs on. Zigbee and WiFi share spectrum. Take a peek at: https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html. 15 tends to be a decent channel.

My WiFi is on channel 11 so opposite end to ZigBee channel 11. 

It is getting a bit annoying didn't have any issues now quite frequently I'll turn off 3 bulbs and 1 of them will stay on. Never used to happen. Or if I turn on 3 bulbs only 1 will.

dmulcahey, 2020-09-06:
I’m sorry it’s annoying but your logs show that there are delivery errors. There is nothing we can do from the ZHA side to remediate that and there are many things that can cause this. I know this isn’t an answer anyone likes but I’m out of ideas. If it were me I’d start looking at the path the packets are taking to the devices. Maybe set up the zha_map component and the network visualization card and see what the troubled device is connected to.

Adminiuga, 2020-09-06:
What about neighbors? Do wifi scan

TheJulianJES, 2020-09-07:
Some other people also reported having issues with ConBee, although I'm unsure whether or not this could even be related: https://github.com/home-assistant/core/issues/38642#issuecomment-688252747

blair287, 2020-09-07:
> What about neighbors? Do a wifi scan

I did channel 11 on ZigBee is the most empty channel for the WiFi around me.

Adminiuga, 2020-09-07:
So I take the "most empty" != empty. Zigbee channel 11still overlaps with WiFi channel 1. Move to channel 25 or 26

blair287, 2020-09-08:
> So I take the "most empty" != empty. Zigbee channel 11still overlaps with WiFi channel 1. Move to channel 25 or 26

ZigBee channel 11 and WiFi channel 11 are the furthest apart you can be.

blair287, 2020-09-10:
> So I take the "most empty" != empty. Zigbee channel 11still overlaps with WiFi channel 1. Move to channel 25 or 26

Moved my router to channel 1 and set ZigBee to channel 25 absolutely no change same issues. Sometimes it works perfectly other times there is a huge delay and you can see home assistant hanging at exciting scene when it contains hue bulbs. Or you send off to 3 bulbs and only 1 or 2 turn off.

dmulcahey, 2020-09-27:
@blair287 Are you still having this issue? IIRC you were having issues w/ IKEA blinds and I believe I saw that you mentioned that those issues just vanished... Is there any chance that they resolved after this change in channels mentioned above?

blair287, 2020-10-01:
> @blair287 Are you still having this issue? IIRC you were having issues w/ IKEA blinds and I believe I saw that you mentioned that those issues just vanished... Is there any chance that they resolved after this change in channels mentioned above?

Still have the issue with hue bulbs the blind started working fine before the channel change no idea why that issue went away but saw someone else with same issue with blind so not sure what's going on.

dmulcahey, 2020-10-10:
how do you have the conbee stick setup? How is it connected to the machine?

blair287, 2020-10-10:
> how do you have the conbee stick setup? How is it connected to the machine?

Plugged into intel nuc via USB 2.0 2m extension

The blind delay is back after updating HA from 114.4 to 116.2 5 seconds again from command issue to it moving.

closed 2021-08-13 by github-actions[bot]