zigpy/zha-device-handlers #466: [BUG] Yale Real Living YRD210 Lock not updating state in HA

meremortals70, 2020-08-26:
I have just newly installed this lock.  It added to HA and is using the zhaquirk.

It does not matter if the door is locked or unlocked the state is not updating.

State always shows Locked
[image]

Double-checked in Developer>States with the same result.

[image]

Adminiuga, 2020-08-26:
Install a Zigbee router as well

meremortals70, 2020-08-26:
? What an inane response.  Over half of my zigbee devices are routers. This device was joined via a router.

[image]

dmulcahey, 2020-08-26:
@meremortals70 Was that necessary?

Adminiuga, 2020-08-27:
response was given based on information you presented and there are reasons for that specific response. 
zha-device-handlers is not a problem here. 
Enable debug logging and confirm that attribute reporting is successfully configured, then confirm that device actually sends attribute updates and lock events.
Ah, and check the correct part number is being used. some people tried to use C4 one and it didn't work, although it would pair fine.

meremortals70, 2020-08-28:
> @meremortals70 Was that necessary?

Sorry David, but when I get a response like that what else can be expected? It didn't ask for further information, just install a router, where? The lock is less than 3m from my Conbee and I have around 70 routers in my network.

Inane (without meaning) aptly characterises the response.

I would rather no response than something like that.

meremortals70, 2020-08-28:
> response was given based on information you presented and there are reasons for that specific response.
>
> zha-device-handlers is not a problem here.
>
> Enable debug logging and confirm that attribute [...]

Much better response thanks. Strangely after an update to HA 0.114.4 the lock disconnected from the network and showed as unknown manufacturer. I deleted it from the network, rebooted HA rejoined it to the network and state is now working.

Adminiuga, 2020-08-28:
> It didn't ask for further information, just install a router

Exactly same feeling about `I have just newly installed this lock.`
So??? is it your 1st device you installed? Do you have any other devices on your network? are they routers? Did you look in the open/closed issues list to see if there was any similar issues, like https://github.com/home-assistant/core/issues/36824 you should use that issue as an example on how to report the problem. Much better chances to get help...

closed 2020-08-28 by Adminiuga

meremortals70, 2020-09-01:
Well it wasn't the same issue at all. How about we agree there was fault on both sides and move on.

I have certainly found people on HA seem really sensitive even though the docs are less than they should be. I have offered many times to help improve docs with no response.