zigpy/zha-device-handlers #469: [Device Support Request] Xiaomi Motion Sensor RTCQ11LM (auto reset)

pipiche38, 2020-09-01:
The Xiaomi motion sensor ( zhaquirks.xiaomi.aqara.motion_aq2.MotionAQ2 ) is only sending a Occupancy bitmap8.1 when a motion is detected. However it never send bitmap8.0 .

Would it be possible that the quirk is making a bitmap8.0 after 15s ?

dmulcahey, 2020-09-01:
Quirks do handle this already

pipiche38, 2020-09-01:
Great. I'm new in the zigpy worlkd, trying to develop a plugin for Domoticz.
Do I have to do something to enable it ? Do I have to use a specific version ?

Adminiuga, 2020-09-01:
https://github.com/zigpy/zha-device-handlers/blob/dev/CONTRIBUTING.md

`import zhaquirks` before you start the radio.

dmulcahey, 2020-09-01:
Just import it before starting up. Look at the ZHA code in HA

Adminiuga, 2020-09-01:
1st one

dmulcahey, 2020-09-01:
Damn, Alexi beat me again 😂

pipiche38, 2020-09-01:
So did I, otherwise it wouldn't have been identified as  zhaquirks.xiaomi.aqara.motion_aq2.MotionAQ2

Adminiuga, 2020-09-01:
it should send the command to IAS cluster subscribers then

pipiche38, 2020-09-01:
As said, I received:

Received an attribute update 0=bitmap8.1 on cluster <zigpy.zcl.clusters.measurement.OccupancySensing object at 0xaae5c9a0> from device <zigpy.device.Device object at 0xaae5c628>/00:15:8d:00:03:02:1

but nothing else . Of course I received correctly the Lux on Cluster 0x0400

Adminiuga, 2020-09-01:
quirks is not going to generate Zigbee messages. you need to add your "listeneres" to the cluster. quirks will send commands to those listeners.

Adminiuga, 2020-09-01:
https://github.com/zigpy/zha-device-handlers/blob/1235e8434a68db70590cdb9ee77dbd0477f8e819/zhaquirks/xiaomi/__init__.py#L312-L334

pipiche38, 2020-09-01:
For each new device I'm doing the following, is that not enough ?

```
            for cluster in endpoint.in_clusters.values():
                # The context listener passes its own object as the first argument
                # to the callback
                cluster.add_context_listener(self)
```

Do you mean that I have to set the timer myself ?

Adminiuga, 2020-09-01:
the `self` instance, need's to have `attribute_updated()` method to receive notification about attribute change. For IAS clusters it needs to have `cluster_command()` method. 
Again, look how ZHA does it in zha/core/channels/*

Adminiuga, 2020-09-01:
I can't answer your question, because I don't know what `self` is in your case

Adminiuga, 2020-09-01:
you may look at HA versions about 0.80.0, a bit easier to understand

pipiche38, 2020-09-02:
not solved, just gave up

closed 2020-09-02 by pipiche38

Hedda, 2021-06-08:
https://github.com/pipiche38/Domoticz-Zigpy#to-be-addressed

> This is using the "quirk" part (from ZHA Device Handlers), and I do not fully understand how it works (especially in regards to that issue https://github.com/zigpy/zha-device-handlers/issues/469 - I do not understand what Cluster IAS should do here!)

@pipiche38 By the way, noticed that Adminiuga comment at the bottom of this other discussion which might be relevant here:

https://github.com/zigpy/zha-device-handlers/discussions/625

"**_ZHA really doesn't care if there's a single basic cluster on one of the endpoints or if there are multiple basic clusters. It always would use the 1st basic cluster it founds when processing all of the endpoints._**"