zigpy/zha-device-handlers #476: Re-Pair does not working for IKEA tradfri E1810 and E1744 Symfonisk

peterson1190, 2020-09-07:
**Describe the bug**
Sorry if I'm at the wrong topic.
I'm using 114.4 HASSIO with Tasmotized Sonoff Zigbee Bridge with ncp-uart-sw_6.7.6_115200.ota on it.

First I paired it successfully, but the dimmer somehow controlled my Xiaomi Lumi plugs with the power button, without any configuration. So I went ahead and tried to re-pair it, but with no luck.

Pairing log
`[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_req: [60, <Bool.false: 0>]
[0xa151:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=124 command_id=Command.Read_Attributes>
[0xa151:1:0x000a] ZCL request 0x0000: [[0]]
[0x67df:1:0x0b04]: async_update
[0x0d4b:1:0x0b04]: async_update
Device 0xd3ab (68:0a:e2:ff:fe:d2:54:b0) joined the network
[0xd3ab] Requesting 'Node Descriptor'
Tries remaining: 2
[0xd3ab] Extending timeout for 0xb2 request
Device 0xd3ab (68:0a:e2:ff:fe:d2:54:b0) joined the network
Skip initialization for existing device 68:0a:e2:ff:fe:d2:54:b0
[0x2650:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=112 command_id=Command.Read_Attributes>
[0x2650:1:0x000a] ZCL request 0x0000: [[0]]
Device 0xd3ab (68:0a:e2:ff:fe:d2:54:b0) joined the network
Skip initialization for existing device 68:0a:e2:ff:fe:d2:54:b0
[0x6c0f:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=58 command_id=Command.Report_Attributes>
[0x6c0f:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=5805>>]]
[0x6c0f:1:0x0405] Attribute report received: measured_value=5805
[0x67df:1:0x0b04]: async_update
[0x0d4b:1:0x0b04]: async_update
[0xd3ab] Delivery error for seq # 0xb2, on endpoint id 0 cluster 0x0002: message send failure
Tries remaining: 1
[0xd3ab] Extending timeout for 0xb5 request
[0x67df:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=206 command_id=Command.Read_Attributes>
[0x67df:1:0x000a] ZCL request 0x0000: [[0]]
[0xf830:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=113 command_id=Command.Read_Attributes>
[0xf830:1:0x000a] ZCL request 0x0000: [[0]]
Device 0xd3ab (68:0a:e2:ff:fe:d2:54:b0) joined the network
Skip initialization for existing device 68:0a:e2:ff:fe:d2:54:b0
Device 0xd3ab (68:0a:e2:ff:fe:d2:54:b0) joined the network
Skip initialization for existing device 68:0a:e2:ff:fe:d2:54:b0
Device 0xd3ab (68:0a:e2:ff:fe:d2:54:b0) joined the network
Skip initialization for existing device 68:0a:e2:ff:fe:d2:54:b0
[0xa151:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=125 command_id=Command.Read_Attributes>
[0xa151:1:0x000a] ZCL request 0x0000: [[0]]
[0x67df:1:0x0b04]: async_update
[0x0d4b:1:0x0b04]: async_update
[0xd3ab] Delivery error for seq # 0xb5, on endpoint id 0 cluster 0x0002: message send failure
[0xd3ab] Requesting Node Descriptor failed
[0xd3ab] Discovering endpoints
Tries remaining: 3
[0xd3ab] Extending timeout for 0xba request
[0x6c0f:1:0x0405] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=59 command_id=Command.Report_Attributes>
[0x6c0f:1:0x0405] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=uint16_t, value=5865>>]]
[0x6c0f:1:0x0405] Attribute report received: measured_value=5865
[0x2650:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=113 command_id=Command.Read_Attributes>
[0x2650:1:0x000a] ZCL request 0x0000: [[0]]
[0x67df:1:0x0b04]: async_update
[0x0d4b:1:0x0b04]: async_update
[0xd3ab] Delivery error for seq # 0xba, on endpoint id 0 cluster 0x0005: message send failure
[0x67df:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=7 command_id=Command.Read_Attributes>
[0x67df:1:0x000a] ZCL request 0x0000: [[0]]
Tries remaining: 2
[0xd3ab] Extending timeout for 0xbe request
[0xf830:1:0x000a] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=114 command_id=Command.Read_Attributes>
[0xf830:1:0x000a] ZCL request 0x0000: [[0]]
`

*To Reproduce*
1. Pair the above mentionet devices.
2. Delete the devices under ZHA integration
3. Try to Re-Pair it.

Adminiuga, 2020-09-07:
Try the latest try the latest ha dev tree and/or new beta once it is released today

peterson1190, 2020-09-07:
Sadly I can not join beta in Supervisor, I press join and for the popup also but nothing happens,I'm still on the stable channel.

So I need to do it manually and I found another issue with similar devices where you told the guy to download ZHA and put it in custom_components folder then edit the manifest Json inside it, with a Git link.
The problem is that this link [git+https://github.com/zigpy/bellows@multiversion#bellows==0.22.0] is not working anymore.
Could you please provide a new one, this is the only missing piece from my puzzle.
Thanks.

Adminiuga, 2020-09-07:
Yes, I've deleted that branch since it was merged. New link 
[git+https://github.com/zigpy/bellows@dev#bellows==0.22.0] the part after @ is the branch name

peterson1190, 2020-09-07:
Okay, so I did put zha folder into custom components. I edited the manifest.json at config/custom_components/zha

Now it looks like this:

`{
  "domain": "zha",
  "name": "Zigbee Home Automation",
  "config_flow": true,
  "documentation": "https://www.home-assistant.io/integrations/zha",
  "requirements": [
    "git+https://github.com/zigpy/bellows@dev#bellows==0.22.0",
    "pyserial==3.4",
    "zha-quirks==0.0.44",
    "zigpy-cc==0.5.2",
    "zigpy-deconz==0.9.2",
    "zigpy==0.23.1",
    "zigpy-xbee==0.13.0",
    "zigpy-zigate==0.6.2",
    "zigpy-znp==0.1.1"
  ],
  "codeowners": ["@dmulcahey", "@adminiuga"]
}
`
I tried including it in config.yaml as zha:
When HA starts it says unable the load zha component.
Do I need to remove the original component from integrations before?
 
Error log: Setup failed for zha: Unable to import component: cannot import name 'PERCENTAGE' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)

Adminiuga, 2020-09-07:
The dev must have incompatible changes with master. Just wait till beta is released.

peterson1190, 2020-09-07:
Updated tasmota to 8.5.0.0, now it worked on first try, but I'm facing the same issue as before. 
The freshly paired dimmer controls my Lumi zigbee plugs without any configuration. (When I press the on/off button on the 5 button dimmer the plugs turn on and when I press it again they turn off.)
What can cause this issue?

Adminiuga, 2020-09-07:
Dimmer could be sending a multicast on group 0x0000 on broadcast endpoint.
I think opple switch behaves the same way until you bind it to a group. Hard to say without debug logs and/or traffic capture

peterson1190, 2020-09-08:
These are the button presses:
[0x4256:1:0x0006] ZCL request 0x0002: []
[0x4256:1:0x0006] No handler for cluster command 2
[0xe7c2:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=88 command_id=Command.Report_Attributes>
[0xe7c2:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.false>>, <Attribute attrid=61440 value=<TypeValue type=uint32_t, value=121787981>>]]
[0xe7c2:1:0x0006] Attribute report received: on_off=0, 61440=121787981
[0x4256:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=78 command_id=2>
[0x4256:1:0x0006] ZCL request 0x0002: []
[0x4256:1:0x0006] No handler for cluster command 2
[0xe7c2:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=93 command_id=Command.Report_Attributes>
[0xe7c2:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.true>>, <Attribute attrid=61440 value=<TypeValue type=uint32_t, value=121787982>>]]
[0xe7c2:1:0x0006] Attribute report received: on_off=1, 61440=121787982
[0xba97:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=14 command_id=Command.Report_Attributes>
[0xba97:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.true>>]]
[0xba97:1:0x0006] Attribute report received: on_off=1
[0x4256:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=79 command_id=2>
[0x4256:1:0x0006] ZCL request 0x0002: []
[0x4256:1:0x0006] No handler for cluster command 2
[0xe7c2:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=106 command_id=Command.Report_Attributes>
[0xe7c2:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.false>>, <Attribute attrid=61440 value=<TypeValue type=uint32_t, value=121787983>>]]
[0xe7c2:1:0x0006] Attribute report received: on_off=0, 61440=121787983
[0x4256:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=80 command_id=2>
[0x4256:1:0x0006] ZCL request 0x0002: []
[0x4256:1:0x0006] No handler for cluster command 2
[0xe7c2:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=109 command_id=Command.Report_Attributes>
[0xe7c2:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.true>>, <Attribute attrid=61440 value=<TypeValue type=uint32_t, value=121787984>>]]
[0xe7c2:1:0x0006] Attribute report received: on_off=1, 61440=121787984
[0x67df:1:0x0b04]: async_update
[0x0d4b:1:0x0b04]: async_update
[0xe7c2:1:0x0b04]: async_update
[0x4256:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=81 command_id=2>
[0x4256:1:0x0006] ZCL request 0x0002: []
[0x4256:1:0x0006] No handler for cluster command 2
[0xe7c2:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=115 command_id=Command.Report_Attributes>
[0xe7c2:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.false>>, <Attribute attrid=61440 value=<TypeValue type=uint32_t, value=121787985>>]]
[0xe7c2:1:0x0006] Attribute report received: on_off=0, 61440=121787985
[0x4256:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=82 command_id=2>
[0x4256:1:0x0006] ZCL request 0x0002: []
[0x4256:1:0x0006] No handler for cluster command 2
[0xe7c2:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=118 command_id=Command.Report_Attributes>
[0xe7c2:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.true>>, <Attribute attrid=61440 value=<TypeValue type=uint32_t, value=121787986>>]]
[0xe7c2:1:0x0006] Attribute report received: on_off=1, 61440=121787986
[0x4256:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=CLUSTER_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=83 command_id=2>
[0x4256:1:0x0006] ZCL request 0x0002: []
[0x4256:1:0x0006] No handler for cluster command 2
[... cut ...]
[0x4256:1:0x0006] No handler for cluster command 2
[0xe7c2:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=123 command_id=Command.Report_Attributes>
[0xe7c2:1:0x0006] ZCL request 0x000a: [[<Attribute attrid=0 value=<TypeValue type=Bool, value=Bool.true>>, <Attribute attrid=61440 value=<TypeValue type=uint32_t, value=121787988>>]]
[0xe7c2:1:0x0006] Attribute report received: on_off=1, 61440=121787988

closed 2020-09-08 by peterson1190